dresden-elektronik/deconz-rest-plugin #6773: Cannot Decouple Aqara WS-EUK04 (2-gang w/ neutral wire)

Marfre888, 2023-03-02:
## Describe the bug

As the title suggests, I am trying to decouple an Aqara WS-EU04 using deconz. However, when I try to read the attributes from 'Xiomi Disconnect 1' and 'Xiomi Disconnect 2' they greyed out, despite showing up as a rw attribute.

## Steps to reproduce the behavior

Click on 'Read', to read attributes, inside of deconz gui.

## Expected behavior

Gui should allow me to change switch to decoupled mode.

## Screenshots

[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian (Homebridge)
 - Firmware version: (26780700)
 - deCONZ version: (2.20.1)
 - Device: ConBee II
 - Do you use an USB extension cable:  no
 - Is there any other USB or serial devices connected to the host system? no

BabaIsYou, 2023-03-03:
Hi,

When attributes become greyed after clicking on button `read`, it only means that this device doesn't present this attributes. Then they cannot be read, neither written.

Marfre888, 2023-03-03:
> Hi,
> 
> AFAIK, when attributes become greyed after clicking on button `read`, it only means that this device doesn't present this attributes. Then they cannot be read, neither written.
> 
> Suggest to have a look at #6140 to see how it's handled for this device thru `config/clickmode` using REST-API

Yes understood, however reading through other issues it is definitely possible to change this attribute.

Am I missing something? Could load a DDF solve this issue?

BabaIsYou, 2023-03-03:
The DDF for it is in the previously referenced issue. But DDF could not activate an attribute in GUI that the device seems not exposed. You could use Rest-ApI (curl, http request or third party app) to get access to this.

Marfre888, 2023-03-03:
> The DDF for it is in the previously referenced issue. But DDF could not activate an attribute in GUI that the device seems not exposed. You could use Rest-ApI (curl, http request or third party app) to get access to this.

I don't currently have access to the switches to test, but should do so this weekend.

Is there any documentation that can aid me with decoupling these switches?

I've already had a look at the rest-api but didn't notice anything referencing a coupled/decoupled state.

BabaIsYou, 2023-03-03:
Don't know your level of knowledge about REST-API and DDF then I suggest to pass thru : https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet and https://dresden-elektronik.github.io/deconz-rest-doc/about_rest/

But it seems that your specific device WS-EUK04 doesn't have decoupled mode in the DDF I referenced. Only WS-EUK01 and WS-EUK03 seems to have this feature if DDF are complete in #6140 

What are the issues you read that say it's possible to change this attribute ?

BabaIsYou, 2023-03-03:
> What I am referencing is #5460. I might be wrong but it seems the comment states that decoupled works for WS-EUK01/02/03/04.

As I can understand after reading this issue,  it seems to not be possible for this device at this time …

Marfre888, 2023-03-04:
> > What I am referencing is #5460. I might be wrong but it seems the comment states that decoupled works for WS-EUK01/02/03/04.
> 
> As I can understand after reading this issue, it seems to not be possible for this device at this time …

Oh no, that’s horrible! I have a bunch of Philips Hue lights that I was hoping to control with the switch, and is primarily the reason why I want to move the lights into deconz.

Is it possible to get it done? 

Would it be possible to get it to decoupled mode in the Aqara hub, then move it to Deconz in the same state?

SwoopX, 2023-03-04:
Gentlemen, as this is not a bug, I'm closing this issue.

The device should be reset and re-paired with the most recent firmware. #5460 explains how to set decoupled mode.

closed 2023-03-04 by SwoopX

Marfre888, 2023-03-05:
> Gentlemen, as this is not a bug, I'm closing this issue.
> 
> The device should be reset and re-paired with the most recent firmware. #5460 explains how to set decoupled mode.

Apologies.

Going through #5460 I foolishly assumed attribute 0x0200 simply locked the switch and made it unusable. Can say I’ve successfully managed to decouple the switch, and it seems the button events still reads through.

My next question is: Assuming I now want to add a stateless switch to the decoupled switches, is there an easy way to do this without having to manually add a CLIPSwitch for each switch? I’m using Homebridge Hue to push the switches into Homekit.

If not could you point me to the documentation that explains how to create a CLIPSwitch. It’s been a while since I last created these in Deconz, and can’t find the documentation that explains how anymore!

Thank you so very much for your help!