dresden-elektronik/deconz-rest-plugin #6775: SYMFONISK sound remote gen2

LiamTheCat, 2023-03-03:
## Device
- Product name: SYMFONISK gen2
- Manufacturer: IKEA of Sweden
- Model identifier: SYMFONISK sound remote gen2
- Device type : 
  - Switch

## Screenshots
[image]

### Basic
[image]

### Power Configuration
[image]

### Identify
[image]

### Poll Control
[image]

### ZLL Commissioning
[image]

### Identify
[image]

### Group
[image]

### On/Off 
[image]

### Level Control
[image]

### OTAU
[image]

### ZLL Commissioning
[image]

Mimiix, 2023-03-03:
Hi,

Looks like the values are not populated in the screenshots. can you please click "read" on all cluster screenshots and make sure the values are populated?

Kind regards,

LiamTheCat, 2023-03-03:
Thank you very much.
I have clicked read everywhere. Here are the new screenshots: 

### Basic
[image]

### Power Configuration
[image]

### Identify
[image]

### Poll Control 
[image]

### OTAU
[image]

BR

ebaauw, 2023-03-03:
We need to sniff the commands it sends on each button.

I'll pick one up, next time I'm at IKEA.
See https://www.ikea.com/nl/nl/p/symfonisk-afstandsbediening-voor-geluid-gen-2-90524607/.

bierlos, 2023-03-04:
i have the same problem. bought on today. seems, it's like a symfonisk gen 1 plus 2 scene buttons with one and two points. those scene buttons have 3 steps: click, doubleclick and long click

ebaauw, 2023-03-05:
> it's like a symfonisk gen 1

Huh?  The first SYMFONISK controller has a dial and a single button.
https://www.ikea.com/au/en/p/symfonisk-sound-remote-black-70433788/

ebaauw, 2023-03-15:
Picked one up today.  It pairs with deCONZ immediately and a ZHASwitch resource is created automatically (probably by the legacy code doing a _Model Identifier_ starts with).  Firmware v1.0.012.  Haven't yet paired it to the dirigeria hub.

Sniffer reveals the following commands, all sent as unicast to the coordinator, without any bindings.

Button | Action | Command
-- | -- | --
_Play_ | Press/Release | _Toggle_
_Play_ | Hold | _Toggle_
_Play_ | Release after Hold | --
`+` | Press/Release | _Move with On/Off_ (_Up_, 255)
`+` | Hold | _Move_ (_Up_, 255), repeated every 0.2s while holding
`+` | Release after Hold | --
`-` | Press/Release | _Move with On/Off_ (_Down_, 255)
`-` | Hold | _Move_ (_Down_, 255), repeated every 0.2s while holding
`-` | Release after Hold | --
_Previous_ | Press/Release | _Step_ (_Down_, 1, 0)
_Previous_ | Hold | _Step_ (_Down_, 1, 0)
_Previous_ | Release after Hold | --
_Next_ | Press/Release |  _Step_ (_Up_, 1, 0)
_Next_ | Hold |  _Step_ (_Up_, 1, 0)
_Next_ | Release after Hold | --
`.` | Press/Release | 0xFC7C/0x01 (0101)
`.` | Double Press | 0xFC7C/0x01 (0102)
`.` | Hold | 0xFC7C, cmd 0x01 (0103)<br>_Identify_ (broadcast !)
`.` | Release after Hold | --
`..` | Press/Release | 0xFC7C, cmd 0x01 (0201)
`..` | Double Press | 0xFC7C, cmd 0x01 (0202)
`..` | Hold | 0xFC7C, cmd 0x01 (0203)
`..` | Release after Hold | --

So that would lead to the following `buttonevent` values:

`buttonevent` | Button | Action
-- | -- | --
1002 | _Play_ | Press/Release or Hold
2001 | `+` | Hold (every 0.2s)
2002 | `+` | Press/Release
3001 | `+` | Hold (every 0.2s)
3002 | `-` | Press/Release
4002 | _Previous_ | Press/Release or Hold
5002 | _Next_ | Press/Release or Hold
6002 | `.` | Press/Release (Single Press)
6003 | `.` | Long Press
6004 | `.` | Double Press
7002 | `..` | Press/Release (Single Press)
7003 | `..` | Long Press
7004 | `..` | Double Press

Note that per Hue specs, x001 events are expected to be issued repeatedly while holding the button.  I'm not sure I like the absence of a "closing" 2003 or 3003, but I don't want the API plugin to generate fake events based on not receiving anything for some time.

ebaauw, 2023-03-15:
Paired it to the Dirigera, and recevied a firmware update to v1.0.32.  That changes the device...  A lot.
[image]

I now has three endpoints; the `.` now sends commands from the 0xFC80 on endpoint 0x02; the `..` from 0xFC80 on endpoint 0x03.  It does require explicit bindings now (maybe the legacy API code created the bindings before?).  And _Battery Percentage Remaining_ now adheres to the standard, reporting the value in 0.5% - that's a first for IKEA.

The _Play_ buttons is bound to a group (as reported by the _ZLL Commissioning_ cluster).  Not sure if that's the device or the legacy code of the API plugin.  Consequently, it sends _Toggle_ to that group, even after binding _On/Off_ to the coordinator.  I cannot seem to ditch the group binding.  It visible in the binding table, but an _Unbind Requests_ results in a _No Entry_.

For `.` and `..` now seem to send
- 0x01 on initial press
- 0x02 on hold (only once)
- 0x03 on release after press
- 0x04 on release after hold
- 0x06 on double press

So
- press/release results in 0x01, 0x03;
- double press in 0x01, 0x06,
- long press in 0x01, 0x02 on press/hold, followed by 0x04 on release.

So that would result in `buttonevent` values 6000/6002; 6000/6004, and 6000/6001/6003.

Note to self: damn, we need a separate indication in the capabilities whether the x001 repeats or not.

ebaauw, 2023-03-15:
Above commit adds support for firmware v1.0.32.  Unfortunately, this firmware isn't (yet?) retrieved by `ikea-ota-download.py`.  The image type is 0x110E.

It should work for the original firmware as well, except for `.` and `..`.  I don't want to add support for `.` and `..` on firmware v1.0.012, since that would require C++ code to handle the two-byte payload for the 0xFC7C/0x01 command.

closed 2023-03-17 by manup

Mimiix, 2023-03-28:
@ebaauw Got one myself aswell. Would you mind sharing the OTA file so i can update it?

ebaauw, 2023-04-17:
Upgraded mine to 1.0.35 (found on https://github.com/Koenkk/zigbee-OTA).  No noticeable changes from 1.0.32.

Idaho947, 2023-04-30:
> Upgraded mine to 1.0.35 (found on https://github.com/Koenkk/zigbee-OTA). No noticeable changes from 1.0.32.

Hello how do you do ? When I try to select the file the field stay white :
[image]