dresden-elektronik/deconz-rest-plugin #6780: Aqara P1 presence sensor is reset to default values

JanLp, 2023-03-05:
Conbee II
Version 2.20.01
Firmware 26780700

The problem is only related to this sensor. I have several sensors where there are no problems.

I have a problem with RTCGQ14LM ([lumi.motion.ac02](https://dresden-elektronik.github.io/deconz-rest-doc/devices/xiaomi/xiaomi_rtcgq14lm_p1_presence_sensor/#aqara-p1-presence-sensor-rtcgq14lm))

I put e.g. duration to 120

```
{
"duration": 120,
}
```

or sensitivity to 1

```
{
"sensitivity": 1,
}
```

Shortly afterwards, these values are reset to the standard `"duration": 30` and `"sensitivity": 3`, the same applies to `ledindication`

```
{
    "config": {
        "battery": 100,
        "duration": 30,
        "ledindication": false,
        "on": true,
        "reachable": true,
        "sensitivity": 3
    },
    "ep": 1,
    "etag": "a996cf7fa7d89b88ab9ec03b120823fc",
    "lastannounced": null,
    "lastseen": "2023-03-04T11:38Z",
    "manufacturername": "LUMI",
    "modelid": "lumi.motion.ac02",
    "name": "Motion 1",
    "state": {
        "lastupdated": "2023-03-04T11:39:20.044",
        "presence": true
    },
    "swversion": "0.0.0_0006",
    "type": "ZHAPresence",
    "uniqueid": "54:ef:44:10:00:52:c4:d2-01-0406"
}
```

Mimiix, 2023-03-05:
Can you please stick to the template?

Mimiix, 2023-03-08:
> > Can you please stick to the template?
> 
> What information do you need to troubleshoot?

All that's in the template.

JanLp, 2023-03-08:
Less can't do it? It's not about complaining, just for the sake of complaining?
Please close it as incomplete if you think information is missing.

Mimiix, 2023-03-08:
We ask for the details for a purpose. It helps us help you.

closed 2023-03-23 by JanLp