dresden-elektronik/deconz-rest-plugin #6785: Aeotec Range Extender Zi

stijnmeijer, 2023-03-06:
## Device

- Product name: Range Extender Zi
- Manufacturer: AL001
- Model identifier: WG001-Z01
- Device type : 
  - Other: Range extender

## Screenshots

[image]

### Basic

[image]

### Identify

[image]
_Note: after pressing Read and/or Exec, the value remains 0_

### OTAU

[image]
[image]

All other clusters do not seem to contain any attributes, or are duplicate to the above.

Smanar, 2023-03-06:
>I'm not sure if this is correct, as it seems too straightforward.

Lol, yes. And it will be the first DDF for range extender

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "AL001",
  "modelid": "WG001-Z01",
  "product": "Aeotec Range Extender Zi",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_RANGE_EXTENDER",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```

IDK if your device have special feature that can be used in the DDF ?

Idaho947, 2023-03-09:
Works for me

Smanar, 2023-03-09:
You have this device too ?

Idaho947, 2023-03-09:
Yes I bought it. i am a device tester in real life.

Smanar, 2023-03-10:
And this device haven't special feature ? (So much endpoints and clusters for nothing)

Idaho947, 2023-03-10:
I don't know. For me it's just an extender

wvuyk, 2023-03-11:
why to expose an extender to the api if there is no added value then extend signals.....?

Idaho947, 2023-03-12:
For exemple to check if it's always on the network.

wvuyk, 2023-03-12:
You will not see any updates in the API to tell the device is still alive? You need to check on the deCONZ map. That is the place to check is a device is still on the network. 
Adding an extender to the api does not any value, there are no controls and there are no updates. Instead apps using the api need to provision ways to ignore these devices.  It really adds no value to the api?

Idaho947, 2023-03-12:
Ok but if your using a third party app for your domotic use and this third party use the api ?

wvuyk, 2023-03-12:
Yes, I am maintaining my own app here

wvuyk, 2023-03-12:
Sorry, my last answer was unclear. I am a writer of a thrid party app. That is also the reason for my earlier answer. Why would a third party app show a device that has no controls and no updates that are of interest for the user? For my app, I hide these devices for the user as the app can not show any assured confirmation it is alive?

Smanar, 2023-03-12:
Using state/reachable don't work ?

Idaho947, 2023-03-12:
Yes it works. But in my memories, my app use the api to import and use it.

Schmandre, 2024-01-09:
Hello,

I just tried to pair the device but can’t see anything in Web UI. Any information how to get that to work?

Using Phoscon in docker