dresden-elektronik/deconz-rest-plugin #6798: Moes Smart Zigbee Push Button Switch 2 Gang: _TZ3000_18ejxno0 / TS0012

MrMEScott, 2023-03-13:
## Device

- Product name: Moes Smart Zigbee Push Button Switch 2 Gang
- Manufacturer: _TZ3000_18ejxno0
- Model identifier: TS0012
- Device type : 
  - Switch

## Screenshots

[image]

### Basic
[image]

### Identify
[image]

### Groups
[image]

### Scenes
[image]

### On/Off
[image]

Mimiix, 2023-03-13:
Hi,

Please add all screenshots of all clusters in full. including the attributes section being read.

Kind regards,

MrMEScott, 2023-03-13:
Sorry about that. The cluster info panels are longer than my screen, so these screengrabs come in sections.
The content seems to be the same for both switch 1 and switch 2, with the exception of the On/Off attributes.
### Identify 
#### 1 of 2
[image]
#### 2 of 2
[image]
### Groups
#### 1 of 3
[image]
#### 2 of 3
[image]
#### 3 of 3
[image]
### Scenes
#### 1 of 5
[image]
#### 2 of 5
[image]
#### 3 of 5
[image]
#### 4 of 5
[image]
#### 5 of 5
[image]
### On/Off
#### Switch 1, 1 of 2
[image]
#### Switch 1, 2 of 2
[image]
#### Switch 2, attribute section (the switch was on when I grabbed this)
[image]

BabaIsYou, 2023-03-13:
Could you try this DDF ?
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZ3000_fvh3pjaz","_TZ3000_18ejxno0"],
  "modelid": ["TS0012","TS0012"],
  "product": "Battery 2 gang switches",
  "sleeper": true,
  "status": "Silver",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0000",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001"
        ],
        "out": [
          "0x0006"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x02"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```

MrMEScott, 2023-03-13:
I'd be happy to, if I knew how to! :)
Where do I put it / how do I install it?

MrMEScott, 2023-03-13:
Suppose I copied the DDF @BabaIsYou  posted, then what? I restart deconz, then does the device become discoverable by Phoscon? Do I have to re-pair it? Or is there some other step I need to take to make it visible to the Home Assistant deconz integration?

Mimiix, 2023-03-13:
> Suppose I copied the DDF @BabaIsYou posted, then what? I restart deconz, then does the device become discoverable by Phoscon? Do I have to re-pair it? Or is there some other step I need to take to make it visible to the Home Assistant deconz integration?

Restart deconz. Re pair might also work after that.

Mimiix, 2023-03-13:
> > Suppose I copied the DDF @BabaIsYou posted, then what? I restart deconz, then does the device become discoverable by Phoscon? Do I have to re-pair it? Or is there some other step I need to take to make it visible to the Home Assistant deconz integration?
> 
Restart deconz. Should show up in the rest api.

You can see the events in Phoscon under help, then api information.

Mimiix, 2023-03-13:
If it works, the hex name should be gone and has something like switch.

Additionally i am not sure if we need a button map?

BabaIsYou, 2023-03-13:
If you gain access to DeConz gui then you can edit DDF and simulate a save as … just to get the path ;-)

MrMEScott, 2023-03-13:
> If you gain access to DeConz gui then you can edit DDF and simulate a save as … just to get the path ;-)

I don't see any option to edit DDF in the deConz gui. I'm accessing it via VNC, and the only window on the desktop is `deCONZ - Conbee II (/dev/ttyACM0)`. I have no way to open other windows.
Unde deConz preferences I see an interface to add/remove ZCLDB XML files but I guess that's something different?

BabaIsYou, 2023-03-14:
You should have this
[image]
and then after selecting a device on the right map, you have to select the far right radio button to display the clusters list
[image]
after that slect the basic cluster and do a click on "read" button above attributes list
[image]
Then with Control+E or in the Edit menu you gain access to the DDF Editor
[image]

Then perhaps could you load the previously copied DDF and save it the place Deconz offers you to to do with Save as option

[image]

Hope this help with the other indication given by Smanar and Mimiix

BabaIsYou, 2023-03-14:
I also modified the DDF in the initial post because the fingerprint part was useless and false, then I propose a specific DDF for this device. Hope that it doesn't need buttonmap as Mimiix suggested.

MrMEScott, 2023-03-14:
I don't see a Control entry in the menu. Here's what I have:
[image]

This has opened a whole new can of worms.

I'm running the `marthoc/deconz` Docker image, which I hadn'r realised is now deprecated in favour of `deconzcommunity/deconz`. 
Should I migrate to the new one? Or maybe there's some startup parameter I can use on the existing container to surface the Control panel?

Mimiix, 2023-03-14:
> I don't see a Control entry in the menu. Here's what I have: [image]
>
> This has opened a whole new can of worms.
>
> I'm running the `marthoc/deconz` Docker image, which I hadn'r realised is now [...]

Yes, you need to migrate. Your on a very old version of deCONZ.

Smanar, 2023-03-14:
Just for information there is a shared folder in the official deconz docker, you can just put your DDF in the `opt/deCONZ/devices` folder.

BabaIsYou, 2023-03-14:
Just to be sure, is it working each button individualy ?

MrMEScott, 2023-03-14:
> Just for information there is a shared/persistent folder in the official deconz docker, you can just put your DDF in the `opt/deCONZ/devices` folder.
> 
> [deconz-community/deconz-docker#163](https://github.com/deconz-community/deconz-docker/issues/163)

I found it, and that's where I put the DDF.

MrMEScott, 2023-03-14:
> Just to be sure, is it working each button individualy ?

Yes, I have two entities in Home Assistant and I can control them individually.
[image]

although it displays in deconz gui with only the name I gave to the #1 endpoint.
[image]

MrMEScott, 2023-03-15:
I think this can close now?

Mimiix, 2023-03-15:
No, don't close yet. I've linked the PR to this issue and if that one is merged this will auto close :)

closed 2023-04-03 by manup