dresden-elektronik/deconz-rest-plugin #6805: Moes AM43 issues with deConz

Johan-Arens, 2023-03-16:
Hey guys, I have a strange issue with 2 motors, I'd like to pickup your brain. With the motors paired to deConz and exposed to HA, I was able to use the motors.

However, since last week, one motor can't be controlled anymore, it is exposing it's state to HA (open, close, position) but any up/down orders are ignored. I've tried to delete the node, repairing it, factory default, repairing it, rebooting the deconz add-on, restarting the integration in HA, nothing.

Another motor is seen as disconnected in HA, but it is well there in deConz, I see it as active, I can read descriptors etc.

Any idea what could be going on ?

Thanks !

Johan

Johan-Arens, 2023-03-16:
Smanar, 

Thanks ! 
For the issue #1 where the motor is ignoring up and down requests, problem appeared with no changes on the backend. 
Trying to fix the issue, I have updated HA and deConz to the lastest versions. 

This is where issue #2 appeared. 

Disconnected node :  
[image]

I will look at the device json in a subsequent message. 
Thanks !

Johan-Arens, 2023-03-16:
For the disconnected node : 
{
    "capabilities": {
        "alerts": [
            "none",
            "select",
            "lselect"
        ]
    },
    "config": {
        "groups": [
            "0"
        ]
    },
    "etag": "5bd74bd618041854d010e921b56c785d",
    "hascolor": false,
    "lastannounced": "2023-03-15T23:00:39Z",
    "lastseen": "2023-03-16T15:11Z",
    "manufacturername": "_TZE200_zah67ekd",
    "modelid": "TS0601",
    "name": "Smart plug 24",
    "productid": "Tuya_COVD AM43-0.45-40",
    "state": {
        "alert": "none",
        "bri": 0,
        "lift": 0,
        "on": false,
        "open": true,
        "reachable": true
    },
    "swversion": null,
    "type": "Window covering device",
    "uniqueid": "b4:e3:f9:ff:fe:3c:40:c4-01"
}

Smanar, 2023-03-16:
For this one, it's strange where you can see it disconnected ? on deconz itself ?
Because it was see recently "`lastseen": "2023-03-16T15:11Z",`
Ans still visible on the API `"reachable": true`

Smanar, 2023-03-16:
I realy have no clue where this message is from.
If all look fine in the API (mean all type are "Window covering device" and all device have "lift" and "open" can you make a try directly with a REST client, to by-pass HA ?

https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

To set a position you need to send a PUT request to `http://IP:PORT/api/KEY/lights/24/state` with the body `{"lift":50}`

Johan-Arens, 2023-03-16:
I'm having hard time to work this out with the API key. I'm trying to make this work from Node-Red, I've tried to use the key for HA, I only get 403 messages in return. I will try to generate this from HA though.

Smanar, 2023-03-16:
Ha yes, using node red is a good method too.
Can use the same api key for all third app, 403 error to get the api key ? it's not because you forget to "authenticate" ? else can generate the key on the same host (no need autenticate, but if you use docker, not easier)

Johan-Arens, 2023-03-16:
Yes, I'm with Docker. I've tried to generate a new key but I got the same error, using way they said in the documentation.

Johan-Arens, 2023-03-16:
OK I figured out what was going on, I did generate a new key (hoping that won't erase the existing key). 
Smart plug 24 doesn't work, 28 is working though. So the method works, something is going on with 24.

Mimiix, 2023-03-16:
Hi

As this is not a bug report as described in #5113 i am closing this issue.

Please open an issue on the forums.

Kind regards,

closed 2023-03-16 by Mimiix