dresden-elektronik/deconz-rest-plugin #6806: BECA TRV model BRT-100 doesn't work

guiumb, 2023-03-16:
Dear All,
I bought 2 new identical white BECA TRV model BRT-100 (https://zigbee.blakadder.com/Beca_BRT-100.html) and I connect them with  DeConz in HomeAssistant.
1 device works well: the description report Home Automation Smart Plug but DeConz set device name automatically as Themostat 74 and create climate entity in HA.
Other device is paired with DeConz but is not recognized as TRV and I can not find any new entity in HA.

## Steps to reproduce the behavior
I have connected 2 devices multiple times with DeConz restart and reset to default the hardware but only 1 devices is working.

## Expected behavior
The 2 devices have the same model and manufactured but only 1 works

## Screenshots
working TRV:
[image]
[image]

not working TRV:
[image]
[image]

## Environment
DeConz firmware 26720700
DeConz Version: 2.20.01 / 19/9/2022
 - Running method: Home Assistent deCONZ Add-on
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

Smanar, 2023-03-16:
Hello, but they have exactly same model id and manufacture name ?
Can you check the manufacture number ? visible in node information.

guiumb, 2023-03-16:
It seems that related fields is not filled correctly !
working device:
[image]

not working device:
[image]

It is strange... I have bought 2 devices at the same shop in one transaction. Should be a problem on the device firmware?

Many thanks for your help.

Smanar, 2023-03-17:
>It seems that related fields is not filled correctly !

Just need to press the "read" button on the basic cluster, but I already have the needed information.
They are "2 stuffs" to make this device reconized.

```
                    else if (node->nodeDescriptor().manufacturerCode() == VENDOR_EMBER &&
                             (manufacturer.endsWith(QLatin1String("oclfnxz")) ||
                              manufacturer.endsWith(QLatin1String("6wax7g0")) ||
                              manufacturer.endsWith(QLatin1String("88teujp"))))
                    {
                        fpThermostatSensor.inClusters.push_back(TUYA_CLUSTER_ID);
                    }
```

This part work only for manufacture number = 0x1002, and 

```
                case TUYA_CLUSTER_ID:
                {
                    if (manufacturer.endsWith(QLatin1String("kud7u2l")) ||
                        manufacturer.endsWith(QLatin1String("GbxAXL2")) ||
                        manufacturer.endsWith(QLatin1String("eaxp72v")) ||
                        manufacturer.endsWith(QLatin1String("fvq6avy")) ||
                        manufacturer.endsWith(QLatin1String("uhszj9s")) ||
                        manufacturer.endsWith(QLatin1String("oclfnxz")) ||
                        manufacturer.endsWith(QLatin1String("w7cahqs")) ||
                        manufacturer.endsWith(QLatin1String("wdxldoj")) ||
                        manufacturer.endsWith(QLatin1String("hn3negr")) ||
                        manufacturer.endsWith(QLatin1String("6wax7g0")) ||
                        manufacturer.endsWith(QLatin1String("88teujp")))
                    {
                        fpThermostatSensor.inClusters.push_back(TUYA_CLUSTER_ID);
                    }
```

This one can work after the tuya cluster is detected.
So with playing with the API it's possible, first never delete the device, set phoscon in permit join (add new light) and read all attributes in the basic cluster (Cluster 0x0000, button "read") and the descriptor (the second circle on the node title)
If it work the node name wil be something better than 0x78F3.

If it don't success, can use a DDF, to test

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_b6wax7g0",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "Generic TRV",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "meta": {
        "values": {
          "config/mode": {"auto": 0, "heat": 1, "off": 2}
        }
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 14, "eval": "Item.val = Attr.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/windowopen_set",
          "parse": {"fn": "tuya", "dpid": 8, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 8, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/heatsetpoint",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = Attr.val * 50;"},
          "write": {"fn": "tuya", "dpid": 2, "dt": "0x2b", "eval": "Item.val / 50;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "config/locked",
          "parse": {"fn": "tuya", "dpid": 13, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 13, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/mode",
          "values": [
              ["auto", 0], ["heat", 1], ["off", 2]
          ],
          "parse": {"fn": "tuya", "dpid": 2, "eval": "if (Attr.val == 0) { Item.val = 'auto' } else if (Attr.val == 1) { Item.val = 'heat' } else { Item.val = 'off' }"},
          "write": {"fn": "tuya", "dpid": 2, "dt": "0x30", "eval": "if (Item.val == 'auto') { 0 } else if (Item.val == 'heat') { 1 } else { 2 }"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "config/preset",
          "parse": {"fn": "tuya", "dpid": 4, "script": "tuya_trv_preset.js"},
          "write": {"fn": "tuya", "dpid": 4, "dt": "0x30", "script": "tuya_trv_preset_set.js"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 3, "eval": "Item.val = Attr.val * 10;"},
          "read": {"fn": "none"}
        }
      ]
    }
  ]
}
```

If I m right only mode and preset will have issue with it.

guiumb, 2023-03-17:
I have done procedure with read function without success. Tomorrow I will try with custom DDF and I will provide you a feedback.

Thks

Smanar, 2023-03-18:
Give me the value you have and the one you need to have to correct convertion value (can need *10 or /10 for exemple)

guiumb, 2023-03-18:
Ok. I am out of home for week-end.
I will test it as soon as come back and I will keep you informed.

guiumb, 2023-03-19:
I creat a custom file named generic_TRV.json and I put it into custom folder on docker (/data/.local/share/dresden-elektronik/deCONZ/devices/) but when I try to load the file with DDF editor (in VNC) an error was display: fail to open file.
I have also checked JSON sintax on line but no issue was found.
If I try to copy some JSON file from device/tuya folder into custom folder I am able to open them with DDF editor.
Should be config file faulty?

Smanar, 2023-03-20:
You are using docker ?
On official docker, need to put the DDF in the /opt/deCONZ/devices folder
On HA https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839 `/data/.local/share/dresden-elektronik/deCONZ/devices/`

Smanar, 2023-03-21:
It's like there is a typo in the file, but have tested it, and the synthax is good

But how have you copied the code in the file, as you can't use copy paster on HA ?

guiumb, 2023-03-22:
I have created a new file in a folder that you suggest using nano editor: after that I have copyed and attached file from this post.
No way to load DDF.

Smanar, 2023-03-23:
>No way to load DDF

It need to be loaed automaticaly. Just select the node in deconz , then choose "Edit DDF" and it need to open the good one, else there is a problem on the DDf file.

And I don't see what can be bad in the Json file, there is no more explanation in deconz log when trying to load the DDF ? (with flag "info" "info_l2" , "error" and "error_l2" ?

guiumb, 2023-03-25:
I will try to collect some extra infos with debug option and I will forward.

guiumb, 2023-03-26:
I can not see any evidence related on TRV into logs. 
Should I open a request in pull-request form in order to support this device "natively" with screenshot attached?

Mimiix, 2023-03-27:
> I can not see any evidence related about TRV into logs. Should I open a request in pull-request form in order to support this device "natively" with screenshot attached?

What would it help?

Pull request is for submitting a code change. Not for reporting bugs.

Mimiix, 2023-03-27:
In this case, its rather user error than a bug.

@Smanar is this a device request ?

Smanar, 2023-03-28:
A Bug, an issue, or an improvement, IDK.
He have several model of a device (already supported) , somes are working other can't be included.
The last one have different manufacture number, so the legacy code don't work, and there is problem to load the DDF to test with them.

Mimiix, 2023-03-28:
> A Bug, an issue, or an improvement, IDK. He have several model of a device (already supported) , somes are working lasts ones can't be included. The last one have different manufacture number, so the legacy code don't work, and there is problem to load the DDF to test with them.

Yes, but the issue with the DDF not loading probably has to do with the incorrect folder.

Smanar, 2023-03-28:
IDK, with the HA docker, just need to put the file in `/data/.local/share/dresden-elektronik/deCONZ/devices/` and he have do that.
But something I don't understand is why he can't load the DDF manualy, it's like the DDF was corrupted, but it's just a text file ....

Mimiix, 2023-03-28:
In his screenshot, there should be more files if it was the correct place. Including button maps

guiumb, 2023-03-30:
I am using /data/.local/share/dresden-elektronik/deCONZ/devices/ folder as Smanar suggested.
Anyway I decided to send the device back to the seller and I bought a Hama TRV (https://zigbee.blakadder.com/Hama_00176592.html).

guiumb, 2023-03-30:
Ok. I will keep you updated.

guiumb, 2023-03-31:
I am working with new Hama TRV (https://zigbee.blakadder.com/Hama_00176592.html):
I have deleted Beca TRV device and I have restarted deCONZ. Pairing with new Hama was no success; so I have saved DDF file from original good Beca TRV that is currently working (Thermostat 74) and I have applied this file to the new Hama changing manufacturer name and status to Gold:
[image]
After that deConz change the node name to Thermostat 75 (using suggestet procedure with read button and hot reload menu):
[image]

New device was exported to HA:
[image]

this is node info:
[image]

The procedure was fine but TRV device doesn't have any cluster about thermostat, On/OFF and doen't report any data to API: also in working beca (thermostat 74) all data was extracet from cluster 201 and not from tuya.

[image]

I can not understand if I am wrong in some procedure !

Smanar, 2023-03-31:
>I can not understand if I am wrong in some procedure !

Not your fault. Your procedure is fine, ...., for all stuff other than tuya ^^.
Can you replace your DDF by this one 

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_h4cgnbzg",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "Saswell TRVs",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "meta": {
        "values": {
          "config/mode": {"off": 0, "heat": 1}
        }
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 21, "eval": "Item.val = Attr.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/heatsetpoint",
          "parse": {"fn": "tuya", "dpid": 103, "eval": "Item.val = Attr.val * 10;"},
          "write": {"fn": "tuya", "dpid": 103, "dt": "0x2b", "eval": "Item.val / 10;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "config/locked",
          "parse": {"fn": "tuya", "dpid": 40, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 40, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/windowopen_set",
          "parse": {"fn": "tuya", "dpid": 8, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 8, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/mode",
          "values": [
              ["off", 0], ["heat", 1]
          ],
          "parse": {"fn": "tuya", "dpid": 101, "eval": "if (Attr.val == 0) { Item.val = 'off' } else { Item.val = 'heat' }"},
          "write": {"fn": "tuya", "dpid": 101, "dt": "0x10", "eval": "if (Item.val == 'off') { false } else { true }"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 102, "eval": "Item.val = Attr.val * 10;"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/valve",
          "parse": {"fn": "tuya", "dpid": 104, "eval": "Item.val = Attr.val > 5;"},
          "read": {"fn": "none"}
        }
      ]
    }
  ]
}
```

Or just create it.
But delete the device before use the DDF to not have conflict in the database (the one that will be created by this DDF will be different the one you already have).

You can have convertion issue, for exemple temperature *10 or /10, just tell me the value you have and the one you have to correct the DDF.

guiumb, 2023-04-01:
Using DDF file provided by Smanar TRV start working !
I can see some differences in cluster between BECA (Therm. 74) and new Hama (Term 75) but basic function (set point, open/close and temperature) work correctly. Aldo factor *10 conversion for temperure is correct.
[image]

I add config/battery in DDF because battery value was not present in API:

[image]

I use DP 14 as suggested in old DDF but value is still remain at 0

[image]

Do you have any suggestion? @Mimiix I have to open a request for new device with screenshots attached?

Mimiix, 2023-04-01:
> Do you have any suggestion? @Mimiix I have to open a request for new device with screenshots attached?

Yes.

Smanar, 2023-04-02:
This device don't have battery level but a batterylow warning.

>low battery value is missing and is not reported to API

On the future issue can you post the actual device sjon you have ?

guiumb, 2023-04-03:
@Mimiix I will do that.
@Smanar After some hours low battery level flag was reported to API without any changes on DDF, so climate entity on HA works perfectly.
I have used DDF file that you provide by copy and paste with no changes and DDF editor was able to manage it.
Many thanks for your help.

Mimiix, 2023-04-04:
Hi ,please don't close. Let's wait a PR is made and implemented so other users can benefit as well:)

closed 2023-05-04 by github-actions[bot]