dresden-elektronik/deconz-rest-plugin #6807: Tuya / TONGOU TO-Q-SV1-ZT

GabeHH78, 2023-03-17:
## Device

- Product name: Tuya / TONGOU
- Manufacturer: unknown
- Model identifier: unknown
- Device type : Rail

[image]

[nodeinfo](https://user-images.githubusercontent.com/78250680/225892548-c23544a0-402a-4e83-8688-06f5caf24259.jpg)

Mimiix, 2023-03-17:
Hi,

please add all screenshots.

Mimiix, 2023-03-17:
Please read the template when you created the issue. It is really clear on what is required.

Additionally, edit your opening post.

GabeHH78, 2023-03-17:
I put now everything in, hope it is now ok? Or is anything missing?

Mimiix, 2023-03-17:
Hi,

The attributes are not populated. You need to make sure that they are. To do this, click "read" on the cluster window and see if things change. If they don't, check if the device is not sleeping.

GabeHH78, 2023-03-19:
Hi,

I put new scrrenshots in. I hope it is now ok.

I have the possibility in Deconz to siwtch it on and off, but I couldnt see it in the phoscon app.

Thanks for helping

Mimiix, 2023-03-19:
Basic cluster is still missing

GabeHH78, 2023-03-20:
Hi,

I put now Basic inside

Hope now ok

rgds

GabeHH78, 2023-03-20:
I put a new screenshot in. Is it better?

Mimiix, 2023-03-20:
looks good :)! Thank you

GabeHH78, 2023-03-20:
great Thank you ;)

BabaIsYou, 2023-03-30:
I suppose that this is the model without metering, then could give a try to this DDF :

```
{
	"schema": "devcap1.schema.json",
	"manufacturername": "_TZ3000_gzvniqjb",
	"modelid": "TS0011",
	"product": "TO-Q-SY1-ZT",
	"sleeper": true,
	"status": "Silver",
	"subdevices": [{
		"type": "$TYPE_ON_OFF_OUTPUT",
		"restapi": "/lights",
		"uuid": [
			"$address.ext",
			"0x01"
		],
		"items": [{
				"name": "attr/id"
			},
			{
				"name": "attr/lastannounced"
			},
			{
				"name": "attr/lastseen"
			},
			{
				"name": "attr/manufacturername"
			},
			{
				"name": "attr/modelid"
			},
			{
				"name": "attr/name"
			},
			{
				"name": "attr/swversion"
			},
			{
				"name": "attr/type"
			},
			{
				"name": "attr/uniqueid"
			},
			{
				"name": "state/on"
			},
			{
				"name": "state/reachable"
			}
		]
	}]
}
```

Don't know if bindings are nedded.

BabaIsYou, 2023-03-31:
BTW could add  binding in the DDF : 
```
{
	"schema": "devcap1.schema.json",
	"manufacturername": "_TZ3000_gzvniqjb",
	"modelid": "TS0011",
	"product": "TO-Q-SY1-ZT",
	"sleeper": true,
	"status": "Silver",
	"subdevices": [{
		"type": "$TYPE_ON_OFF_OUTPUT",
		"restapi": "/lights",
		"uuid": [
			"$address.ext",
			"0x01"
		],
		"items": [{
				"name": "attr/id"
			},
			{
				"name": "attr/lastannounced"
			},
			{
				"name": "attr/lastseen"
			},
			{
				"name": "attr/manufacturername"
			},
			{
				"name": "attr/modelid"
			},
			{
				"name": "attr/name"
			},
			{
				"name": "attr/swversion",
				"parse": {
					"fn": "zcl",
					"ep": 1,
					"cl": "0x0000",
					"at": "0x0001",
					"script": "tuya_swversion.js"
				},
				"read": {
					"fn": "zcl",
					"ep": 1,
					"cl": "0x0000",
					"at": "0x0001"
				}
			},
			{
				"name": "attr/type"
			},
			{
				"name": "attr/uniqueid"
			},
			{
				"name": "state/on"
			},
			{
				"name": "state/reachable"
			}
		]
	}],
	"bindings": [{
		"bind": "unicast",
		"src.ep": 1,
		"cl": "0x0006",
		"report": [{
			"at": "0x0000",
			"dt": "0x10",
			"min": 1,
			"max": 300
		}]
	}]
}
```

GabeHH78, 2023-04-01:
Great now everything work fine for me.

Many thanks for your helps!!!

ValleBL, 2023-04-10:
Would this include TO-Q-SY1-JZT?
That’s the version with power metering

PR #6842 (2023-04-15): Add DDF for  Tuya / TONGOU TO-Q-SV1-ZT
Add DDF for  Tuya / TONGOU TO-Q-SV1-ZT din rail switch without metering.

Fix #6807
closed 2023-04-15 by manup