dresden-elektronik/deconz-rest-plugin #6816: Aqara E1 Thermostat does not take values from HomeAssistant via deconz

Wolkenstuermer, 2023-03-22:
## Describe the bug
I was able to onboard Aqara Thermostat E1 on Monday via Homeassistant running on a docker container and phoscon  running also on a docker container. Every other Aqara sensors work as expected. 

However, Temperature Values i set via Homeassisstant are not pushed to the Thermostat. It's not broken completely, as settings i do on the valve are shown in HA, and if i set it offline via HA, the valve goes off.

I am not deep in DDF Files, but if someone tells me, what to search for, i will. (I didn't manage to set the temperature directly in Mesh View)

I was very happy, that in this version, this Thermostat is now supported.

## Steps to reproduce the behavior
1. I am on my Homeassistant. have the integration up and running
2. Set the temperature via HA tile to ie 27 Celsius
3. On the valve nothing happens, after some seconds/minutes, the shown value of the valve in the tiles goes back to what was set manually.
4. Green lights on Valve for Connection is working.

Some communication is working:
1.  If i set the valve on the HA Tile to "Off" it does as expected.
  
## Expected behavior
Set the Temperature at the valve.

 - Host system: HP MicroServer Gen8 with Ubuntu and Docker
 - Running method: ( Ubuntu / Home Assistent Docker COntainer, deCONZ Docker container )
 - Firmware version: 26720700
 - deCONZ version: 2.21.01 / 19.9.2022
 - Device: ConBee II 
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system? If so: Printer

## deCONZ Logs
attached; if anything else is needed, i will provide.

## Additional context

[debug_deconz_20230323.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/11039155/debug_deconz_20230323.txt)

Mimiix, 2023-03-22:
Hi,

This repo does not provide any support or contains any code regarding Home assistant. However, there could still be an issue. 

For this, i'd like you to test if it works using the deCONZ RestAPI (this is how HA talks to deCONZ). 

Documentation can be found here: https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

Additionally: https://dresden-elektronik.github.io/deconz-rest-doc/devices/xiaomi/xiaomi_lumi.airrtc.agl001/

Mimiix, 2023-03-22:
Do values under 27 work?

Wolkenstuermer, 2023-03-22:
"14:17:57:94": {
        "config": {
            "battery": 80,
            "heatsetpoint": 2500,
            "locked": false,
            "mode": "heat",
            "offset": 0,
            "on": true,
            "preset": "manual",
            "reachable": true
        },
        "e": "changed",
        "id": "40",
        "r": "sensors",
        "t": "event",
        "uniqueid": "54:ef:44:10:00:70:5e:19-01-0201"
    },
    "14:18:18:831": {
        "e": "changed",
        "id": "40",
        "r": "sensors",
        "state": {
            "lastupdated": "2023-03-22T13:18:19.343",
            "on": true,
            "temperature": 2480,
            "windowopen": "false"
        },
        "t": "event",
        "uniqueid": "54:ef:44:10:00:70:5e:19-01-0201"
    }, 

No, short time after that, it fails back...
        "e": "changed",
        "id": "40",
        "r": "sensors",
        "t": "event",
        "uniqueid": "54:ef:44:10:00:70:5e:19-01-0201"
    },
    "14:17:43:148": {
        "config": {
            "battery": 80,
            "heatsetpoint": 2270,
            "locked": false,
            "mode": "heat",
            "offset": 0,
            "on": true,
            "preset": "manual",
            "reachable": true
        },
        "e": "changed",
        "id": "40",
        "r": "sensors",
        "t": "event",
        "uniqueid": "54:ef:44:10:00:70:5e:19-01-0201"
    },
    "14:17:44:791": {
        "attr": {
            "colorcapabilities": 0,
            "ctmax": 65279,
            "ctmin": 0,
            "id": "4",
            "lastannounced": "2022-11-05T14:35:23Z",
            "lastseen": "2023-03-22T13:17Z",
            "manufacturername": "Philips",
            "modelid": "LTA009",
            "name": "KellerflurDeconz",
            "swversion": "1.93.7",
            "type": "Color temperature light",
            "uniqueid": "00:17:88:01:0b:1e:97:ce-0b"
        },
        "e": "changed",
        "id": "4",
        "r": "lights",
        "t": "event",
        "uniqueid": "00:17:88:01:0b:1e:97:ce-0b"
    },
    "14:17:45:775": {
        "attr": {
            "id": "3",
            "lastannounced": null,
            "lastseen": "2023-03-22T13:17Z",
            "manufacturername": "dresden elektronik",
            "modelid": "ConBee II",
            "name": "Configuration tool 3",
            "swversion": "0x26720700",
            "type": "Configuration tool",
            "uniqueid": "00:21:2e:ff:ff:07:d9:e4-01"
        },
        "e": "changed",
        "id": "3",
        "r": "lights",
        "t": "event",
        "uniqueid": "00:21:2e:ff:ff:07:d9:e4-01"
    },
    "14:17:52:256": {
        "attr": {
            "id": "6",
            "lastannounced": "2022-11-05T14:35:18Z",
            "lastseen": "2023-03-22T13:17Z",
            "manufacturername": "IKEA of Sweden",
            "modelid": "TRADFRI control outlet",
            "name": "On/Off Steckdose Keller",
            "swversion": "2.0.024",
            "type": "On/Off plug-in unit",
            "uniqueid": "b4:e3:f9:ff:fe:80:df:41-01"
        },
        "e": "changed",
        "id": "6",
        "r": "lights",
        "t": "event",
        "uniqueid": "b4:e3:f9:ff:fe:80:df:41-01"
    },
    "14:17:57:94": {
        "config": {
            "battery": 80,
            "heatsetpoint": 2500,
            "locked": false,
            "mode": "heat",
            "offset": 0,
            "on": true,
            "preset": "manual",
            "reachable": true
        },
        "e": "changed",
        "id": "40",
        "r": "sensors",
        "t": "event",
        "uniqueid": "54:ef:44:10:00:70:5e:19-01-0201"
    },
    "14:18:18:831": {
        "e": "changed",
        "id": "40",
        "r": "sensors",
        "state": {
            "lastupdated": "2023-03-22T13:18:19.343",
            "on": true,
            "temperature": 2480,
            "windowopen": "false"
        },
        "t": "event",
        "uniqueid": "54:ef:44:10:00:70:5e:19-01-0201"
    },

Wolkenstuermer, 2023-03-22:
[deconz_events_20230322.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/11040436/deconz_events_20230322.txt)

Mimiix, 2023-03-22:
I change the Title to be more comprehensive in this case as the issue is not with HA.

chrissiboy, 2023-03-26:
I can confirm that behaviour since version 2.21.0 and 2.21.1 too - in version 2.20 the setting of the heatsetpoint works with postman and IP Symcon / DeConz.

Wolkenstuermer, 2023-04-17:
Is there any possibility of downgrading the firmware? Anything i can test or provide?

chrissiboy, 2023-04-17:
Yes, after downgrade on version 2.20 the setting of the heatsetpoint works fine

closed 2023-04-17 by manup

amir-hadi, 2023-04-23:
@SwoopX You mentioned in your PR that you cannot test this with the Aqara / Xiaomi Thermostat. I have multiple of them and I remember it didn't work before downgrading to 2.20.1. Now I tested 2.22.00 and they seem to work. Thank you very much for your effort.

Paeddi359, 2025-01-05:
Hello,
I have 3 Aqara E1 thermostats and none of them accept the heatsetpoint.

I have already tested various deconz versions but unfortunately I had no success...

ConBee II on a Raspberry PI 4B (Raspian)
Firmware: 0x26780700

Phoscon
v2.28.1
v2.22.2
v2.22.0
v2.26.3
v2.20.1

The write entry in the generic DDF is there.

The room temperature is also read back and the setpoint is also included in the Rest API but it is not send to the Radiator

what else can I do?

Thank you