dresden-elektronik/deconz-rest-plugin #6817: V2.21.01 and Ikea Symfonisk controller is not sending events?

carsten-h, 2023-03-22:
## Describe the bug

Hello!

A few days ago I installed deCONZ 2.21.02 on a Pi 4 with Raspberry Pi OS and a Conbee II with Firmware 26780700.
I have two Ikea Symfonisk controllers (E1744) which had been working without any problem with deCONZ together with Home Assistant (which has not been changed the last days.
I also have Ikea Tradfri E1810 and an Aqara Opple 6 button ZigBee controller which are still working fine and they are sending events to Home Assistant.

But the two Ikea Symfonisk controllers are not sending any events to Home Assistant. Nothing can be seen. I can see the battery status in Hime Assistant and that is fine. I also changed one of the batteries and connected one of the Symfonisk controllers again to deCONZ/Phoscon, but nothing changes. No events of the Symfonisk controllers can be seen. Every other controller is sending without a problem.

Is there any known problem with these Symfonisk controllers?

Thank you!

## Steps to reproduce the behavior

Just use the Symfonisk controller and you see that no events are send (just watch in Home Assistant for deconz_event).

## Environment

 - Host system: (Raspberry Pi)
 - Running method: (Raspbian)
 - Firmware version: (26780700)
 - deCONZ version: (2.21.01)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which? M.2 SSD in Argon One M.2 case
 - Client System: Raspberry Pi 4 with Home Assistant and deCONZ Integration

Mimiix, 2023-03-22:
Not showing in EVENTS either. Have been able to reproduce this.

ebaauw, 2023-03-22:
Is that the gen-1 (with dial) or gen-2 (7-button) SYMFONISK controller?

Mimiix, 2023-03-22:
The dial

carsten-h, 2023-03-24:
I looked what has been changed in version 2.21.01 and I found this: [https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6803](url)

Maybe this change "kills" the older Symfonisk controller?

ebaauw, 2023-03-24:
Indeed, the `button_maps.json` from PR #6803 is messed up and breaks support for both the gen-1 and gen-2 SYMFONISK controller.  My bad.

carsten-h, 2023-03-24:
> breaks support for both the gen-1 and gen-2 SYMFONISK controller

Ah, great that you found it!

> My bad.

No problem! Great that you found a fix!

ebaauw, 2023-03-24:
I hope we can include this in the next v2.21.x release.

In the meantime for deCONZ v2.21.1, you could copy the `button_maps.json` from the new PR to `/usr/share/deCONZ/devices`.  After restarting deCONZ, you should once again be receiving button events.

carsten-h, 2023-03-24:
> you should once again be receiving button events

Yes, it was working instantly! Dank u wel!

binford6000, 2023-03-26:
> In the meantime for deCONZ v2.21.1, you could copy the button_maps.json from the new PR to /usr/share/deCONZ/devices. After restarting deCONZ, you should once again be receiving button events.

How can i apply this with docker? Simply putting the file into the volume under 'devices'-folder is not working...
KR, Sebastian

Mimiix, 2023-03-28:
I replaced the button_map.json in the /usr/share/deCONZ/devices folder and now it works as inteded.

closed 2023-03-29 by manup

binford6000, 2023-03-30:
Ok got it. Thx!

carsten-h, 2023-04-01:
> I hope we can include this in the next v2.21.x release.

The change is in 2.21.2 and it works without any problem!

Thank you!

Egglestron, 2023-04-08:
I can't get it to work on the 2.21.2 release, on both 1.0.32 and 1.0.35 firmwares (latest one). Only the 1002 event is received by HA when single pressing the Play button... 
I checked the _button_map.json_ file and it matches what is mentioned in this thread. :/

carsten-h, 2023-04-08:
> Only the 1002 event is received by HA when single pressing the Play button...

Did you tried putting the battery out and in again? That sometimes heals such situations.

Egglestron, 2023-04-08:
> > Only the 1002 event is received by HA when single pressing the Play button...
> 
> Did you tried putting the battery out and in again? That sometimes heals such situations.

I did, as well as repairing the remote multiple times, and rebooting deCONZ.

Mimiix, 2023-04-08:
In your case i recommend opening a topic on the forum as the implementation is fine