dresden-elektronik/deconz-rest-plugin #6818: Aqara WXKG20LM

agramner, 2023-03-22:
## Device

- Product name: Aqara WXKG20LM.
- Manufacturer: LUMI.
- Model identifier: lumi.remote.acn007
- Device type :
  - Remote
  - Switch

[image]

## Screenshots

[image]

[image]

### Basic
[image]

### Identify
[image]

### Power Configuration
[image]

### Multistate Input (Basic)
[image]

### Lumi specific
[image]
[image]
[image]

BabaIsYou, 2023-03-24:
Hi,

Are you sure you clicked on "read" before taking screenshot of Power Configuration cluster ?

agramner, 2023-03-24:
You're right! I updated the screenshot :)

BabaIsYou, 2023-03-24:
Trying to make a @SwoopX of me ... ;-) adapting https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5862#issuecomment-1136459355

Could you give a try to this DDF ? But button mapping should also have to be made after that.

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "$MF_LUMI",
  "modelid": "lumi.remote.acn007",
  "matchexpr": "R.endpoints.length === 1",
  "vendor": "Xiaomi",
  "product": "Aqara Single Wall Button E1 WXKG20LM",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0000",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0012",
          "0xFCC0"
        ],
        "out": [
          "0x0006",
          "0x0019"
        ]
      },
      "meta": {"group.endpoints": [1] },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid",
          "awake": true
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x00f7",
            "ep": 1,
            "fn": "xiaomi:special",
            "idx": "0x01",
            "script": "xiaomi_battery.js"
          }
        },
        {
          "name": "config/clickmode",
          "refresh.interval": 3000,
          "parse": {
            "at": "0x0125",
            "cl": "0xfcc0",
            "ep": 1,
            "eval": "if (Attr.val == 1) { Item.val = 'highspeed' } else if (Attr.val == 2) { Item.val = 'multiclick' } else { Item.val = 'unknown' }",
            "fn": "zcl",
            "mf": "0x115f"
          },
          "read": {
            "at": "0x0125",
            "cl": "0xfcc0",
            "ep": 1,
            "fn": "zcl",
            "mf": "0x115f"
          },
          "write": {
            "at": "0x0125",
            "cl": "0xfcc0",
            "dt": "0x20",
            "ep": 1,
            "eval": "if (Item.val == 'highspeed') { 1 } else if (Item.val == 'multiclick') { 2 }",
            "fn": "zcl",
            "mf": "0x115f"
          },
          "values": [
            ["\"highspeed\"", "Buttons emit only single click events, but fast"],
            ["\"multiclick\"", "Emits single, double and treble presses"] 
          ],
          "default": "multiclick"
        },
        {
          "name": "config/devicemode",
          "refresh.interval": 3000,
          "parse": {
            "at": "0x0009",
            "cl": "0xfcc0",
            "ep": 1,
            "eval": "if (Attr.val == 1) { Item.val = 'compatibility' } else if (Attr.val == 2) { Item.val = 'zigbee' } else { Item.val = 'unknown' }",
            "fn": "zcl",
            "mf": "0x115f"
          },
          "read": {
            "at": "0x0009",
            "cl": "0xfcc0",
            "ep": 1,
            "fn": "zcl",
            "mf": "0x115f"
          },
          "write": {
            "at": "0x0009",
            "cl": "0xfcc0",
            "dt": "0x20",
            "ep": 1,
            "eval": "if (Item.val == 'compatibility') { 1 } else if (Item.val == 'zigbee') { 2 }",
            "fn": "zcl",
            "mf": "0x115f"
          },
          "values": [
            ["\"compatibility\"", "Default mode for Xiaomi devices"],
            ["\"zigbee\"", "Closer to zigbee standard"] 
          ],
          "default": "compatibility"
        },
        {
          "name": "config/group",
          "default": "auto"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ]
}
```

agramner, 2023-03-24:
Nice, will try later tonight. I guess there's no way to copy paste it into Deconz UI? Need to find a way to ssh into Home assistant OS otherwise

agramner, 2023-03-24:
Great, I tested it and it got a name in deconz UI and battry entity appeared in Home Assistant.

[image]
[image]

agramner, 2023-03-24:
I checked out `button_maps.json` and guess that something need to be added there?
If it is of any help the `Preset value` in the `Multistate input` cluster change value like this:
- Long hold - `0`
- Single click - `1`
- Double click - `2`

BabaIsYou, 2023-03-24:
May be you could see what is in button_maps.json for lumi.remote.acn003 to get inspired … if there is something for it

BabaIsYou, 2023-03-25:
Seems could just try to add modelid to the part `xiaomiSwitchB186acn01Map` in `button_maps.json` because seems to be the same mapping ;-)

agramner, 2023-03-25:
Yes tried that yesterday and it worked 😊. Battery status doesn't seem to work though, will look in to that later

BabaIsYou, 2023-03-25:
Did you try with my last proposition for battery ?

agramner, 2023-03-25:
Oh didn't see your comment about battery. Will try that tonight 👍

SwoopX, 2023-03-25:
Battery is not working as the required .js file is not in the same location as the DDF. Either you copy it from the current location on your system or create a new one with the content from the repo. Please stick with the Xiaomi special reporting here.

Additionally, the swversion item can be replaced with this, again the respective .js file needs to be available:

```
        {
          "name": "attr/swversion",
          "awake": true,
          "parse": {
            "at": "0x00f7",
            "ep": 1,
            "fn": "xiaomi:special",
            "idx": "0x08",
            "script": "xiaomi_swversion.js"
          },
          "read": {
            "fn": "none"
          }
        },
```

Other than that, the sensor uuid is currently wrong, as it uses the on/off cluster, but should use multistate input.

Regarding the groups, I somewhat doubt that the switch correctly supports it. That would imply it also sends something, which can be seen in the debug log and is usable.

agramner, 2023-03-25:
Thanks! It seem to work fine now. Changed to use xaomi:special for battery and swversion. And changed the uuid as suggested (not sure how that matters).

Also, clickmode doesn't seem to be supported. The attribute is greyed out. Should i remove config/clickmode?
[image]

And what about devicemode? It's 0. Should it be removed?

BabaIsYou, 2023-03-27:
> Other than that, the sensor uuid is currently wrong, as it uses the on/off cluster, but should use multistate input.

Thanks @SwoopX, but why should this zone use multistate rather thaan on/off like the one in #5862 ?

SwoopX, 2023-03-27:
Because that was a 1:1 copy from the H1 DDF, having just exchanged the model ID. And the H1 makes use of the on/off cluster.

SwoopX, 2023-03-27:
> Thanks! It seem to work fine now. Changed to use xaomi:special for battery and swversion. And changed the uuid as suggested (not sure how that matters).
>
> Also, clickmode doesn't seem to be [...]

Yes, clickmode can be removed. I'd keep devicemode for the time being, it should set itself to 1 (compatibility) after some time.

agramner, 2023-03-28:
Ok great! I'll create a PR soon

PR #6828 (2023-04-13): DDF for Aqara mini switch WXKG20LM
Support for Aqara mini switch WXKG20LM

Fixes #6818
closed 2023-04-13 by manup