dresden-elektronik/deconz-rest-plugin #6819: Tuya Motion Sensor Model ZG-204Z

nelbs, 2023-03-22:
## Device

Recently a PR for the device ZG-204ZL was merged #6458. This device request is about the ZG-204Z which is basically the same device but without a luminance sensor. https://de.aliexpress.com/item/1005004135075743.html?gatewayAdapt=glo2deu

- Product name: Tuya Motion Sensor Model ZG-204Z
- Manufacturer: _TZE200_3towulqd
- Model identifier: TS0601
- Device type : Please remove all unrelated device types. 
  - Sensor

## Screenshots
### endpoint menu
[image]

### node info
[image]

### Basic
[image]

### power configuration
[image]

### IAS zone
[image]

### OTAU
[image]

BabaIsYou, 2023-03-27:
Well, this device has he same combination Manufacturer/ModelId than the other one  https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6458 (?!) then the DDF should be picked up and do the trick.

Mimiix, 2023-03-27:
So it's a duplicate ?

BabaIsYou, 2023-03-27:
> So it's a duplicate ?

Not really sure, seems to be perhaps two different devices (one with luminance and second not) but with the same combination Manufacturer/ModelId if I believe the screenshots. But in that case I don't understand why the actual DDF was not picked up.

Depending of answer we get may be it could be a duplicate, or a strange affair ;-)

BabaIsYou, 2023-03-28:
> So it's a duplicate ?

Then I think it is ! ;-)

Mimiix, 2023-03-28:
Okido, thanks guys :)