dresden-elektronik/deconz-rest-plugin #6823: Tuya(Meian) Flooding detector SW02-ZB

alex-jung, 2023-03-24:
## Device

- Product name: Smart Flooding Detector SW02-ZB
- Manufacturer: _TZ3000_kyb656no
- Model identifier: TS0207
- Device type :
  - Sensor

## Screenshots
### Endpoint
[image]

### Node info
[image]

### Basic
[image]

### Identify
[image]

## Power configuration
[image]

### IAS Zone
[image]

### OTAU
[image]

BabaIsYou, 2023-03-24:
Can you give a try to this DDF ?

{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_kyb656no",
  "modelid": "TS0207",
  "vendor": "Meian",
  "product": "Flooding detector SW02-ZB",
  "sleeper": true,
  "status": "Silver",
  "subdevices": [
    {
      "type": "$TYPE_WATER_LEAK_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0003",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid",
          "awake": true
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2",
            "fn": "zcl"
          }
        },
        {
          "name": "config/checkin",
          "awake": true
        },
        {
          "name": "config/enrolled",
          "public": true
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/water",
          "awake": true
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 3600,
          "max": 14400,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500"
    }
  ]
}

alex-jung, 2023-03-27:
Hi,

Unfortunately no changes :( But I'm not 100% if I'm doing the update correctly. Please let me know if I'm doing it wrong.

Following steps I did performed:
1. Store provided content in a JSON file inside the deCONZ docker image
2. Open "deCONZ" Webview
3. Right click on "problematic" node
4. Edit DDF
5. File -> Open created JSON file
6. Open Phoscon
7. Try to add a new sensor

[image]

BabaIsYou, 2023-03-27:
DDF has to be stored in a stecific folder (may be could find hints here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6798#issuecomment-1466483973). Then just rejoining the device may be sufficient.

alex-jung, 2023-03-29:
Thank you for tips :) After I enabled Silver DDF level, sensor was updated in deCONZ:
[image]

in phoscon works sensor as expected too.

But in Home Assistant I see only Battery Entity:
[image]

Do you know where I can address this issue?

BabaIsYou, 2023-03-29:
Well done ! I'm not familiar with HA, but do you have some errors in log, in Deconz or HA part about this device ?

BabaIsYou, 2023-03-29:
@alex-jung can you see the sensor thru Phoscon or REST-API ?

alex-jung, 2023-04-02:
Hi,

Sorry for late answer, my HA was broken for some reason, I had to reinstall it.
Now after I installed Home Assistant OS, the sensor work as expected:
[image]

Thank you a lot for providing a solution and have nice Easter :)

BR
Alex

PR #6833 (2023-04-17): DDF for various TS0207 Tuya based water leak sensors
Fix #6823
closed 2023-04-17 by manup