dresden-elektronik/deconz-rest-plugin #6843: GUI bug: non-consecutive values for enum command parameters

ebaauw, 2023-04-01:
## Describe the bug

The GUI doesn't handle command parameters of type enum with non-consecutive values.

## Steps to reproduce the behavior

Try and issue a _Move Color Temperature_ with _Move Mode_ _Down_ from the GUI.  The GUI actually sends value 2 as parameter, since _Down_ is the third item on the value list.  This results in an _Invalid Value_ error.

## Expected behavior

The GUI should send the defined value from `general.xml`, 3.

## Screenshots

The command definition from `general.xml`:
```
        <command id="0x4b" dir="recv" name="Move Color Temperature" required="m">
          <payload>
            <attribute id="0x0000" type="enum8" name="Move Mode" required="m">
              <value name="Stop" value="0x00"></value>
              <value name="Up" value="0x01"></value>
              <value name="Down" value="0x03"></value>
            </attribute>
            <attribute id="0x0001" type="u16" name="Rate" required="m">
              <description>Steps per second.</description>
            </attribute>
            <attribute id="0x0002" type="u16" name="Color Temperature Min Mireds" required="m">
              <description>Specifies a lower bound on the color temperature for the current move operation.</description>
            </attribute>
            <attribute id="0x0003" type="u16" name="Color Temperature Max Mireds" required="m">
              <description>Specifies a upper bound on the color temperature for the current move operation.</description>
            </attribute>
          </payload>
        </command>
```

## Environment

 - Host system: All
 - Running method: All
 - Firmware version: All
 - deCONZ version: All, including v2.21.2
 - Device: All
 - Do you use an USB extension cable: n/a
 - Is there any other USB or serial devices connected to the host system? If so: Which? n/a

## deCONZ Logs

n/a

## Additional context

See Discord, #core-issues channel.

This used to be a bug for enum attributes as well, but that has been fixed (see e.g. _Startup OnOff_).

The same workaround as for attributes still works, e.g. for _Step Color Temperature_:
```
        <command id="0x4c" dir="recv" name="Step Color Temperature" required="m">
          <payload>
            <attribute id="0x0000" type="enum8" name="Step Mode" required="m">
              <value name="GUI Bug" value="0x00"></value>
              <value name="Up" value="0x01"></value>
              <value name="GUI Bug" value="0x02"></value>
              <value name="Down" value="0x03"></value>
            </attribute>
            <attribute id="0x0001" type="u16" name="Step Size" required="m"></attribute>
            <attribute id="0x0002" type="u16" name="Transition Time" required="m">
              <description>The transition time in 1/10 seconds.</description>
            </attribute>
            <attribute id="0x0003" type="u16" name="Color Temperature Min Mireds" required="m">
              <description>Specifies a lower bound on the color temperature for the current step operation.</description>
            </attribute>
            <attribute id="0x0004" type="u16" name="Color Temperature Max Mireds" required="m">
              <description>Specifies a upper bound on the color temperature for the current step operation.</description>
            </attribute>
          </payload>
        </command>
```

manup, 2023-04-17:
Found the issue in the code, it's now fixed and will be available in v2.22.0-beta.

manup, 2023-04-19:
Closing for now as fixed in v2.22.0-beta.

closed 2023-04-19 by manup