dresden-elektronik/deconz-rest-plugin #6845: Philips Outdoor Sensor pairing uncomplete after OTA

olicooper, 2023-04-01:
## Describe the bug

I performed an OTA update on the sensor and after the update I noticed deconz was flashing green next to the device every few seconds. I found a related GitHub issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1569 which states it is to do with pairing not being complete. I looked for logs related to this (see below).

## Steps to reproduce the behavior

Perform and update using this ota file: https://otau.meethue.com/storage/ZGB_100B_010D/bd3f218f-190b-4498-b6b3-69aea563fd9d/Sensor-ATmega_6.1.1.27575_0012.sbl-ota

This should update the sensor:
- **`Date Code`** `20180828` to `20190219`
- **`SW Build ID`** `6.1.0.25261` to `6.1.1.27575`

## Expected behavior

Pairing should already be complete but if it has to pair again after the update then this should work without intervention.

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26720700
 - deCONZ version: 2.21.01
 - Device: ConBee II
 - Do you use an USB extension cable: yes

## deCONZ Logs

`16:11:37:489 No presence sensor found for 0x00178801XXXXXXXX, endpoint: 0x02`

ebaauw, 2023-04-01:
That's not what the green light means.  See https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-for-Dummies#node-status-led.

olicooper, 2023-04-01:
Okay thank you for the quick response and for clarifying that.. I tired looking in the [deconz handbook PDF](https://www.dresden-elektronik.com/wireless/software/deconz.html?file=files/dresden-elektronik/content/downloads/dokumente/funktechnik/deCONZ-BHB-en.pdf) but the status LEDs are not mentioned in there.

Having left the device for a while the presence detection seems to have come back to life!

I am apprehensive to update my other sensors due to the log entries: `No presence sensor found for 0x00178801XXXXXXXX, endpoint: 0x02`... Should I keep this issue open to address a possible DDF problem?

ebaauw, 2023-04-01:
> There seems to be a lot of activity happening on the device since the update despite being on battery power.

The Hue motion sensors poll their parent router every five seconds or so.  If the RaspBee/ConBee happens to be their parent, you'll see the virtual LED blinking green.  This is perfectly normal behaviour; if you were to sniff the traffic when the sensor is attached to the Hue bridge, you'd see exactly the same.  And, afaik, this has been the behaviour since the very first firmware version of the original Hue motion sensor.

> Having left the device for a while the presence detection seems to have come back to life!

deCONZ needs to configure bindings and attribute reporting for the sensor to report motion.

> `No presence sensor found for 0x00178801XXXXXXXX, endpoint: 0x02`

I've been getting these messages as well, for all my 15 Hue motion sensors (13x SML001, 1x SML003, and 1x SML004). However, this has nothing to do with the firmware update.

Looking at the code, the are issued by `handleOccupancySensingClusterIndication()` when `getSensorNodeForAddressEndpointAndCluster()` doesn't find the sensor node.  `getSensorNodeForAddressEndpointAndCluster()` seems to be looking at the fingerprint to see if the device has _Occupancy Sensing_ on the specified endpoint.

Looking at the database, the `fingerprint` in the `sensors` table is empty for the Hue motion sensors?!  In the `device_descriptors` table, the _Simple Descriptor_  for endpoint 0x02 is registered correctly: `02 04 01 07 01 00 07 00 00 01 00 03 00 06 04 00 04 02 04 19 00 01 19 00`, including the 0x0406 cluster.

BTW, the `fingerprint` is also empty for:
- innr SP120 ZHAConsumption and ZHAPower
- Hue dimmer switch (new model) RWL022
- Hue smart button ROM001
- IKEA VINDSTYRKA
- Aqara mmWave senor

It's filled for my other Zigbee sensors.  As far as I can tell, the missing fingerprints are all for devices exposed through DDF without a `fingerprint` definition, whereas the sensors with fingerprint are exposed through DDF with `fingerprint` or through legacy code.

olicooper, 2023-04-02:
Perfect, great work, thank you for looking in to this for me! 👍 I will not worry about battery usage then, it's surprising that they don't run out of battery sooner with a poll frequency that high!

closed 2023-05-02 by github-actions[bot]