dresden-elektronik/deconz-rest-plugin #6855: Aqara High Precision Monitor sensor RTCGQ13LM functionality

andys73, 2023-04-02:
## Describe the bug
Running Home Assistant OS and deconz addon I get the following behavior:

    Regardless of the setting delay (and I press the button on the sensor to wake it up) it seems that the cool off period is 5 min which is way to big.
    Every hour even if they are no movement it registers a detection and clears it up right away.

When I pair it in Aqara I can see two settings. Sensitivity and cool off period. With sensitivity medium and cool off period at 30 seconds the sensor reports all great…

## Steps to reproduce the behavior
See bug description

## Expected behavior
Sensor to function within the cool off period and not to trigger false movement every hour

## Screenshots
[image]

## Environment

 - Host system: Raspberry Pi 4B
 - Running method: Home Assistant deCONZ Add-on 
 - Firmware version: 26780700
 - deCONZ version: 2.20.01
 - Device: ConBee II 
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system? If so: Which? Yes, SSD, Bluetooth dongle (on extension), USB UPS system 

## deCONZ Logs
no logs included

## Additional context
See forum https://forum.phoscon.de/t/aqara-high-precision-motion-sensor/3434 SwoopX provided additional DDF for testing

andys73, 2023-04-03:
Thanks for the information. Will try to test it out!

Finne75, 2023-04-08:
I have the same device (and they same issue). With this DDF, presence is always reported as "on" for me for some reason.

SwoopX, 2023-04-09:
@Finne75 Thanks for your feedback and testing it out. I assume the reason is rather simple: the device is missing a cluster in the fingerprint and therefore skips the necessary code to do the reset. Could you please add cluster `0x0406` to the fingerprint and restart deconz to test?

[image]

Finne75, 2023-04-10:
Ok I did, now I have the interesting phenomenon that the device is recognized as lumi.motion.agl04 in deconz (like before), but it does not appear in Phoscon and is also not exposed via API anymore. Still, if I right click to edit DDF, it seems to be the correct file. Note that the "values" part in the sensitivity settings are somehow dropped by the DDF editor, though (at least if I click on the "preview" pane).

7leman7, 2023-05-03:
The issue with false detection reported and cleared within the same second is still there...

andys73, 2023-05-03:
thank you @7leman7  and @Finne75 for testing the ddf, just never got around to be able to test the ddf myself.
was the cool period resolved with the ddf?

I am guessing that the issue with the false detection is when the device wakes up to report something back.. can this be fixed somewhere in the ddf?

7leman7, 2023-05-03:
Sorry, I guess I did not phrase that well enough. I did not test the ddf. I was able to change the cool down period okay though, I'm running deconz in ha, deconz version is listed as 2.21.2

AarneAarebye, 2023-05-11:
I took the DDF file, added the missing fingerprint parameter and can now successfully set "sensitivity" and "duration". But...

1. In Phoscon App it appears as "Version: unknown" and "Status: Not reachable". Usually, w./o. the new DDF and after some time, the corrrect values are been displayed and the device output can be used. Not with the new DDF.
2. I can query the device in node-red but the result is always "lastupdated: none" and "presence: false", so not motion detection at all (with the Aqara app I did check the device is working properly)

Environment:
- node-red with node-red-contrib-deconz
- Raspi 4B, newest OS & deconz, with RaspBee II
- Phoscon in Chrome browser under newest OSX
- DDF tested in folders ~/.local/share/dresden-elektronik/deCONZ/devices and /usr/share/deCONZ/devices/xiaomi 
- 28 Aqara sensors and 2 magic cubes handled by RaspBee II

[xiaomi_rtcgq13lm_high_precision_motion_sensor.json.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/11449864/xiaomi_rtcgq13lm_high_precision_motion_sensor.json.txt)

andys73, 2023-05-19:
that is fantastic. Can someone push that to the repository so it gets into a future release? Many thanks!!!

andys73, 2023-05-29:
Hi @SwoopX , @BabaIsYou or @Smanar can you push the ddf mentioned in the next release please as was successfully tested by @AarneAarebye ?

Many thanks

Smanar, 2023-05-29:
@AarneAarebye do you want to make the PR with your DDF or do you want I make it ?

AarneAarebye, 2023-05-29:
@Smanar: I would prefer if you do it. :-) Thank you!

SwoopX, 2023-05-29:
I can highly recommend reading deconz' release notes 🙂

closed 2023-05-29 by SwoopX