dresden-elektronik/deconz-rest-plugin #6856: Device name

guiumb, 2023-04-03:
Request for support device Hama TRV (https://zigbee.blakadder.com/Hama_00176592.html)
DDF file suggested by @Smanar in issue #6806 permit device DeConz enablement.

- Product name: Hama Smart Radiator Thermostat
- Manufacturer: _TZE200_h4cgnbzg
- Model identifier: TS0601
- Device type : Thermostat

Node Info:

[image]

Basic:

[image]

Groups:

[image]
[image]

Scenes

[image]
[image]
[image]
[image]

Tuya specific Cluster:

[image]

OTAU:

[image]
[image]

Time:

[image]

Smanar, 2023-04-03:
The used DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_h4cgnbzg",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "Saswell TRVs",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "meta": {
        "values": {
          "config/mode": {"off": 0, "heat": 1}
        }
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/lowbattery",
          "parse": {"fn": "tuya", "dpid": 105, "eval": "Item.val = Attr.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/heatsetpoint",
          "parse": {"fn": "tuya", "dpid": 103, "eval": "Item.val = Attr.val * 10;"},
          "write": {"fn": "tuya", "dpid": 103, "dt": "0x2b", "eval": "Item.val / 10;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "config/locked",
          "parse": {"fn": "tuya", "dpid": 40, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 40, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/windowopen_set",
          "parse": {"fn": "tuya", "dpid": 8, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 8, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/mode",
          "values": [
              ["off", 0], ["heat", 1]
          ],
          "parse": {"fn": "tuya", "dpid": 101, "eval": "if (Attr.val == 0) { Item.val = 'off' } else { Item.val = 'heat' }"},
          "write": {"fn": "tuya", "dpid": 101, "dt": "0x10", "eval": "if (Item.val == 'off') { false } else { true }"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 102, "eval": "Item.val = Attr.val * 10;"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/valve",
          "parse": {"fn": "tuya", "dpid": 104, "eval": "Item.val = Attr.val > 5;"},
          "read": {"fn": "none"}
        }
      ]
    }
  ]
}
```

If all is working just tell me if you want to make the PR (to submit the DDF officialy) or want I make it.

guiumb, 2023-04-03:
DDF file works fine. If you could open a PR I would be very grateful, and I think that would also help future buyers of this device.
I am very grateful for your help and support.

Smanar, 2023-04-04:
>that would also help future buyers of this device

And more, this device have so much clones
PR done, thx for help too.

https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6859

closed 2023-05-04 by github-actions[bot]