dresden-elektronik/deconz-rest-plugin #6861: Candeo Modmote Zigbee Scene Switch

lworbey, 2023-04-04:
## Device

- Product name: Candeo Modmote Zigbee Scene Switch
- Manufacturer: _TZ3000_czuyt8lz
- Model identifier: TS004F
- Device type : Please remove all unrelated device types. 
  - Remote
  - Switch

## Screenshots
[image]

### Basic
[image]

### Identify
[image]

### Groups
[image]

### On/Off
[image]

### Other clusters that are not mentioned above
[image]

BabaIsYou, 2023-04-05:
Some interesting information from manufacturer https://candeo.io/support-cmm-1/ , specially this file will help https://candeo.io/wp-content/uploads/2022/05/CandeoModmoteDTH.zip to have button map.

BabaIsYou, 2023-04-11:
May be you can try with this DDF 
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZ3000_rco1yzb1","_TZ3000_czuyt8lz"],
  "modelid": ["TS004F","TS004F"],
  "product": "TS004F Scene Switch",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          }
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 3600,
          "max": 14400,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0008"
    }
  ]
}
```

And add the device in the list in the button_maps.json file (you will found it in the "devices" folder) :  

```
        "Tuya3gangMap": {
            "vendor": "Tuya",
            "doc": "3-gang remote",
            "modelids": ["_TZ3000_ee8nrt2l", "_TZ3000_ygvf9xzp", "_TZ3000_t8hzpgnd", "_TZ3000_wkai4ga5", "_TZ3000_bi6lpsew", "_TZ3400_keyjhapk", "_TYZB02_key8kk7r", "_TZ3400_keyjqthh", "_TZ3400_key8kk7r", "_TZ3000_vp6clf9d", "_TYZB02_keyjqthh", "_TZ3000_peszejy7", "_TZ3000_qzjcsmar", "_TZ3000_owgcnkrh", "_TZ3000_adkvzooy", "_TZ3000_arfwfgoa", "_TZ3000_a7ouggvs", "_TZ3000_rrjr1q0u", "_TZ3000_abci1hiu", "_TZ3000_dfgbtub0", "_TZ3000_rco1yzb1", "_TZ3000_gbm10jnj", "_TZ3000_5e235jpa", "_TZ3000_sj7jbgks", "_TZ3000_w8jwkczz", "_TZ3000_dziaict4", "_TZ3000_famkxci2", "_TZ3000_itb0omhv", "_TZ3000_tzvbimpq", "_TZ3000_mh9px7cq", "_TZ3000_b7bxojrg", "_TZ3000_ufhtxr59", "_TZ3000_4upl1fcj", "_TZ3000_i3rjdrwu", "_TZ3000_abci1hiu",_TZ3000_czuyt8lz"],
```

Don't take care about the title `Tuya3gangMap`

closed 2023-05-11 by github-actions[bot]