dresden-elektronik/deconz-rest-plugin #6862: After update using reverse proxy sensors page is not reachable

raidenlal, 2023-04-04:
## Describe the bug

  1. Go to deCONZ plugin in hass
  2. Click on 'Phoscon'
  3. Click on hamburger menu: Devices->Lights
  4. Observed error

## Expected behavior
Devices get shown

## Screenshots

## Environment

 - Host system: HASS OS
 - Running method: RPI
 - Firmware version: (26xxyy00)
 - deCONZ version: 6.19.0
 - Device: ConBee
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? If so: zwave dongle

## deCONZ Logs

## Additional context

Mimiix, 2023-04-04:
Hi,

Phoscon bug reports are collected on the forums, as is described in #5113.

Closing this issue

closed 2023-04-04 by Mimiix

raidenlal, 2023-04-04:
Is this a joke?