dresden-elektronik/deconz-rest-plugin #6870: Device name

rweise, 2023-04-07:
## Device

- Product name: (Tuya) 2CH Zigbee Switch Modul-L  
- Manufacturer: _TZ3000_jl7qyupf
- Model identifier: TS0012
- Device type : 2-Way Switch without N  eVatmaster Consulting GmbH
 
[image]

## Screenshots

[image]
[image]

If there are more Information requested, i will do what i can :-) and sorry if it is not a new device. I do not found TZ3000_ji7qyupf in open requests

Mimiix, 2023-04-07:
Please add all screenshots of all clusters being read( as the template explains).

rweise, 2023-04-07:
I will do my best to add all screenshots, but it is not easy because that i only have VNC access. So i can not change resolution and deconz gives me only limited access to extend the window. For today i use the _TZ3000_...qew...tch_3gangs.json and modify it like the add on at the end of this post.
Now i can see it in Phoscon and in IOBroker. It works to switch both channels on/off equal witch one i use. It is not perfekt but enough for today. Like i say, i will post all screenshots the next days . 

{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_jl7qyupf",
  "modelid": "TS0012",
  "product": "Wired Switch 2 gangs",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/_TZ3000_jl7qyupf_wired_2gangs.json",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x02"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 5,
          "max": 1800
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 2,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 5,
          "max": 1800
        }
      ]
    }
  ]
}

Mimiix, 2023-04-07:
> I will do my best to add all screenshots, but it is not easy because that i only have VNC access. So i can not change resolution and deconz gives me only limited access to extend the window. For [...]

Without the screenshots, we can't proceed and a dev can't help.

Please add the screenshots.

Smanar, 2023-04-07:
>Now i can see it in Phoscon and in IOBroker. It works to switch both channels on/off equal witch one i use

You mean you can switch only both channel in same time ?

rweise, 2023-04-08:
Hi Smanar, yes, if i switch one on in Phoscon or iobroker it takes both to on and vice versa. Only at the switch i can switch them separately.  
[image]
[image]
Here are Identify:
[image]
[image]
Groups :

[image]
[image]
[image]
Scenes :
[image]
[image]
[image]
[image]
[image]
On/Off (first)
[image]
[image]
[image]
Unknown  is empty
Tuya specific : 
[image]
Basic: 
[image]
[image]
Otau: 
[image]
[image]
[image]
Time is empty 
Groups and Scenes from part 2 are the same as the first ones 
And here is On/Off from part 2: 
[image]

[image]

[image]

Smanar, 2023-04-08:
Try with this DDF, I have added the tuya unlock sequence

```
{
   "schema":"devcap1.schema.json",
   "manufacturername":"_TZ3000_jl7qyupf",
   "modelid":"TS0012",
   "product":"Wired Switch 2 gangs",
   "sleeper":false,
   "status":"Gold",
   "path":"/devices/_TZ3000_jl7qyupf_wired_2gangs.json",
   "subdevices":[
      {
         "type":"$TYPE_ON_OFF_LIGHT",
         "restapi":"/lights",
         "uuid":[
            "$address.ext",
            "0x01"
         ],
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"config/interfacemode",
               "refresh.interval":86400,
               "read":{
                  "fn":"zcl",
                  "ep":1,
                  "cl":"0x0000",
                  "at":[
                     "0x0004",
                     "0x0000",
                     "0x0001",
                     "0x0005",
                     "0x0007",
                     "0xfffe"
                  ]
               },
               "parse":{
                  "fn":"zcl",
                  "ep":1,
                  "cl":"0x0000",
                  "at":"0xfffe",
                  "eval":"Item.val = Attr.val"
               },
               "public":false,
               "default":0
            },
            {
               "name":"state/on",
               "refresh.interval":5
            },
            {
               "name":"state/reachable"
            }
         ]
      },
      {
         "type":"$TYPE_ON_OFF_LIGHT",
         "restapi":"/lights",
         "uuid":[
            "$address.ext",
            "0x02"
         ],
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"state/on",
               "refresh.interval":5
            },
            {
               "name":"state/reachable"
            }
         ]
      }
   ],
   "bindings":[
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0006",
         "report":[
            {
               "at":"0x0000",
               "dt":"0x10",
               "min":5,
               "max":1800
            }
         ]
      },
      {
         "bind":"unicast",
         "src.ep":2,
         "cl":"0x0006",
         "report":[
            {
               "at":"0x0000",
               "dt":"0x10",
               "min":5,
               "max":1800
            }
         ]
      }
   ]
}
```

You can force it with re-including the device (no need to remove it before)

BabaIsYou, 2023-04-08:
@Smanar , would have been a good candidate for your new specific attrib https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/generic/items/config_tuya_unlock_item.json ? ;-)

rweise, 2023-04-09:
@Smanar 
Thank you very much, now it works perfekt.
So the Request might be closed. 

Thanks to all at  dresden-elektronik for the phantastic work and also to all supporters and programmers 🥇

Mimiix, 2023-04-09:
> @Smanar Thank you very much, now it works perfekt. So the Request might be closed.
> 
> Thanks to all at dresden-elektronik for the phantastic work and also to all supporters and programmers 🥇

No closing yet, now we make a PR :)

Smanar, 2023-04-09:
> @Smanar , would have been a good candidate for your new specific attrib https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/generic/items/config_tuya_unlock_item.json ? ;-)

Lol, yes but it's new, not sure all users have it, I will ask on next one ^^

closed 2023-04-17 by manup