dresden-elektronik/deconz-rest-plugin #6871: Salus PS600 Pipe Temperature Sensor

Carl-mgb66, 2023-04-07:
## Device

- Product name: Salus PS600 Temperature Sensor
- Manufacturer: Computime
- Model identifier: SS909ZB
- Device type :
  - Sensor

## Screenshots

## Clusters of the Node

[image]

## Node Info Panel

[image]

### Basic

[image]

### Identify

[image]

### Diagnostics

[image]

### Other clusters that are not mentioned above

### Power Configuration

[image]

### Poll Control

[image]

### Temperature Measurement

[image]

### IAS Zone

[image]

BabaIsYou, 2023-04-09:
Hi,

May be you have to create two files under the folder for DDF files on your system

The first one with a name like `ss909zb_temp_sensor.json `containing 

```

{
  "schema": "devcap1.schema.json",
  "manufacturername": "Computime",
  "modelid": "SS909ZB",
  "product": "Salus PS600 Temperature Sensor",
  "sleeper": true,
  "status": "Silver",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x09",
        "0x0402"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0302",
        "endpoint": "0x09",
        "in": [
          "0x0001",
          "0x0402",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername",
          "awake": true
        },
        {
          "name": "attr/modelid",
          "awake": true
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "awake": true,
          "read": {
            "at": "0x0006",
            "cl": "0x0000",
            "ep": 9,
            "fn": "zcl",
          },
          "parse": {
            "at": "0x0006",
            "cl": "0x0000",
            "ep": 9,
            "fn": "zcl",
            "eval": "Item.val = Attr.val;"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0020",
            "cl": "0x0001",
            "ep": 9,
            "script": "computime_battery.js"
          }
        },
        {
          "name": "config/offset"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "awake": true
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 9,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0020",
          "dt": "0x20",
          "min": 3600,
          "max": 43200,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 9,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 600,
          "change": "0x0000000A"
        }
      ]
    }
  ]
}

```

Then a second file in the same place named `computime_battery.js`
```
const vmax = 30;
let bat = Attr.val;

if      (bat > vmax) { bat = vmax; }
else if (bat < vmin) { bat = vmin; }

bat = ((bat - vmin) /(vmax - vmin)) * 100;

if      (bat > 100) { bat = 100; }
else if (bat <= 0)  { bat = 1; } // ?

Item.val = bat;
```

Carl-mgb66, 2023-04-09:
Ok, thanks. I'm using deconz in Home Assistant so I'll have to work out where to put these.

BabaIsYou, 2023-04-09:
You'll have to change DDF level https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-change-DDF-modes-and-enable-or-disable-DDF-levels

Carl-mgb66, 2023-04-10:
Hi @BabaIsYou - thanks for the help. I've put your suggested files in the appropriate folder, but deconz doesn't seem to be picking them up and if I try to open in the DDF editor it give an error Failed to Open - any thoughts?

Carl-mgb66, 2023-04-10:
@BabaIsYou - thanks very much - I've managed to get it working now. I found a couple of other typos in the file, I suspect from when I copied it across - all seems good now.

Thanks very much.

BabaIsYou, 2023-04-10:
Good ! Do you want to make the pull request (PR) to have included in a next release ?

Carl-mgb66, 2023-04-10:
I would do, just not sure how!

BabaIsYou, 2023-04-10:
This time it’s not as simple as just adding a .json to a folder ;-)
The folder had to be created (under `devices`) and named `computime-pruse` (would serve for both because Pruse is a sub-brand of Computime and same device could have two different names as clones)  and then the files are to be created in it. 

Github will ask you to commit on master branch then add comments and something like fix #6871 to associate PR to this request.

And that’s it. Don’t worry PR will be reviewed and if something went wrong just delete the PR ;-)

Carl-mgb66, 2023-04-11:
OK - think I've done that right - let's see, thanks for the help.

BabaIsYou, 2023-04-11:
Looks Ok, bravo ! ;-)
Just perhaps change "See:" by "Fix" without ":", then github will be able to do the link between the PR and the issue and then close automaticaly the issue once the PR will be merge.

Carl-mgb66, 2023-04-11:
> I see you did it and the issue is now linked to the PR. Thx

No problem. Thanks for the help.

closed 2023-04-19 by manup