dresden-elektronik/deconz-rest-plugin #6873: Aqara / Mi Magic Cube MFKZQ01LM functionality

TC1666, 2023-04-08:
## Describe the bug
DDF for Aqara Cube does not function properly. There a two clusters for this device, analog input (rotation) and multi-state (gestures). Analog input works correctly, gestures does not. Setting DDF to Draft no longer works since C++ device code has been removed.

## Steps to reproduce the behavior

1. Install latest deconz version v2.21.2. 
2. Perform a gesture with cube, such as shake/stir, flip to another side, etc. REST API or values shown in Phoscon do not update.
3. Rotate the cube, values do update properly.
4. Rollback to a deconz version that still contains C++ device code, such as v2.18.2
5. Gestures once again work.

## Expected behavior
Multi-state/gestures should work.

## Screenshots
[image]

## Environment

 - Host system: Windows PC
 - Running method: Windows standalone
 - Firmware version: 26780700
 - deCONZ version: 2.21.2
 - Device: ConBee II 
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context
Uninstalling/deleting the device and re-adding does not work to solve issue.

SwoopX, 2023-04-08:
I have  to say, I'm tempted to close this one as my cubes update alright and otherwise, quite some stuff wouldn't work over here. Can you please share how the 2 sensors of the cube look like from an API perspective?

Btw, those are the websocket events when doing some gestures and rotations:

## Gestures

```
{
    "21:14:39:59": {
        "attr": {
            "lastseen": "2023-04-08T19:14Z"
        },
        "e": "changed",
        "id": "188",
        "r": "sensors",
        "t": "event",
        "uniqueid": "00:15:8d:00:12:34:56:78-02-0012"
    },
    "21:14:39:62": {
        "e": "changed",
        "id": "188",
        "r": "sensors",
        "state": {
            "buttonevent": 7000,
            "gesture": 0,
            "lastupdated": "2023-04-08T19:14:38.382"
        },
        "t": "event",
        "uniqueid": "00:15:8d:00:12:34:56:78-02-0012"
    },
    "21:14:53:512": {
        "e": "changed",
        "id": "188",
        "r": "sensors",
        "state": {
            "buttonevent": 2004,
            "gesture": 3,
            "lastupdated": "2023-04-08T19:14:52.844"
        },
        "t": "event",
        "uniqueid": "00:15:8d:00:12:34:56:78-02-0012"
    },
    "21:15:02:956": {
        "e": "changed",
        "id": "188",
        "r": "sensors",
        "state": {
            "buttonevent": 7007,
            "gesture": 1,
            "lastupdated": "2023-04-08T19:15:02.281"
        },
        "t": "event",
        "uniqueid": "00:15:8d:00:12:34:56:78-02-0012"
    },
    "21:18:09:354": {
        "attr": {
            "lastseen": "2023-04-08T19:18Z"
        },
        "e": "changed",
        "id": "188",
        "r": "sensors",
        "t": "event",
        "uniqueid": "00:15:8d:00:12:34:56:78-02-0012"
    },
    "21:18:09:355": {
        "e": "changed",
        "id": "188",
        "r": "sensors",
        "state": {
            "buttonevent": 7000,
            "gesture": 0,
            "lastupdated": "2023-04-08T19:18:08.673"
        },
        "t": "event",
        "uniqueid": "00:15:8d:00:12:34:56:78-02-0012"
    },
    "21:18:09:842": {
        "e": "changed",
        "id": "188",
        "r": "sensors",
        "state": {
            "buttonevent": 3002,
            "gesture": 3,
            "lastupdated": "2023-04-08T19:18:09.170"
        },
        "t": "event",
        "uniqueid": "00:15:8d:00:12:34:56:78-02-0012"
    },
    "21:18:31:870": {
        "e": "changed",
        "id": "188",
        "r": "sensors",
        "state": {
            "buttonevent": 1003,
            "lastupdated": "2023-04-08T19:18:31.201"
        },
        "t": "event",
        "uniqueid": "00:15:8d:00:12:34:56:78-02-0012"
    },
    "21:18:31:874": {
        "e": "changed",
        "id": "188",
        "r": "sensors",
        "state": {
            "buttonevent": 1003
        },
        "t": "event",
        "uniqueid": "00:15:8d:00:12:34:56:78-02-0012"
    },
    "21:18:37:868": {
        "e": "changed",
        "id": "188",
        "r": "sensors",
        "state": {
            "buttonevent": 6001,
            "gesture": 4,
            "lastupdated": "2023-04-08T19:18:37.192"
        },
        "t": "event",
        "uniqueid": "00:15:8d:00:12:34:56:78-02-0012"
    }
}
```

## Rotation

```
{
    "21:14:39:58": {
        "attr": {
            "lastseen": "2023-04-08T19:14Z"
        },
        "e": "changed",
        "id": "187",
        "r": "sensors",
        "t": "event",
        "uniqueid": "00:15:8d:00:12:34:56:78-03-000c"
    },
    "21:18:09:347": {
        "attr": {
            "lastseen": "2023-04-08T19:18Z"
        },
        "e": "changed",
        "id": "187",
        "r": "sensors",
        "t": "event",
        "uniqueid": "00:15:8d:00:12:34:56:78-03-000c"
    },
    "21:18:13:293": {
        "e": "changed",
        "id": "187",
        "r": "sensors",
        "state": {
            "buttonevent": -9382,
            "gesture": 8,
            "lastupdated": "2023-04-08T19:18:12.602"
        },
        "t": "event",
        "uniqueid": "00:15:8d:00:12:34:56:78-03-000c"
    },
    "21:18:25:84": {
        "e": "changed",
        "id": "187",
        "r": "sensors",
        "state": {
            "buttonevent": 11998,
            "gesture": 7,
            "lastupdated": "2023-04-08T19:18:24.410"
        },
        "t": "event",
        "uniqueid": "00:15:8d:00:12:34:56:78-03-000c"
    }
}
```

TC1666, 2023-04-08:
I totally understand your feelings, when it first stopped working I checked with other Cube users I know and they said they have no problem. The only thing I can think of is that I have a variation of the model? Aqara seems to be notorious for that.

When you say " 2 sensors of the cube look like from an API perspective?" where/how should I be capturing that?

SwoopX, 2023-04-09:
In Phoscon, click on **Help**, then **API information**. Now select the switch filter and choose the respective sensors to provide the requested info.

TC1666, 2023-04-11:
Here's the API Info below. Another clue to note, after I install a newer version of deconz that no longer has the internal device code (just the DDF), it ends up renaming the cube to something generic. In other words, I had previously renamed the cube to "Aqara Cube" but after installing the latest deconz, the device gets renamed to something generic such as "Switch 31". It's like deconz doesn't fully recognize the device. I have several other devices in the network and none of them have gotten renamed during a deconz update. 

**Rotation** 
```
{
    "config": {
        "battery": 22,
        "on": true,
        "reachable": true
    },
    "ep": 3,
    "etag": "fbdc954ad101cf69f6081d24cd9fda87",
    "lastannounced": null,
    "lastseen": "2023-04-11T00:03Z",
    "manufacturername": "LUMI",
    "mode": 1,
    "modelid": "lumi.sensor_cube.aqgl01",
    "name": "Switch 31",
    "state": {
        "buttonevent": -8674,
        "gesture": 8,
        "lastupdated": "2023-04-11T00:03:45.057"
    },
    "swversion": "20160704",
    "type": "ZHASwitch",
    "uniqueid": "00:15:8d:00:05:42:ae:09-03-000c"
}
```
**Gesture**
```
{
    "config": {
        "battery": 22,
        "on": true,
        "reachable": true
    },
    "ep": 2,
    "etag": "6319fa7fab8d63bde737888bf3d64a46",
    "lastannounced": null,
    "lastseen": "2023-04-11T00:04Z",
    "manufacturername": "LUMI",
    "mode": 1,
    "modelid": "lumi.sensor_cube.aqgl01",
    "name": "Switch 32",
    "state": {
        "buttonevent": 7007,
        "gesture": 1,
        "lastupdated": "2023-03-18T19:34:08.303"
    },
    "swversion": "20160704",
    "type": "ZHASwitch",
    "uniqueid": "00:15:8d:00:05:42:ae:09-02-0012"
}

```

SwoopX, 2023-04-11:
I actually meant what you shared at the beginning of your post, thanks. I'd advise to change the battery, Xiaomi devices start acting weirdly if battery is low and yours is done. Had the same 3 weeks ago and after change, the cube was usable again.

TC1666, 2023-04-11:
Unfortunately, battery is not the issue. One more piece of strange information: when I upgrade to a version that only has the DDF, the device will report a low battery of about 22%. Rolling back to an earlier version and the battery is at 100% (I did change it to a fresh battery previously). It still doesn't explain why rotation works without issues but gestures does not. Again, simply rolling back to an earlier deconz version and everything works 100%.

TC1666, 2023-04-14:
@SwoopX is there anything additional I can provide to help troubleshoot this issue?

SwoopX, 2023-04-17:
Checked the Windows version and it has indeed issues with file `xiaomi_cube_gestures.js`. Can you please find and open it on your system to remove the degree sign (°)? That made it work in my tests here.

TC1666, 2023-04-18:
@SwoopX that solved the problem. Who would have thought that a special character in a comment would screw things up? Comments are supposed to be ignored by interpreters/compilers, right? I'm assuming the fact the character is at EOL did something?

Still two open questions on my side:

- Why does upgrading from to v 2.21.2 from a 2.18.x rename my cube? Is it because deconz is now using a DDF instead of internal code for that device?? None of my other devices get their name changed during updates.
- Why does deconz now show my cube battery at 22% when it was 100% before the update?

Thanks for your patience in sticking with this one.