dresden-elektronik/deconz-rest-plugin #6876: resetPassword - /api/config/password can't be use without phoscon

Zehir, 2023-04-10:
## Describe the bug

The [documentation](https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/configuration/#reset-password) say that I can reset the password by using a `DELETE` request on `/api/config/password` but each time I get an `unauthorized user` response.

## Steps to reproduce the behavior

Make a `DELETE` request on `/api/config/password`

## Expected behavior

The gateway should reset the password and return a 200 code.

## Screenshots

## Environment

 - Host system: Windows
 - Running method: Windows
 - Firmware version: 0x26720700
 - deCONZ version: 2.21.1
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

For some reason the code is checking for the `Referer` header and the pattern set seems to accept only Phoscon like login URL.
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/4d366e41856bf7b06c6b76b116dcc48ef29b6c9d/rest_configuration.cpp#L2728-L2732

Can we remove this code ?

Thanks

SwoopX, 2023-04-10:
I assume, the intention to allow the reset only from one place or under certain conditions doesn't qualify as a bug?

Zehir, 2023-04-10:
I don't know why there check is here but in that case the documentation should mention it. But is this was made for security reason any web app can provide a work around.

It's more a question than an issue

SwoopX, 2023-04-10:
Surely agree on the documentation. The rest can/should be discussed (guess where) 🙂

Zehir, 2023-04-11:
This check was added by @manup 6 years ago in https://github.com/dresden-elektronik/deconz-rest-plugin/commit/6f846c9bf45c7f4e1b45be6c68f13596a9bace9e maybe he can tell us more about it ?

Mimiix, 2023-04-11:
What happens if you do the reset procedure as Phoscon has? That you need to physically RESTART the device. 

[image]

Zehir, 2023-04-11:
On phoscon it's working

Zehir, 2023-04-11:
The issue is not the power cycle it's the refered.
See screenshoot of postman;
Without Referer:
[image]

With Referer: 
[image]

Mimiix, 2023-04-11:
Did you note you had to put in the APIKEY in the request?

Zehir, 2023-04-11:
> Did you note you had to put in the APIKEY in the request?

It's what the documentation say but it's not true.

Even phoscon don't do it : 
[image]

And the code don't look for it :
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/4d366e41856bf7b06c6b76b116dcc48ef29b6c9d/rest_configuration.cpp#L669-L673

I made a PR to update the documentation : https://github.com/dresden-elektronik/deconz-rest-doc/pull/31

Mimiix, 2023-04-11:
Intresting.

But does it work according to the documentation?

Zehir, 2023-04-11:
No, either we add a warning message on the doc that say the Referer need to contain "login.html" or update the code to remove the check. In both case the documentation need to be updated because you don't need the api key to reset the password. Who make sense because if you don't know the password you may not have a valid api key to reset it

Mimiix, 2023-04-11:
Agree.

Zehir, 2023-04-11:
If you want you cant test it on the playground page of the rest-client I building : 
https://deconz-community.github.io/rest-client/
https://deconz-community.github.io/rest-client/login.html/

look for `resetPassword` on the page to find the correct endpoint to call

Zehir, 2023-05-03:
Up

Zehir, 2023-05-25:
Not resolved

Zehir, 2023-06-16:
Up

closed 2023-07-15 by github-actions[bot]