dresden-elektronik/deconz-rest-plugin #6881: Can't set code0 of alarm system on deconz running on windows

Zehir, 2023-04-11:
## Describe the bug

It's look like we can't set the alarm system pin code0 when deconz is running on windows. I made the test with the same request on my prod env running under home assistant docker thingy and on it it's working.

## Steps to reproduce the behavior

Make a PUT request on `http://localhost/api/SECRET/alarmsystems/1/config` with data 
```json
{"code0": "1234"}
```

And I get the result :
```json
[
    {
        "error":{
            "address":"/alarmsystems/1",
            "description":"internal error, failed to set code, occured",
            "type":901
        }
    }
]
```

## Expected behavior

I should get this response instead :
```json
[
    {
        "success":{
            "/alarmsystems/1/config/configured":true
        }
    }
]
```

## Screenshots

## Environment

 - Host system: PC
 - Running method: Windows
 - Firmware version: 0x26720700
 - deCONZ version: 2.21.1
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

```txt
11:15:50:279 HTTP API OPTIONS /api/SECRET/alarmsystems/1/config - 127.0.0.1
11:15:50:279 ApiMode: 0
11:15:50:593 HTTP API PUT /api/SECRET/alarmsystems/1/config - 127.0.0.1
11:15:50:593 Text Data: 	{"code0":"1234"}
11:15:50:593 ApiMode: 0
11:15:50:593 API error 901, /alarmsystems/1, internal error, failed to set code, occured
11:15:50:594 [{"error":{"address":"/alarmsystems/1","description":"internal error, failed to set code, occured","type":901}}]
```

## Additional context

It's a clean database, I run a resetGateway `/api/:apiKey/config/reset` with both resetGW and deleteDB yesterday.

Zehir, 2023-05-11:
Not resolved

Mimiix, 2023-05-11:
Please keep active :)

manup, 2023-06-03:
One of these problems, after testing the released versions for a while it works on my machines.
Turns out that the crypto libraries depend on msvcr100.dll a MS redistributable thing. Which was installed on my machines but might be missing in your setup.

I've now added this msvcr100.dll to be included in the deCONZ Windows package so with v2.22.1 this should work without having to install anything extra.

closed 2023-06-03 by manup