dresden-elektronik/deconz-rest-plugin #6885: TS0011 - BSeed Zigbee Smart Wall-Switch

dpozimski, 2023-04-13:
## Device

- Product name: BSEED Smart Zigbee Wall-Switch.
- Manufacturer: _TZ3000_hafsqare
- Model identifier: TS0011
- Device type:
- Other: Wall-Switch

## Screenshots

### Basic
[image]
[image]
[image]

### Identify
[image]

### Groups
[image]
[image]

### Scenes
[image]
[image]

### On/Off
[image]
[image]

### OTAU
[image]

### Tuya Specific
[image]

dpozimski, 2023-04-13:
### Basic
[image]

### Identify

[image]

### Alarms

### Device Temperature

### Groups

[image]

### Scenes
[image]

### On/Off
[image]

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
[image]
[image]

@BabaIsYou

BabaIsYou, 2023-04-13:
By the mean time, could perhaps try to adapt the DDF from #6422

dpozimski, 2023-04-15:
It was not possible to paste it to deconz gui from home Assistant. The ddf json that i found was in preview tab and it could not be edited.

BabaIsYou, 2023-04-15:
> t was not possible to paste it to deconz gui from home Assistant

It wasn't possible in Deconz GUI, from HA or whatever you use !

> The ddf json that i found was in preview tab and it could not be edited.

DDF has to be stored in a stecific folder (may be could find hints here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6798#issuecomment-1466483973). Then just rejoining the device may be sufficient.

dpozimski, 2023-04-29:
hi, what is the status of the issue?

BabaIsYou, 2023-04-30:
Did you have a look at my previous post to give a try to the DDF ? Without someone testing it we couldn’t add it to a version

closed 2023-05-28 by github-actions[bot]