dresden-elektronik/deconz-rest-plugin #6886: _TZ3000_pfc7i3kt - Behaves like all the switches are grouped

sciorty, 2023-04-13:
## Describe the bug
The device has been correctly paired but when i turn on/off **any** of the switches (from deCONZ or HA GUI) all the remaining 2 turn on/off as well. This does not occours when i manually press the physical switches, in that case they correctly behave as individual switches.

## Steps to reproduce the behavior
Just toggle any of the switches from deCONZ or HA GUI

## Expected behavior
Switches should be toggled individually

## Screenshots

[image]

## Environment

 - Host system: Raspberry
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26610700
 - deCONZ version: 2.21.02
 - Device: RaspBee II
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
No logs changes when toggling the lights

## Additional context

[image]

[image]

closed 2023-04-13 by Mimiix

sciorty, 2023-04-13:
Sorry 

> Hi,
> 
> Thanks for the report!
> 
> This is not a bug, this is a device not implemented yet. See : #6886
> 
> I have re-opened the above :)

I0m sorry, which issue did you reopen? You referenced this same issue :)

sciorty, 2023-04-19:
@Mimiix ?

Mimiix, 2023-04-20:
Huh?!

I check this morning.

Mimiix, 2023-04-20:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6682 was the one i reopened.

Looks like it was also fixed in last beta ;)

sciorty, 2023-04-20:
Thanbks for the update. I still have the problem though, i'll ask on https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6682