dresden-elektronik/deconz-rest-plugin #6901: Pairing Broken - KEEN Smart Vent

mmiller7, 2023-04-15:
## Describe the bug

When pairing a KEEN smart-vent, the "level control output" is not created in the "lights" page anymore.  This means that it is not possible to finish setting up a new vent.

## Steps to reproduce the behavior

Enter pairing mode in deCONZ
Insert batteries in a factory-reset vent, attempt to connect to network
Wait for join timeout

## Expected behavior

The level-controllable output is created, and the multi-sensor is created

## Screenshots

## Environment

 - Host system: PC
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: (26xxyy00)
 - deCONZ version: Last working: 06.11.1; First known broken: 06.16.0
 - Device: ConBee II
 - Do you use an USB extension cable: Yes-- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which? Aeotec Z-Wave, RTL-SDR
 
## deCONZ Logs

## Additional context

Previously added smart-vents work fine, but new smart vents will not add correctly in newer deCONZ version.  Had to roll back all the way to 06.11.1 to get it working properly.

Mimiix, 2023-04-16:
Can you please use deconz versions and not addon versions?

mmiller7, 2023-04-16:
> Can you please use deconz versions and not addon versions?

I did not know there are different version numbers.  How would I find the correct version number?

Mimiix, 2023-04-16:
Deconz versions can be found under gateway in Phoscon and under about or help in deconz.

Can you also provide the manufacturer name and model id from the basic cluster for this device?

In the wiki you can find how to read clusters

mmiller7, 2023-04-17:
Found under Gateway, the last working version I have is 2.13.04 / 12/18/2021

[image]

[image]

Hopefully that is the info you needed?

Later this week if you need samples from a broken newer instance I can do a backup and then test that for you, but I don't know what additional details to capture.

I do know AFTER paired, the newer version continues to work fine...it seems to only be a failure DURING pairing the "light" adjustable level output cover portion is not created in newer versions during pairing

Mimiix, 2023-04-17:
Seems like the legacy code is not used anymore/removed and there is no DDF yet. 

@ebaauw  I am not sure what to do here as there was never a request for this device(specifically), but i did find you integrating a similar one: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/269. 

Perhaps we just need a new DDF?

manup, 2023-04-17:
Perhaps we should add a more generic `state/level` attribute similar like it is described in the ZCL Level Control Cluster, which isn't limited to brightness?

mmiller7, 2023-05-09:
As far as I know this hasn't been fixed.

Anything I can do in the meantime other than sitting on the older version that works?

mmiller7, 2023-06-07:
Uh, can this be kept open?

As far as I know this hasn't been fixed?

Mimiix, 2023-06-07:
Hi, Need to respond in time when the bot pings.

mmiller7, 2023-06-08:
> Hi, Need to respond in time when the bot pings.

Sorry - I thought I was, I didn't see the one between the last reply and when it suddenly said closed

mmiller7, 2023-06-27:
bump to keep open

mmiller7, 2023-07-19:
Still open

closed 2023-08-18 by github-actions[bot]

mmiller7, 2023-08-19:
Still an issue