dresden-elektronik/deconz-rest-plugin #6905: Tuya Water Leak Sensor TS0207 | _TZ3000_upgcbody

theflyer08, 2023-04-17:


Mimiix, 2023-04-17:
Hi,

You need to add the screenshots as you own the device.

BabaIsYou, 2023-04-17:
As soon as the request is completed, we will quickly see if it can still be included in the pending PR #6833 or if a new DDF is needed in case this device is specific and not only a clone.

theflyer08, 2023-04-17:
> Hi,
> 
> You need to add the screenshots as you own the device.

Can you tell me how I can extract the Screenshots?

Mimiix, 2023-04-17:
Please check the template that was presented to you during creation. That forwards you to this wiki page:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

theflyer08, 2023-04-17:
> You need deCONZ-Gui for this. It can't be achieved with Phoscon.

@Mimiix Thanks a lot for the explanation. Now I get it. Therefore, I need to install a deskop OS and install deCONZ-Gui.

PR #6920 (2023-05-21): Combine two Tuya water leak sensor DDFs
Combine _TZ3000_TS0207_water_leak_sensor.json and  _TZ3000_water_sensor.json in a unique DDF.
Fix #6905
closed 2023-05-16 by github-actions[bot]