dresden-elektronik/deconz-rest-plugin #6912: Luminea Home Control Bewegungs-Melder: ZigBee-kompatibler Outdoor-PIR-Sensor mit Solarpanel, App, IP55 (Bewegungsmelder Solarpanel)

wunni1, 2023-04-19:
## Device

- Product name: Luminea Home Control Bewegungs-Melder: ZigBee-kompatibler Outdoor-PIR-Sensor mit Solarpanel, App, IP55 (Bewegungsmelder Solarpanel)
- Manufacturer: _TZ3000_jmrgyl7o
- Model identifier: TS0202
- Device type : Sensor
  

## Screenshots
[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

BabaIsYou, 2023-04-20:
Can you redo screenshots after clicking on "read" button please ?

wunni1, 2023-04-20:
Oh sorry, here are the new screenshots:

Basic

[image]

Identify

[image]

IAS Zone 1/2

[image]

IAS Zone 2/2
[image]

Node Info

[image]

OTAU Cluster 1/2

[image]

OTAU Cluster 2/2

[image]

Power Configuration Cluster

[image]

wunni1, 2023-04-20:
Thank you, I’ve tried it several times with an other sensor. But I didn’t match it. Run deconz under Home Assistant (no Docker). Can‘t found the path to the file we’re to save. So I‘ve to wait for the next update…

BabaIsYou, 2023-04-20:
You’ll probably get some hints reading this https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6798#issuecomment-1466483973

PR #6915 (2023-05-12): DDF update for various ts0202 presence sensors
DDF update for various ts0202 presence sensors.
Guess will close #6912
closed 2023-05-12 by manup