dresden-elektronik/deconz-rest-plugin #6916: Fantem ZB003-X

pocipocster, 2023-04-20:
## Device

- Product name: The device name as shown on the product or package.
- Manufacturer: As per deCONZ GUI Basic cluster.
- Model identifier: As per deCONZ GUI Basic cluster.
- Device type : Please remove all unrelated device types. 
  - Light
  - Lock
  - Remote
  - Sensor
  - Siren
  - Switch
  - Thermostat
  - Other: 

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2023-04-20:
Hi, 

Looks like the template is empty.

Can you please make sure all info is added?

pocipocster, 2023-04-20:
My mistake!. Selected Edit but can't seem too

Mimiix, 2023-04-20:
Feel free to open a new one if you can't edit  :)

Mimiix, 2023-04-20:
Closing this one.

closed 2023-04-20 by Mimiix

pocipocster, 2023-04-20:
So new one ok ?

Cheers