dresden-elektronik/deconz-rest-plugin #6917: Fantem ZB003-X

pocipocster, 2023-04-20:
## Device

- Product name: Fantem ZB003-X
- Manufacturer: _TZ3210_wuhzzfqg
- Model identifier: TS0202
- Device type : Please remove all unrelated device types. 
  - Light
  - Sensor PIR
  - Thermostat
  - Other: Humidity

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Mimiix, 2023-04-21:
Looks complete :)

pocipocster, 2023-04-21:
Awesome ! Thanks

Smanar, 2023-04-21:
Hello, can you test this DDF ?

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3210_wuhzzfqg",
  "modelid": "TS0202",
  "vendor": "Fantem",
  "product": "Multi sensor ZB003-X",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",                   
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "../tuya/tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/duration",
	      "default": 120	
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
[... 301 lines cut ...]
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 3,
      "cl": "0x0400",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 600,
          "change": "0x000007d0"
        }
      ]
    }
  ]
}
```

To work this DDF need to be in the "tuya" folder, else if you have HA and can do nothing, you need to add in the persistent folder 
- tuya_swversion.js else you will miss the swversion (not a big problem)
- sml_light_level else you will miss the light values

And ofc you need to edit the path in the DDF according to waht you will make.

pocipocster, 2023-04-21:
Hello

Sorry - you are going to have to guide me where this file goes !!

Thanks
P.s newbie !

Smanar, 2023-04-21:
What is your OS ?

pocipocster, 2023-04-21:
Thanks for your help
Windows 11

Smanar, 2023-04-21:
ha ?
Hu .... Honnestly I have no idea where DDF need to be on windows ?
Can you make a search for exemple on the file "_TZ3000__TS0207_range_extender.json" ?

pocipocster, 2023-04-21:
Hey

Hmmm, I’m looking!. Can’t find anything. Was hoping they might be in the deconz directory but it would appear not!
Can I ask what DDF stands for?

Once I find where this should go I can test it yes?. If it works do you then add it to the next conbee flash update? ( is that how it works??)

Cheers

Smanar, 2023-04-22:
>Was hoping they might be in the deconz directory but it would appear not!

So I have installed it on my side
You have the system folder in the_folder_you_have_installed_deconz/devices

You just need to create a text file called what_you_want.json in the folder devices/tuya with the DDF contain, they restart deconz and re-include the device.

pocipocster, 2023-04-23:
Hi,

Have installed the DDF and had some luck!
Motion appears to work as does luminosity .

Temperature and humidity not though.

You can see in Phoscon they are both 0; in my HA they aren't read correctly...
[image]
[image]

pocipocster, 2023-04-23:
[image]

Smanar, 2023-04-23:
Ha yes I remember now, we have this issue too on first version (with legacy code), we need to enable a reporting mode.
Can you try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3210_wuhzzfqg",
  "modelid": "TS0202",
  "vendor": "Fantem",
  "product": "Multi sensor ZB003-X",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",                   
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "../tuya/tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/duration",
	      "default": 120	
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
[... 313 lines cut ...]
      "cl": "0x0400",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 600,
          "change": "0x000007d0"
        }
      ]
    }
  ]
}
```

You will have a new field in the ZHAPresence sensor. a config/configured, I think this value will be false. You need to set it to true.
IDK if you can do that using your third app.

It will be possible using curl with the request

`curl -H 'Content-Type: application/json' -X PUT -d '{"configured": true}' http://IP:PORT/api/APIKEY/sensors/ID/config`

IP and PORT are the one visible on your browser when accessing phoscon, or deconz ip,port.
APIKEY is an apikey, can use the same used by your third app
ID is the device ID, I think "2" according your capture.

pocipocster, 2023-04-24:
Hi,

I tried the new DDF. It's completely unclear what I need to do in your instructions at the end.
At the moment it still isn't reporting humidity nor temperature.

What is this new field you mention?

[image]
[image]

Smanar, 2023-04-24:
With the new DDF you will see a new field in the ZHAPresence sensor.
(I think just a deconz restart is enought, noneed to re-include the device)
You can take a look in phoscon/help/API/information/sensors/2, not sure for the 2, but it's your device id.

pocipocster, 2023-04-25:
Ok. I've been reading and learning!! ( a bit! ).
How do I do your curl request as a POST? ( or is it a PUT????).
Once done is this change permanent to the device? i.e. after power on/off will it retain it?. Need I do this to all sensors of this type??

thanks in a advance

pocipocster, 2023-04-25:
This the new field?

   {
          "name": "config/configured",
          "parse": {"fn": "tuya", "dpid": 112, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 112, "dt": "0x10", "eval": "Item.val;"},
          "default": false

Can I not just set default to true in the schema? - forgive if a stupid question!

Smanar, 2023-04-25:
`You can take a look in phoscon/help/API/information/sensors/2, not sure for the 2, but it's your device id.`

You third app don't display the used API key ?

>Can I not just set default to true in the schema? 

IDK, from the code yes, but never worked on my side, we can try after all other tests.

pocipocster, 2023-04-25:
I have the api key now.
But I don’t understand how to covert your curl request to http.

If I set scheme configure flag to true what do we expect to happen?
This flag just for presence detection yes? ; not fix temperature/humidty not being read.

thanks again.

Smanar, 2023-04-25:
>This flag just for presence detection yes?

This flag enable the reporting, I have put it in the ZHApresence sensor, because it's the "main" sensor, but nothing to see with it.

>But I don’t understand how to convert your curl request to http.

Just use the request in a console.

pocipocster, 2023-04-25:
OK!. I am trying to learn all this as fast as I can! - Will try !!!

Smanar, 2023-04-25:
There is a more detailled procedure here https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/
But just for test I can give you command line to use on console.

pocipocster, 2023-04-25:
I followed that guide and managed to get api key as it suggests.

Can you give me the exact command line to use for curl?
When I try your suggested line I get “could not resolve host application “ error in cmd.

thanks

pocipocster, 2023-04-25:
Yes. This is exactly what I do.

So I’m not sure what brackets, quotes, ‘ “ { } etc. you need exactly

errors in cmd are:

Could not resolve host: application
unmatched close brace/bracket in URL position 5: true}'

pocipocster, 2023-04-26:
Had some success.

Can send your curl command now! 

But! - I don't think the device id is 2 ( where did you get/see that? ); looking at api in postman - it looks like 6. But still gives an error " resource, /sensors(2 or 6 no matter which), not available "
[image]

pocipocster, 2023-04-26:
I think it is visible. But I still get error when I post that resource is not available
[image]

pocipocster, 2023-04-26:
Ok. Notice something!
Edit DDF is not the same as the DDF file!.
I tried reloading deconz; removing, re-adding the device.

I assume this is the problem!!! - but why?
[image]

pocipocster, 2023-04-26:
[image]

Smanar, 2023-04-27:
>I think it is visible

No, I m sure it's not here ^^, it will be visible here
[image]

But it can be hidden, you have `"public":false` on one DDF, it hide the field on the API, so not possible to be sure if it's here or not just looking in the API.

IDK what happen, some DDF part are good, but deconz have added some lines ...

You can use the feature "load" in the DDF editor ? To load the good DDF ?
Else can just check the box "Public item" save the file and make a hot reload.

pocipocster, 2023-04-27:
Curl error
[image]

pocipocster, 2023-04-27:
My BAD! - Curl works with a PUT.

Though humidity/temperature still show 0. Reloaded deconz ; re added device. Still nothing :-(

pocipocster, 2023-04-27:
This is strange.
I send the curl. It says OK.
I then GET the sensor/config - but it still shows false......
[image]
[image]
[image]

pocipocster, 2023-04-27:
> {
> 	"schema": "resourceitem1.schema.json",
> 	"id": "config/configured",
> 	"datatype": "Bool",
> 	"access": "R",
> 	"public": true,
> 	"description": "Determines that a resource is configured."
> }
>
> Perhaps because the field I have chosed is write only ? Can mke a try with "config/interfacemode" to test.
>
> Can be something bad in the request and it's the device itself that return back to "false".
> This is visible on deconz logs with the flag "info" and "DDF"
>
>         DBG_Printf(DBG_INFO, "TY_DATA_%s: seq %u, dpid: 0x%02X, type: 0x%02X, length: %u, val: %d\n",
>                    rt, seq, dpid, dataType, dataLength, num.s32);

I’m sorry I don’t understand what you want me to do here!

Create this schema? Do what with it? - understand the problem you are suggesting but don’t understand what I do here!

pocipocster, 2023-04-27:
Not easier just to send me another complete ddf for this device ???

pocipocster, 2023-04-27:
Or I post you a sensor ( no need to return it ! )

Smanar, 2023-04-28:
No, we are close to the solution, we have same issue on previous integration https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4511#issuecomment-832872212

Here a new DDF, but better to delete, replace the DDF and re-include the device

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3210_wuhzzfqg",
  "modelid": "TS0202",
  "vendor": "Fantem",
  "product": "Multi sensor ZB003-X",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",                   
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "../tuya/tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/duration",
	      "default": 120	
        },
        {
          "name": "config/on"
        },
        {
[... 309 lines cut ...]
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 3,
      "cl": "0x0400",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 600,
          "change": "0x000007d0"
        }
      ]
    }
  ]
}
```

I have replaced the config/configured by config/interfacemode for test.

So can you make same test with `{"interfacemode":true}`
If it don't work with `{"interfacemode":false}`
If it still don't work whith taking a look in logs in same time

Smanar, 2023-04-29:
Yep, IDK why it don't work using DDF, but we just need to make a zigbee tuya request to enable the reporting for the 2 missing sensor, it was already done on legacy code, so there is nothing magic, just I don't see what can be the problem using DDF.

pocipocster, 2023-05-08:
Hello!

I have returned.
Have added new DDF and re-added device.
Please tele exactly what checks you need me to do - as I've forgotten/confused still! :-)

Smanar, 2023-05-08:
So you are using this DDF ATM https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6917#issuecomment-1527667071 ?

It's still not working for temperature and humidity ?
If no, can you check the value of "config/interfacemode" ?
Can you try to change the value to true and false.

pocipocster, 2023-05-08:
using latest DDF yes.
Still no temp or humidity.

Setting interface mode to true returns OK.

But reading it with GET it is still zero.....

pocipocster, 2023-05-08:
I tried true and false. Also set debug log DDF and info_l2. Can't see anything relating to this sensor!......

[image]

pocipocster, 2023-05-08:
This help?

DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode, lastseen, lastannounced) VALUES ('26', 'TS0202', 'ZHAPresence', 'TS0202', '_TZ3210_wuhzzfqg', '8c:f6:81:ff:fe:bf:9b:6a-01-0500', '2.2.1', '{"lastupdated":"2023-05-08T16:18:04.767","lowbattery":false,"presence":false,"tampered":false}', '{"battery":0,"duration":120,"enrolled":1,"interfacemode":0,"on":true,"pending":0,"reachable":true}', '{"d":1026,"ep":1,"in":[0,1,1280],"p":260}', 'normal', '1', '2023-05-08T16:17Z', '')
17:18:23:128 DB saved in 15 ms
17:18:23:212 Websocket 192.168.8.147:50621 send message: {"attr":{"id":"26","lastannounced":null,"lastseen":"2023-05-08T16:18Z","manufacturername":"_TZ3210_wuhzzfqg","modelid":"TS0202","name":"TS0202","swversion":"2.2.1","type":"ZHAPresence","uniqueid":"8c:f6:81:ff:fe:bf:9b:6a-01-0500"},"e":"changed","id":"26","r":"sensors","t":"event","uniqueid":"8c:f6:81:ff:fe:bf:9b:6a-01-0500"} (ret = 326)
17:18:23:213 Websocket 192.168.8.147:50624 send message: {"attr":{"id":"26","lastannounced":null,"lastseen":"2023-05-08T16:18Z","manufacturername":"_TZ3210_wuhzzfqg","modelid":"TS0202","name":"TS0202","swversion":"2.2.1","type":"ZHAPresence","uniqueid":"8c:f6:81:ff:fe:bf:9b:6a-01-0500"},"e":"changed","id":"26","r":"sensors","t":"event","uniqueid":"8c:f6:81:ff:fe:bf:9b:6a-01-0500"} (ret = 326)
17:18:23:215 Websocket 192.168.8.147:50621 send message: {"attr":{"id":"29","lastannounced":null,"lastseen":"2023-05-08T16:18Z","manufacturername":"_TZ3210_wuhzzfqg","modelid":"TS0202","name":"LightLevel 29","swversion":null,"type":"ZHALightLevel","uniqueid":"8c:f6:81:ff:fe:bf:9b:6a-03-0400"},"e":"changed","id":"29","r":"sensors","t":"event","uniqueid":"8c:f6:81:ff:fe:bf:9b:6a-03-0400"} (ret = 332)
17:18:23:216 Websocket 192.168.8.147:50624 send message: {"attr":{"id":"29","lastannounced":null,"lastseen":"2023-05-08T16:18Z","manufacturername":"_TZ3210_wuhzzfqg","modelid":"TS0202","name":"LightLevel 29","swversion":null,"type":"ZHALightLevel","uniqueid":"8c:f6:81:ff:fe:bf:9b:6a-03-0400"},"e":"changed","id":"29","r":"sensors","t":"event","uniqueid":"8c:f6:81:ff:fe:bf:9b:6a-03-0400"} (ret = 332)
17:18:23:217 Websocket 192.168.8.147:50621 send message: {"attr":{"id":"27","lastannounced":null,"lastseen":"2023-05-08T16:18Z","manufacturername":"_TZ3210_wuhzzfqg","modelid":"TS0202","name":"Temperature 27","swversion":"2.2.1","type":"ZHATemperature","uniqueid":"8c:f6:81:ff:fe:bf:9b:6a-01-0402"},"e":"changed","id":"27","r":"sensors","t":"event","uniqueid":"8c:f6:81:ff:fe:bf:9b:6a-01-0402"} (ret = 337)
17:18:23:217 Websocket 192.168.8.147:50624 send message: {"attr":{"id":"27","lastannounced":null,"lastseen":"2023-05-08T16:18Z","manufacturername":"_TZ3210_wuhzzfqg","modelid":"TS0202","name":"Temperature 27","swversion":"2.2.1","type":"ZHATemperature","uniqueid":"8c:f6:81:ff:fe:bf:9b:6a-01-0402"},"e":"changed","id":"27","r":"sensors","t":"event","uniqueid":"8c:f6:81:ff:fe:bf:9b:6a-01-0402"} (ret = 337)
17:18:23:218 Websocket 192.168.8.147:50621 send message: {"attr":{"id":"28","lastannounced":null,"lastseen":"2023-05-08T16:18Z","manufacturername":"_TZ3210_wuhzzfqg","modelid":"TS0202","name":"Humidity 28","swversion":"2.2.1","type":"ZHAHumidity","uniqueid":"8c:f6:81:ff:fe:bf:9b:6a-01-0405"},"e":"changed","id":"28","r":"sensors","t":"event","uniqueid":"8c:f6:81:ff:fe:bf:9b:6a-01-0405"} (ret = 331)
17:18:23:219 Websocket 192.168.8.147:50624 send message: {"attr":{"id":"28","lastannounced":null,"lastseen":"2023-05-08T16:18Z","manufacturername":"_TZ3210_wuhzzfqg","modelid":"TS0202","name":"Humidity 28","swversion":"2.2.1","type":"ZHAHumidity","uniqueid":"8c:f6:81:ff:fe:bf:9b:6a-01-0405"},"e":"changed","id":"28","r":"sensors","t":"event","uniqueid":"8c:f6:81:ff:fe:bf:9b:6a-01-0405"} (ret = 331)
17:18:23:544 poll node 70:ac:08:ff:fe:45:ca:3e-01
17:18:23:545 Poll light node UGF UFH 3 Kitchen 2
17:18:23:545 Idle timer triggered
17:18:23:545 Force binding of attribute reporting for node TS0202
17:18:23:827 Poll APS request to 0x70AC08FFFE45CA3E cluster: 0x0006 dropped, values are fresh enough

Smanar, 2023-05-09:
There is no zigbee request about "interfacemode", there is only something for battery and error message about "attr/swversion", the second one is easy to correct, in wich one folder have you put the DDF ? (I will use a relative path for the script)

But I can see `"interfacemode":0,"` was not true or false, this one use a value, so can you try  with 0 and 1 ?

pocipocster, 2023-05-09:
I'll ignore "name": "attr/swversion" issue for now.

in log just "info" is set.

But nothing comes up when I do the PUT REST request Curl. I am probably doing something wrong!!!

[image]

pocipocster, 2023-05-09:
This is all I get

[image]

Smanar, 2023-05-09:
You forget the body, or there is a problem on the application you are using.

`{"interfacemode":1}`

pocipocster, 2023-05-09:
Sorry I have no idea anymore. Postman says it has imported the curl command - but I don't think it has.
[image]

pocipocster, 2023-05-10:
As it doesn't import the curl properly ( this in itself is an issue ) you'll need to help me and tell me what exactly goes in the body/header and anything else.

Smanar, 2023-05-10:
Select the "body" tab, inside you need to have selected (for exemple) "raw" and in the body part `{"interfacemode":1}`

pocipocster, 2023-05-11:
This works! ; not true or false just 1 or 0

But how does this help with the original problem?????

[image]

Smanar, 2023-05-11:
If I m right if "interfacemode" is set to 1, it will enable the reporting, so you will have value for temperature and humidity.

pocipocster, 2023-05-11:
Nope !
I assume if I delete and add the device is no good ? Interfacemode default to 0

pocipocster, 2023-05-11:
[image]

Smanar, 2023-05-12:
Ok so here I don't understand, the request is fine (dpid = 0x70, type 0x10, value 1). It the same I m doing for the other device.

Perhaps it's because I have put `"read": {"fn": "none"},` for both fonctions. Can you try to change one, for exemple the one for temperature by `"read": {"fn": "tuya"},`.

Or the request was not listen by the device because it was in sleeping mode ? On the previous code we are making the request during the inclusion.

pocipocster, 2023-05-12:
I change temperature in DDF to as you said.
I re add device and set interface mode to 1

Still no temperature reading...

[image]

Smanar, 2023-05-12:
Sorry, I haven't more idea. The temperature and humidity is still null or 0 ?

You haven't a zigbee error for this device after having in log the line "witetuyadata" ? There is a way to force a request during inclusion using DDF (with setting a defaut value), but it have never worked for me.

pocipocster, 2023-05-12:
Hey

We had a good try !
I started with knowledge of 0 ! ; now have a bit more than 0 😁
Please let me post you the item ( no need to return ) .PM me details . I have about 15 of these to install ! Temperature and presence were the most important- as I use them to control heating !!

Smanar, 2023-05-12:
Where are you from ? You can contact me on discord (Smanar too, PM are possible on github ?).
But I m not understand what can be the problem, I know the DDF core is not working same but here ....

You have tried with the "device awekening" ?

pocipocster, 2023-05-12:
I’m in U.K.
Device is usb powered so always awake ( I don’t like battery )
Sorry what is discord ! 🤣 - everything new too me

pocipocster, 2023-05-12:
I have to learn so much!!! LOL!
Discord says I need your 4 digit tag code???! - then I can send a friend request.
Might be easier I post my email address here! - far simpler; am I allowed to do that!????

Smanar, 2023-05-12:
>Device is usb powered so always awake ( I don’t like battery )

Ha ? So it's perhaps not a sleeper, can you edit in the DDF

`  "sleeper": true,`

And set it to false ?

>Might be easier I post my email address here! - far simpler; am I allowed to do that!????

IDK but will be more "private" on discord. On my side I just use the online web application, you can find me on the deconz server or the digit code are "Smanar#0359"

[14 more replies]

closed 2023-05-31 by Smanar