dresden-elektronik/deconz-rest-plugin #6918: Device name

Sjonnie2018, 2023-04-20:
Hi All,

Received the new deCONZ Beta V2.22.0 container today and tested the MOES 3 Gang Switch (TZ3000_w3c7ouru) within deCONZ and Domoticz. The good news is that the switch is seen as a 3 gang switch (4 gang icon in Phoscon but that is probably unrelated) and single click and double click works as expected. I only have a hard time to get the long click (press ;-) ) recognized.

I had the following results in Domoticz:

Left button:
Single click -> value 10
Double click -> value 30
Long click -> no response

Middle button:
Single click -> value 40
Double click -> value 60
Long click -> no response

Right button:
Single click -> value 70
Double click -> value 90
Long click -> no rsponse.

I was expecting values of 20, 50 and 80 respectively.

Is there something I can do to make long click functional?

Note that I am using deCONZ in a container and it seems to be impossible to create a button_maps.json for this device.

Any help appreciated!

## Device

- 
    Product name: MOES Wireless Remote Switch
    Manufacturer: _TZ3000_w3c7ouru
    Model identifier: TS0043
    Device type : 3 gang switch
    Remote: remote

## Screenshots
Sorry that I made screenshots that are not following guidelines but I think you will see the information required. All screenshots were made after pressing the appropriate button on the switch.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Mimiix, 2023-04-20:
Please proceed in the original issue. This scatters the info more.

Closing this.

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6724

closed 2023-04-20 by Mimiix

Sjonnie2018, 2023-04-20:
I will transfer the information to the original thread tomorrow. Sorry for any inconvenience I caused.