dresden-elektronik/deconz-rest-plugin #6921: Moes Dual Gang switch

pocipocster, 2023-04-21:
## Device

- Product name: Smart switch modules - Zigbee only ( not RF )
- Manufacturer: _TZ3000_pmz6mjyu
- Model identifier: TS011F
- Device type : Please remove all unrelated device types. 

  - Switch dual gang
 

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
[image]
[image]
[image]
[image]
[image]

pocipocster, 2023-04-21:
Shows in Deconz but not found in Phoscon.

pocipocster, 2023-04-21:
Correction! - this may work and shows up in Phoscon as a light rather than a switch.....

pocipocster, 2023-04-21:
Phoscon shows it as 2 separate switches (is this correct for 2 gang?). Deconz just shows 1 switch.
What should it be?

Smanar, 2023-04-21:
Yep, deconz show node, so only 1 by device
Phoson show API entry, so 1 by gangs (and more if you have too power measurement for exemple).

And wired switch are "light", they receive order like a bulb, battery switch can only send order.

pocipocster, 2023-04-24:
> Regarding to previews post is there still an issue with this switch ? Does it works, each switch separatly ?
> If not could perhaps try this DDF
>
> {
>         "schema": "devcap1.schema.json",
>         "manufacturername": ["_TZ3000_pmz6mjyu", "_TZ3000_zmy1waw6"],
>         "modelid": ["TS011F", "TS011F"],
>         "vendor": "Moes",
>         "product": "MS-104BZ Wired Switch 2 gangs",
>         "sleeper": false,
>         "status": "Gold",
>         "subdevices": [{
>                         "type": "$TYPE_ON_OFF_LIGHT",
>                         "restapi": "/lights",
>                         "uuid": [
>                                 "$address.ext",
>                                 "0x01"
>                         ],
>                         "items": [{
>                                         "name": "attr/id"
>                                 },
>                                 {
>                                         "name": "attr/lastannounced"
>                                 },
>                                 {
> [... 119 lines cut ...]
>                                 "max": 1800
>                         }]
>                 }
>         ]
> }

Hi

The switch seems to work fine without this schema . Naming of zigbee modules in phoscon seems a bit strange . The name sometimes gets altered to “ on/off switch “ no matter what I named it .

BabaIsYou, 2023-04-24:
Are you able to test the DDF anyway ? If not then just close the issue if all is ok for you. Thx.

pocipocster, 2023-04-24:
I can test this if you wish.
Should i see any differences???

cheers

pocipocster, 2023-04-24:
> You shouldn't ! But that make the switch picked up by DDF and not depending of legacy code ... always better.
> In the worst case scenario you just delete the created DDF and restart DeConz and that's it !

Ok. I will test this DDF later 👍🏻

Tosko4, 2023-05-06:
+1 for me 👍🏻

PR #6959 (2023-05-16): DDF add support for Moes MS-104BZ
Fix #6921
commit f2cc70e4 (2023-05-16): DDF for Moes smart switch MS-104BZ (#6959)

Fix #6921
closed 2023-05-16 by manup