dresden-elektronik/deconz-rest-plugin #6922: Tesla Smart Sensor Temperature and Humidity

kocicky, 2023-04-21:
## Device

- Product name: Smart Sensor Temperature and Humidity
- Manufacturer: Tesla Global Limited
- Model identifier: TSL-SEN-TAH
- Device type : Please remove all unrelated device types. 
  - Sensor

## Screenshots
[image]
[image]

### Basic
[image]

### Identify
[image]

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
[image]
[image]

https://www.teslasmart.com/tesla-smart-sensor-temperature-and-humidity

BabaIsYou, 2023-04-21:
Can you please stick to the template ? Add clusters list screenshot, click on "read" button before taking screenshot, etc ... Thx

kocicky, 2023-04-21:
Sorry, corrected.

BabaIsYou, 2023-04-21:
Could you try this DDF ?
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_j5fbnjeh",
  "modelid": "TS0201",
  "vendor": "Tesla Global Limited",
  "product": "TSL-SEN-TAH",
  "sleeper": true,
  "status": "Silver",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "awake": true,
          "default": 0
        }
      ]
    },
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          },
          "default": 0
        },
        {
          "name": "config/offset"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 3600,
          "max": 14400,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    }
  ]
}
```

kocicky, 2023-04-21:
Thanks, but temperature doesn't seem right
[image]

BabaIsYou, 2023-04-22:
OK. Even if you force reading of cluster 0x0402 in Deconz-GUI this value stay to zero ?

BabaIsYou, 2023-04-23:
@kocicky, Can you confirm that is ok after the changes ?

kocicky, 2023-04-23:
It seems allright now. But I cannot see the sensor entity in Phoscon-GW in Home Assistant.

kocicky, 2023-04-23:
Sorry, after core update sensor is on. Thanks for support.

PR #6924 (2023-05-21): DDF for Tesla TSL-SEN-TAH multi sensor
Fix #6922
commit 4edce2b1 (2023-05-21): DDF for Tesla TSL-SEN-TAH multi sensor (#6924)

Fix #6922
closed 2023-04-23 by kocicky