dresden-elektronik/deconz-rest-plugin #6923: Salus PS600 Pipe Temperatur not in Phoscon App and API

sansonifabio, 2023-04-22:
## Describe the bug

  The Salus Pipe PS 600 temperature sensor has been supported since version 2.22.00. I can pairing it via the Phoscon App, but it is not displayed. Not in the Phoscon App and also in the Api. I only see it as a tile in Deconz

## Steps to reproduce the behavior

1. Install version 2.22.00
2. Start teach sensor.
3. Put the Palus PS600 in pairing.
4. Then look in deconz

## Screenshots
  
[image]
[image]
[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian 
 - Firmware version: 26780700
 - deCONZ version: 2.22.00 
 - Device: RaspBee

## deCONZ Logs

## Additional context

  DDF
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Computime",
  "modelid": "PS600",
  "product": "PS600",
  "sleeper": false,
  "status": "Draft",
  "subdevices": []
}

BabaIsYou, 2023-04-22:
As you can see in the DDF file recently released it is in "silver" DDF level because has not been tested by a lot of users (https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6882/files#diff-bd4add2588d69dd5ed080f741056351d66693f47e3346222915d88f36dd9aa7f)

Then you have to enable Silver level DDF in DeCONZ : https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-change-DDF-modes-and-enable-or-disable-DDF-levels

Mimiix, 2023-04-22:
As this is not a bug as described in #5113 i am closing this issue.

Feel free to open an topic on the forums.

Kind regards
Dennis

closed 2023-04-22 by Mimiix

sansonifabio, 2023-04-23:
Hello [BabaIsYou](https://github.com/BabaIsYou) thanks but I have absolutely no idea about DDF etc. The only thing that counts for me is that the Salus was added to the devices and it still doesn't work. Thanks for your help.

**I also think it's a shame that it's just being closed here.**
Greetings
Fabio

BabaIsYou, 2023-04-23:
Hi,

it’s not a big but a parameter that have to be set in your configuration as explain in the link I referenced. That’s why, according to rules of this GitHub, and because solution has been proposed, this issue was closed.