dresden-elektronik/deconz-rest-plugin #6925: LoraTap Tuya ZigBee 3.0 Wireless 2 Gang Remote Control Switch

sanluca, 2023-04-23:
Hello,
I have a problem with these remote controls, at the beginning they had a problem that after a day the battery drained immediately, but on openhab I was able to see the battery status, now the battery problem has been solved, which does not drain immediately after a day, the remote control has been working for two weeks, but on openhab I can no longer view the battery status, what could it be? Thank you

BabaIsYou, 2023-04-23:
> what could it be?

So many things ! What was the laste reported value for battery ? Is your device still reacheable ? seen by controller ? Does it work when pressing on buttons ?

Mimiix, 2023-04-23:
Hi,

Please use the template.

sanluca, 2023-04-23:
yes the works, I would like to review the state of the battery as before to understand how long the button lasts.
what model should i use?
Thank you

closed 2023-05-22 by github-actions[bot]