dresden-elektronik/deconz-rest-plugin #6930: deCONZ  GUI shows Raspbee2 MAC addres, Homebridge deCONZ does not see it

brickelbrit, 2023-04-27:
## Describe the bug
The Homebridge deCONZ plugin spits out below error:
[4/27/2023, 9:41:21 AM] Homebridge v1.6.0 (HAP v0.11.0) (Homebridge 3D7D) is running on port 51154.
[4/27/2023, 9:41:21 AM] [deCONZ] hardware: Raspberry Pi 4B 1.5 (4GB)
[4/27/2023, 9:41:21 AM] [deCONZ] os: Raspbian GNU/Linux 10 (buster)
[4/27/2023, 9:41:21 AM] [deCONZ] ui server: listening on http://127.0.0.1:39265/
[4/27/2023, 9:41:22 AM] [deCONZ] warning: 192.168.0.128:80: 0000000000000000: not a RaspBee/ConBee mac address - retrying in 60s

## Steps to reproduce the behavior
None.
Reboot with sudo GCFFlasher_internal -r seems to have done the trick once briefly, but issues is back and won't leave.

## Expected behavior
deCONZ to recognize MAC address of Raspbee2

## Screenshots
[image]

## Environment
 - Host system: Raspberry Pi 4 4GB
 - Running method: deCONZ Raspbian Buster Desktop stable image  
 - Firmware version: 26780700
 - deCONZ version: 2.22.0
 - Device: RaspBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs
APS,ZDP, ZCL and INFO:
https://pastebin.com/fvMPEdW6

## Additional context
Currently only three exterior Philips Hue lamps (no Philips Bridge) are connected to the network. Can control everything easily through the Phoscon Web interface or e.g. Hue Essentials. Just cannot get them into Homebridge and HomeKit. Planning to get several more Ikea bulbs.

BabaIsYou, 2023-04-27:
Can you just force reading of Basic cluster for your Raspbee2 to see if it's changes something ?
[image]
Or just clicking on F5 while selecting youre Raspbee2 device in GUI ?

brickelbrit, 2023-04-27:
[image]
This is what I see. I might be doing something wrong though. Is something noticeable supposed to happen when I click the Raspbee2 in the GUI while holding F5?

I still get the same error message every 60s in the hombridge logs.

ebaauw, 2023-04-27:
Homebridge deCONZ gets the mac address from the API (`/config/bridgeid`).  I don't know what causes the API not to expose the mac address, but it's been suggested it could be related to the _deCONZ Network Settings_ being wrong.

brickelbrit, 2023-04-27:
OK, thanks for the feedback.
Using the device with different software would mean a different plugin? How would I be able to run a different software through the Raspbee2?

ebaauw, 2023-04-27:
Different software than deCONZ, interacting with the device over the serial protocol.  Again I have no first-hand experience with this situation.  Best check the network settings.

Mimiix, 2023-04-27:
Can you share a screen of the network settings

brickelbrit, 2023-04-27:
[image]

ebaauw, 2023-04-27:
The _NWK Ext PAN ID_ should normally be the coordinator's mac address, and the _APS Use Ext PAN ID_ usually is all zeroes.  It's probably enough to delete the _APS Use Ext PAN ID_, for _NWK Ext PAN ID_ to reset itself.  If not, copy the coordinator's mac address to _APS Use Ext PAN ID_.  Make sure to _Save_ and _Leave_/_Join_ the network for the settings to become active.

I think the API should report the correct `bridgeid` after the _Leave_/_Join_.  If not, you might need to restart deCONZ.

Mimiix, 2023-04-27:
Hi,

As this is not a bug report as described in #5113, i am closing this issue.

You indeed need to replace the ext pan id to all 0 .

brickelbrit, 2023-04-28:
Thanks for all the support, regardless. This solved my problem!

Mimiix, 2023-04-28:
Nice!