dresden-elektronik/deconz-rest-plugin #6934: Vendor_TZE204_cjbofhxw  ModelTS0601 - Ampermeter from Matsee

sajikur, 2023-04-28:
## 
Ampermeter from Matsee Model TS0601

- Product name: Ampermeter from Matsee.
- Manufacturer: Vendor_TZE204_cjbofhxw .
- Model identifier: ModelTS0601.
- Device type : Please remove all unrelated device types. 
  - Other: Amperometer

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2023-04-28:
Missing screeenshots. Please add all.

sajikur, 2023-04-28:
Hi, thank you for answer. Could you tell me what is the panel you mean?

Mimiix, 2023-04-28:
The list of clusters under the node on the gui.

sajikur, 2023-04-28:
It's correct?

Mimiix, 2023-04-28:
Now we are still missing screens for groups and scenes cluster

sajikur, 2023-04-28:
well, It's all? Sorry for my little experience in this kind of request

Mimiix, 2023-04-28:
All good , thanks :)

sajikur, 2023-04-28:
> All good , thanks :)

Thank you!

sajikur, 2023-05-12:
hi, any news?

Mimiix, 2023-05-12:
No dev has picked it up, so no news

BabaIsYou, 2023-05-16:
May be could try this DDF. Not quite sure about consumption BTW

```
{
	"schema": "devcap1.schema.json",
	"manufacturername": "_TZE204_cjbofhxw",
	"modelid": "TS0601",
	"product": "Ampermeter from Matsee plus",
	"sleeper": false,
	"status": "Gold",
	"subdevices": [{
			"type": "$TYPE_CONSUMPTION_SENSOR",
			"restapi": "/sensors",
			"uuid": [
				"$address.ext",
				"0x01",
				"0x0702"
			],
			"items": [{
					"name": "attr/id"
				},
				{
					"name": "attr/lastannounced"
				},
				{
					"name": "attr/lastseen"
				},
				{
					"name": "attr/manufacturername"
				},
				{
					"name": "attr/modelid"
				},
				{
					"name": "attr/name"
				},
				{
					"name": "attr/swversion",
					"parse": {
						"fn": "zcl",
						"ep": 1,
						"cl": "0x0000",
						"at": "0x0001",
						"script": "tuya_swversion.js"
					},
					"read": {
						"fn": "zcl",
						"ep": 1,
						"cl": "0x0000",
						"at": "0x0001"
					}
				},
				{
					"name": "attr/type"
				},
				{
					"name": "attr/uniqueid"
				},
				{
					"name": "config/offset",
					"default": 0
				},
				{
					"name": "config/on"
				},
				{
					"name": "config/tuya_unlock"
				},
				{
					"name": "config/reachable"
				},
				{
					"name": "state/consumption",
					"parse": {
						"fn": "tuya",
						"dpid": 101,
						"eval": "Item.val = Attr.val / 1000;"
					},
					"read": {
						"fn": "none"
					},
					"default": 0
				},
				{
					"name": "state/lastupdated"
				}
			]
		},
		{
			"type": "$TYPE_POWER_SENSOR",
			"restapi": "/sensors",
			"uuid": [
				"$address.ext",
				"0x01",
				"0x0b04"
			],
			"items": [{
					"name": "attr/id"
				},
				{
					"name": "attr/lastannounced"
				},
				{
					"name": "attr/lastseen"
				},
				{
					"name": "attr/manufacturername"
				},
				{
					"name": "attr/modelid"
				},
				{
					"name": "attr/name"
				},
				{
					"name": "attr/swversion",
					"parse": {
						"fn": "zcl",
						"ep": 1,
						"cl": "0x0000",
						"at": "0x0001",
						"script": "tuya_swversion.js"
					},
					"read": {
						"fn": "none"
					}
				},
				{
					"name": "attr/type"
				},
				{
					"name": "attr/uniqueid"
				},
				{
					"name": "config/offset",
					"default": 0
				},
				{
					"name": "config/on"
				},
				{
					"name": "config/reachable"
				},
				{
					"name": "state/lastupdated"
				},
				{
					"name": "state/current",
					"parse": {
						"fn": "tuya",
						"dpid": 18,
						"eval": "Item.val = Attr.val / 1000;"
					},
					"read": {
						"fn": "none"
					},
					"default": 0
				},
				{
					"name": "state/voltage",
					"parse": {
						"fn": "tuya",
						"dpid": 20,
						"eval": "Item.val = Attr.val / 10;"
					},
					"read": {
						"fn": "none"
					},
					"default": 0
				},
				{
					"name": "state/power",
					"parse": {
						"fn": "tuya",
						"dpid": 19,
						"eval": "Item.val = Attr.val / 10;"
					},
					"read": {
						"fn": "none"
					},
					"default": 0
				}
			]
		}
	]
}
```

BabaIsYou, 2023-05-22:
> hi, any news?

@sajikur any news from your side ?

sajikur, 2023-05-23:
> > hi, any news?
> 
> @sajikur any news from your side ?

Unfortunately no new news.  Too bad because the device is really very cheap compared to shelly and it's zigbee

sajikur, 2023-05-23:
> Did you give a try to the provided DDF

sorry for my incompetence. Could you explain me your post?
thank you

BabaIsYou, 2023-05-23:
I think you could begin with this : https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

sajikur, 2023-05-23:
> https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

thank you, I'll try

BabaIsYou, 2023-06-14:
Saw it already, but as sajikur has naver mentionned a pairing issue, nor tested the DDF ... issue was staled ...

haegar33, 2023-08-08:
So the conclusion is: there is no solution for this device and no working deConz implementation? Its a wide spread energy meter on Aliexpress now. In my NodeRed/Deconz setup its quickly recognized, but only as power plug. I tried the DDF mentioned above but no differenz 
[image]

pdcmoreira, 2023-08-08:
@haegar33 For now, I guess there is no working solution and I also stopped trying to make it work. Might try again soon.

Mimiix, 2023-08-08:
> So the conclusion is: there is no solution for this device and no working deConz implementation? Its a wide spread energy meter on Aliexpress now. In my NodeRed/Deconz setup its quickly recognized, but only as power plug. I tried the DDF mentioned above but no differenz [image]

The proposed DDF was never tested.

While testing the DDF, you should see things in the rest api .

pdcmoreira, 2023-08-08:
Hey @Mimiix what do you mean by this?
>While testing the DDF, you should see things in the rest api .

Mimiix, 2023-08-08:
The working of the DDF can only be determined by checking the rest api or websocket. Phoscon probably isn't showing anything as it needs to add support on its own.

Mimiix, 2023-08-08:
This repo only provides support until the api itself .

haegar33, 2023-08-08:
ok applying the DDF gets a modified REST-api response, but this does not imply that the device is providing values with the defined "attributes" I see that @pdcmoreira and others have tried without success. 
Looks like this is the first China device not working with my setup ;(
Thanks anyway! Haegar

haegar33, 2023-08-09:
Looking at a similar(?) device discussed here #7151 I changed the DDF entry for ''state/voltage'' as suggested and I get at least a reasonable answer back via REST:
```
   "lastannounced": "2023-08-09T09:12:33Z",
    "lastseen": "2023-08-09T10:21Z",
    "manufacturername": "_TZE204_cjbofhxw",
    "modelid": "TS0601",
    "name": "Power 26",
    "state": {
        "current": 0,
        "lastupdated": "2023-08-09T09:14:29.603",
        "power": 0,
        "voltage": 234
    },
    "type": "ZHAPower",
    "uniqueid": "18:7a:3e:ff:fe:cb:0a:21-01-0b04"
}
```

Smanar, 2023-08-09:
If you have the GUI, you can take a look in logs with flag "info" and "ddf", you will return from the device (even not managed by deconz)

haegar33, 2023-08-10:
I have changed:
```
        {
          "name": "state/voltage",
          "parse": {"fn": "tuya", "dpid": 6, "eval": "Item.val = (Attr.val[0] << 8) | Attr.val[1]) / 10;" },
          "read": {"fn": "none"},
          "default": 0
        },
```
thus reading from dpid 6. As I am not an expert in javascript bitfield decodings I could not get current or power. I will try the deCONZ GUI log file

Smanar, 2023-08-10:
Hu ?
I have checked again, and the model _TZE204_cjbofhxw don't use the dpid 6 but the 20 for voltage ....

Yeah pls, on deconz logs you will be able to see request from the device with "info", and the dpid used by it.

pdcmoreira, 2023-09-01:
Please keep this open.

closed 2023-09-30 by github-actions[bot]