dresden-elektronik/deconz-rest-plugin #6935: v2.22.0-beta: no websocket, no device names

cs42, 2023-04-29:
## Describe the bug

1. deConz does only listen on http port, no websocket, no UDP, so connection from home assistant is not possible. Output of `lsof -i -n -P`:
```
deCONZ    2210              ha    7u  IPv4  39020      0t0  TCP *:42080 (LISTEN)
```
2. the GUI doesn't display device names, just HEX numbers: see screenshot below

## Steps to reproduce the behavior

1. install beta 2.22.0 from official repository

## Expected behavior

1. output of `lsof -i -n -P` should be (and is on 2.12.2):
```
deCONZ     67304              ha    7u  IPv4  5561285      0t0  TCP *:42080 (LISTEN)
deCONZ     67304              ha   11u  IPv4  5561295      0t0  TCP *:42443 (LISTEN)
deCONZ     67304              ha   12u  IPv4  5562438      0t0  TCP 10.50.0.5:42443->10.50.0.5:39514 (ESTABLISHED)
deCONZ     67304              ha   13u  IPv4  5562440      0t0  UDP *:1900
```
2. The GUI on 2.12.2 shows device names: see screenshot below

## Screenshots

- 2.22.0-beta: 
[image]

- 2.12.2: 
[image]

## Environment

 - Host system: PC (thin client)
 - Running method: Ubuntu 
 - Firmware version: 26780700
 - deCONZ version: 2.22.0-beta
 - Device: ConBee II 
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system?: no
 - command to run deconz: `/bin/sh /usr/bin/xvfb-run -n 10 -s -screen 0 1920x1080x24 -f (redacted PATH_TO_XAUTH) -a deCONZ --http-port=42080 --ws-port=42443 --auto-connect=1`

## deCONZ Logs

## Additional context

Mimiix, 2023-04-29:
Hi

I am tempted to close this issue as it doesn't comply to #5113, additionally as this looks to be your installation only.

@SwoopX perhaps you can comment ?

cs42, 2023-04-29:
Oh, sorry that I didn't mention that: I have two systems with this problem!

Smanar, 2023-04-29:
Somes logs can be usefull, this can happen on bad OS configuration, by curiosity you have same result with a "lighter" command line to run deconz ?

cs42, 2023-04-29:
Unfortunately no output on the command line. I tried running `deCONZ --platform minimal`

The `lsof -i -n -P` show:

- on 2.22.0-beta: `deCONZ    5919            root    4u  IPv4 202247      0t0  TCP *:8080 (LISTEN)`
- on 2.21.2 after removing beta and reinstalling:
```
deCONZ    6781            root    4u  IPv4 210661      0t0  TCP *:8080 (LISTEN)
deCONZ    6781            root    9u  IPv4 211326      0t0  TCP *:443 (LISTEN)
deCONZ    6781            root   10u  IPv4 210698      0t0  UDP *:1900
````
- on 2.21.2 (the other system, port 443 in use): 
```
deCONZ     208991            root    4u  IPv4 121209809      0t0  TCP *:8080 (LISTEN)
deCONZ     208991            root   10u  IPv4 121218186      0t0  TCP *:8088 (LISTEN)
deCONZ     208991            root   11u  IPv4 121218226      0t0  UDP *:1900
```

Output on command line starting 2.22.0-beta is the same as for 2.21.2:
```
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'                                                    │hass      1074              ha   94u  IPv4  30150      0t0  TCP 10.50.0.5:50538->18.198.222.235:443 (CLOSE_WAIT)
libpng warning: iCCP: known incorrect sRGB profile                                                                            │hass      1074              ha   98u  IPv4  30152      0t0  TCP 10.50.0.5:42156->10.50.0.25:8009 (ESTABLISHED)
libpng warning: iCCP: known incorrect sRGB profile                                                                            │hass      1074              ha  108u  IPv4  29096      0t0  TCP 10.50.0.5:46542->10.50.0.25:32010 (ESTABLISHED)
This plugin does not support propagateSizeHints()                                                                             │hass      1074              ha  113u  IPv4  29099      0t0  TCP 10.50.0.5:43108->10.50.0.25:32206 (ESTABLISHED)
This plugin does not support propagateSizeHints()                                                                             │hass      1074              ha  115u  IPv4  29103      0t0  TCP 10.50.0.5:55810->10.50.0.29:8009 (ESTABLISHED)
This plugin does not support propagateSizeHints()
```

So I do not thinks it's the port selection or running through Xvfb. 

BTW, both systems are: Ubuntu 22.04.2 LTS

What other logs could I create to help?

Mimiix, 2023-04-29:
The devices showing up with hex names indicate that the zigbee network config has issues. 

We would like logs from deconz with log levels:
aps
Aps_l2, 
error, 
error_l2
Info,
Info_l2

cs42, 2023-04-29:
No need to upload the logs. Using the log output I found the reason myself:

```
11:24:33:658 error loading plugin: Cannot load library /usr/share/deCONZ/plugins/libde_rest_plugin.so: (libcrypto.so.1.1: cannot open shared object file: No such file or directory)
```

Ubuntu 22.04 does not provide `libssl1` and as it seems older versions of deCONZ didn't need it (or provided it). After installing from `http://archive.ubuntu.com/ubuntu/pool/main/o/openssl/libssl1.1_1.1.0g-2ubuntu4_amd64.deb` it works as expected.

So it may be considered a "bug" for systems that do not provide `libcrypto.so.1.1` any longer.

Mimiix, 2023-04-29:
That's interesting indeed. Good find!

I keep it open as a bug and ask manup to check .

SwoopX, 2023-04-29:
Tbh, I was about to close this one, as my dev environment is Ubuntu 22.04 and there, everything works perfectly as well as on my prod environment. However, I'm using the plugin code as it was right before the beta came out (compiling myself) so there must have changed anything in between 🤔

Hypfer, 2023-05-18:
Can confirm that both issues reported also happened here with `2.22.00-ubuntu-bionic-beta`.
Switching to the non-beta `2.21.02-ubuntu-bionic-stable` brought the system back into a working state.

My host OS is Debian Bookworm

CDRX2, 2023-05-22:
I can confirm the same behaviour, running Ubuntu as an OS. Was unable to do much testing, however. It's my production environment and I had to prioritize getting back to a working state.

manup, 2023-05-30:
> 11:24:33:658 error loading plugin: Cannot load library /usr/share/deCONZ/plugins/libde_rest_plugin.so: (libcrypto.so.1.1: cannot open shared object file: No such file or directory)

I think this is due the hard linking of libcrypto added in 2.22.0. The deCONZ and plugin binaries are compiled on older OS versions, normally since that it upwards compatible. But it fails when linked against the full version like `libcrypto.so.1.1` instead of the SO_NAME `libcrypto.so.1` which should be a symlink to whatever is installed.

Can be checked with:
```
ldd /usr/bin/deCONZ
```

[image]

PR #6913 contains code to replace hard linking of libcrypto and also handles the case when the library isn't available.
I'm currently trying to figure out why the MMO hash in the PR doesn't work on Windows, and after that I'll spin up a few Ubuntu versions in Virtualbox to check that dynamic loading works.

closed 2023-06-03 by manup