dresden-elektronik/deconz-rest-plugin #6937: After a sudo apt update && sudo apt full-upgrade and a reboot deconz no longer connects to the conbee 2.

Nastras, 2023-05-01:
## Describe the bug

After a sudo apt update && sudo apt full-upgrade and a reboot deconz no longer connects to the conbee 2.
The problem exists has occurred on three different Raspberry Pi with deconz.

Raspberry Pi 4 Model B Rev 1.1
Debian GNU/Linux 11 (bullseye)
Linux 6.1.21-v8+
deconz 2.21.02

The following packages were installed:
[image]

Idaho947, 2023-05-01:
Hello,

Same on debian 11 with deconz 2.21.02 and ConBeeII_0x26780700.bin.GCF.
Only difference it doesn't freez and I can close the guid windows.

On a windows serveur and debian 10 serveur the key is working. I update the debain 10 serveur to the debian 11 of yesterday and I've the issue.

ANierbeck, 2023-05-01:
Hi, 

on a Raspberry Pi 4 Model B Rev 1.4 same thing.
system itself finds the conbee2 stick: 

> [    2.303833] usb 1-1.2: new full-speed USB device number 3 using xhci_hcd
[    2.412978] usb 1-1.2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[    2.413040] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.413068] usb 1-1.2: Product: ConBee II
[    2.413090] usb 1-1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[    2.413114] usb 1-1.2: SerialNumber: DE2256881

The phoscon gateway shows no found USB stick

[image]

also the gui doesn't show a single connectable device: 

[image]

Mimiix, 2023-05-01:
I've reached out to Manup and some others at DE to check this quickly.

guitardoc, 2023-05-01:
There is a public holiday today in Germany so maybe there will be a delayed response.

BabaIsYou, 2023-05-01:
May be useless but did you try `sudo raspi-config` and update ?

Idaho947, 2023-05-01:
Do you have the same for a debian virtual machine ?

BabaIsYou, 2023-05-01:
> Do you have the same for a debian virtual machine ?

Nope ! But your remark leads me to think then that  this issue is  not Rpi related (just Raspbian) but Debian.

Idaho947, 2023-05-01:
Yes I think too. It is a debian issue.

BabaIsYou, 2023-05-01:
what does dmesg -T say when you plug in the adapter?

Idaho947, 2023-05-01:
After a the key is see :
[image]

Idaho947, 2023-05-01:
Here are the result :
[image]

BabaIsYou, 2023-05-01:
Groups seems correct. And do you have trouble to have `ls -la /dev/serial/by-id/usb-*` ?

Idaho947, 2023-05-01:
Yes it doesn t work :
[image]

Siggi0904, 2023-05-01:
looks like the same error at iobroker community and other smart home forums.

It's an debian error, Symlinks are not set after upgrade.
See https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1035094

Thanks.

ebaauw, 2023-05-01:
In that case, you should be able to specify the device file for the ConBee on the command line, like: `deCONZ --dev=/dev/ttyACM0`

ebaauw, 2023-05-01:
> Can I include this somehow in the service file?

Yes, you could add the parameter to the `ExecStart` line in `/lib/systemd/system/deconz-gui.service`.  However that will get overwritten when deCONZ is updated.

I've created an override, so my changes survive an update of deCONZ.
`/etc/systemd/system/deconz-gui.service.d/override.conf `:
```
[Service]
EnvironmentFile=/etc/default/deconz
ExecStart=
ExecStart=/usr/bin/deCONZ $DECONZ_OPTS
ExecStartPre=/usr/bin/deCONZ-wait-for-X.sh
```
And then specify the command-line options in `/etc/default/deconz`:
```
DECONZ_OPTS="--http-port=80 --dbg-ddf=1 --dbg-error=2 --dbg-js=1"
```
This is my dev/test system, currently debugging the DDFs and JavaScript.

`deCONZ-wait-for-X.sh` prevents deCONZ crashing on reboot, when the X server is not yet up, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2482.

guitardoc, 2023-05-01:
How would you do it on a headless system?
Thanks.

x-MaxG-x, 2023-05-01:
> How would you do it on a headless system? Thanks.

I guess it should work in the same way by adding the param **--dev=/dev/ttyACM0** to the deconz.service file:
 /lib/systemd/system/deconz.service

...
ExecStart=/usr/bin/deCONZ -platform minimal --http-port=80 **--dev=/dev/ttyACM0**
...

Idaho947, 2023-05-01:
You just have to edit the device.confif file with nano and save ?

x-MaxG-x, 2023-05-01:
> You just have to edit the device-gui.service file with nano and save ?

After updating the service file you have to reload them and restart the service afterwards.

> systemctl daemon-reload
> systemctl restart deconz-gui.service

ANierbeck, 2023-05-02:
> > How would you do it on a headless system? Thanks.
> 
> I guess it should work in the same way by adding the param **--dev=/dev/ttyACM0** to the deconz.service file: /lib/systemd/system/deconz.service
> 
> ... ExecStart=/usr/bin/deCONZ -platform minimal --http-port=80 **--dev=/dev/ttyACM0** ...

can confirm, this worked for me. 
Thanks to all the people pointing at this direction. 🙇

manup, 2023-05-02:
Uff that's a nasty bug, good that it's fixed upstream in Debian.
While using `/dev/ttyACMx` is a workaround it could also randomly fail due to how Linux "randomly" assigns these paths so that ttyACM0 sometimes becomes ttyACM1.

Idaho947, 2023-05-02:
OK so we have to wait Debian for the fix and in the meantime juste add --dev=/dev/ttyACM0 (for exemple) in the deconz-gui.service ?

Mimiix, 2023-05-02:
Yes

Mimiix, 2023-05-02:
> Its Possible that i have the Same Problem with my zigbee2mqtt docker configuraiton?
> 
> ```shell
> zigbee2mqtt  | Zigbee2MQTT:error 2023-05-02 08:34:59: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
> zigbee2mqtt  |     at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
> zigbee2mqtt  |     at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
> zigbee2mqtt  |     at Zigbee.start (/app/lib/zigbee.ts:58:27)
> zigbee2mqtt  |     at Controller.start (/app/lib/controller.ts:101:27)
> zigbee2mqtt  |     at start (/app/index.js:107:5)
> ```
> 
> ```yaml
>  zigbee2mqtt:
>     container_name: zigbee2mqtt
>     restart: always
>     image: koenkk/zigbee2mqtt:1.28.4
>     volumes:
>       - ./zigbee2mqtt-data:/app/data
>       - /run/udev:/run/udev:ro
>     ports:
>       - 8080:8080
>     environment:
>       - TZ=Europe/Zurich
>     devices:
>       - /dev/ttyACM0:/dev/ttyACM0
> ```
> 
> ```shell
> [ 1078.028698] usb 1-1.3: USB disconnect, device number 18
> [ 1078.254314] usb 1-1.3: new full-speed USB device number 19 using xhci_hcd
> [ 1078.363336] usb 1-1.3: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
> [ 1078.363350] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [ 1078.363356] usb 1-1.3: Product: ConBee II
> [ 1078.363361] usb 1-1.3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
> [ 1078.363366] usb 1-1.3: SerialNumber: DE2662453
> [ 1078.368759] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
> ```
> 
> my devices info:
> 
> ```shell
> docker@dockerpi:~ $ ls -lah /dev/ttyACM0 
> crw-rw---- 1 root dialout 166, 0 May  2 10:18 /dev/ttyACM0
> 
> docker@dockerpi:~ $ ls -la /dev/serial/by-path/
> total 0
> drwxr-xr-x 2 root root 60 May  2 10:18 .
> drwxr-xr-x 3 root root 60 May  2 10:18 ..
> lrwxrwxrwx 1 root root 13 May  2 10:18 platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.3:1.0 -> ../../ttyACM0
> ```
> 
> anyone knows why i dont have an `/dev/serial/by-id` path?
Please keep this on-topic. This repository is about deCONZ, not docker.

I advice you to reach out to Z2M or docker support channels.

Idaho947, 2023-05-02:
I try to add like this, reboot the vm but it doesn't work.
[image]

I'm wrong ?

Fredi009sg, 2023-05-03:
I have similar issues. I can get connected by typing `deCONZ --dev=/dev/ttyACM0`

KennethLavrsen, 2023-05-03:
Please note that the bug is not to be blamed on Deconz. Deconz will try to autodetect the Conbee based on the naming of the devices in the /dev/serial/by-id directory. Without that you have to tell Deconz the path of your device.

Last year some punk checked in code for the systemd project that destroyed the creation of serial device names. It was soon fixed again.

For some odd reason, the debian project decided to backport some fixes to an old version of systemd and picked the version that is broken. I am not sure why and how but the result is that last week Debian pushed out a broken systemd. Not the kind of thing we normally expect of Debian. And many Debian derived distros just copy these updates and push them out also.

We are all waiting for Debian to push out the fix. It is there waiting to be pushed out and I am puzzled why the fix does not have some more urgency from the Debian project

A temporary fix for Debian is to go to the /usr/lib/udev/rules.d directory and replace the content of the file 60-serial.rules with this

# do not edit this file, it will be overwritten on update

ACTION=="remove", GOTO="serial_end"
SUBSYSTEM!="tty", GOTO="serial_end"

SUBSYSTEMS=="usb", IMPORT{builtin}="usb_id", IMPORT{builtin}="hwdb --subsystem=usb"
SUBSYSTEMS=="pci", ENV{ID_BUS}=="", ENV{ID_BUS}="pci", \
ENV{ID_VENDOR_ID}="$attr{vendor}", ENV{ID_MODEL_ID}="$attr{device}", \
IMPORT{builtin}="hwdb --subsystem=pci"

```
# /dev/serial/by-path/, /dev/serial/by-id/ for USB devices
KERNEL!="ttyUSB[0-9]*|ttyACM[0-9]*", GOTO="serial_end"

SUBSYSTEMS=="usb-serial", ENV{.ID_PORT}="$attr{port_number}"

IMPORT{builtin}="path_id"
ENV{ID_PATH}=="?*", ENV{.ID_PORT}=="", SYMLINK+="serial/by-path/$env{ID_PATH}"
ENV{ID_PATH}=="?*", ENV{.ID_PORT}=="?*", SYMLINK+="serial/by-path/$env{ID_PATH}-port$env{.ID_PORT}"

ENV{ID_BUS}=="", GOTO="serial_end"
ENV{ID_SERIAL}=="", GOTO="serial_end"
ENV{ID_USB_INTERFACE_NUM}=="", GOTO="serial_end"
ENV{.ID_PORT}=="", SYMLINK+="serial/by-id/$env{ID_BUS}-$env{ID_SERIAL}-if$env{ID_USB_INTERFACE_NUM}"
ENV{.ID_PORT}=="?*", SYMLINK+="serial/by-id/$env{ID_BUS}-$env{ID_SERIAL}-if$env{ID_USB_INTERFACE_NUM}-port$env{.ID_PORT}"

LABEL="serial_end"

```

Next time an apt upgrade changes the systemd packages this file will be overwritten. Hopefully when that happens, it will be with a correct version but not likely to be identical content.

boscho87, 2023-05-03:
maybe this will help you:

https://github.com/Koenkk/zigbee2mqtt/issues/17535#issuecomment-1532241700

stoneubi, 2023-05-12:
I do have the same issue. My setup is a virtual machine, dietpi on debian. After a apt get update / uprade deconz gui cannot connect to the usb conbee 2 device. lsusb shows the dongle in the vm but cannot connect. Anyhow, I will try mentioned workaround.

Mimiix, 2023-05-12:
As this issue is something that isn't fixable in deCONZ itself, i am closing it.

closed 2023-05-12 by Mimiix

MSchmitt67, 2023-05-21:
Hallo,
had the same problem this morning and found a solution/workaround.

MSchmitt67, 2023-05-21:
i found it on iobroker forum and it works for me 
[Link to iobroker forum](https://forum.iobroker.net/topic/65017/zigbee-conbee2-funktioniert-nach-update-nicht-mehr-udev/66?_=1684664822472)

KennethLavrsen, 2025-07-01:
The symlink is created by the udev system. Never ever create this manually. The ttyACM# device name is unpredictable. It can be 0 or 1 after a boot. The idea of the by-id symlink is that it is always the same. The issue that this bug report covered was solved years ago.