dresden-elektronik/deconz-rest-plugin #6939: Device name

edwin-2023, 2023-05-01:
## Device

- Product name: The device name as shown on the product or package.
- Manufacturer: As per deCONZ GUI Basic cluster.  LDS
- Model identifier: As per deCONZ GUI Basic cluster.   ZBT-DIMSwitch-D0000
- Device type : Please remove all unrelated device types.   RFD
  - Switch
  - Dim

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

edwin-2023, 2023-05-01:
Somehow the screenshots are not visible
[image]
[image]

Mimiix, 2023-05-01:
Can you please add all screenshots? We need screens of all clusters.

edwin-2023, 2023-05-01:
What do you mean with all clusters? Where can i find the info you need?

edwin-2023, 2023-05-01:
Cluster info is empty
[image]

Mimiix, 2023-05-01:
As mentioned in the issue template:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

and 

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-read-Clusters

edwin-2023, 2023-05-01:
That helps :)
Basic
[image]
Power Configuration
[image]
Identify
[image]
Diagnostics = all 0
ZLL commisioning
[image]
[image]
Do you also need the grey clusters?

Mimiix, 2023-05-01:
No need for the grey clusters. However ,i'd appreciate it if you put the above screens in the first post of this issue :)

Smanar, 2023-05-01:
Hello I hope you haven't HA ?
Can try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": " LDS",
  "modelid": "ZBT-DIMSwitch-D0000",
  "product": "LDS battery switch",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "refresh.interval": 86400,
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl"
          }
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0008"
    }
  ]
}
```

For the button map can try first to add the device in LDS part in the file button_maps.json
```
        "ZBT-DIMController-D0800Map": {
            "vendor": "LDS",
            "doc": "Mueller-Licht tint dimmer",
            "modelids": ["ZBT-DIMController-D0800", "ZBT-DIMSwitch-D0000"],
```

For missing one, just go in deconz log with flag "info" and "info_l2", and get log when pressing a button, something like
`
15:08:58:374 [INFO] - No button map for: 3AFE280100510001, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), command: ATTRIBUTE_REPORT (0x0A), payload: 00001080, zclSeq: 6`

edwin-2023, 2023-05-01:
> No need for the grey clusters. However ,i'd appreciate it if you put the above screens in the first post of this issue :)

Will do, but need a bit of time

Smanar, 2023-05-02:
It's the standard DDF for all battery remote.

>I'm using HomeSeer as domotic system.

So I think you are in this situation ? (no dockers ?)

> All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.

You just need to create a file called what_you_want.json with the DDF contain in a "devices" folder.
You will found the button_maps.json in /usr/share/deCONZ/devices 'can use a text editor to edit it.

edwin-2023, 2023-05-02:
Alright i think i got it....
BTW homeseer is a windows based domotica software system but deconz is running on a raspberry pi with a zigbeeII module. So deconz is basiclly running on linux.

DDF is totally new for me. As i understand it i need to copy your post, past in a editor and save the file in /usr/share/deCONZ/devices/ as a file whatever_name.json
Right?

And if it's not working just remove the .json file i created and undo the changes in button-maps

I guess i have to shutdown deconz and phoscon on the raspberry pi, how do i do that? Which commandline to use in terminal?

Smanar, 2023-05-02:
For exemples, you have 2 "devices" folder.

>And if it's not working just remove the .json file i created and undo the changes in button-maps

No need, if I m bad, it will just not work, no impact on others devices.

>I guess i have to shutdown deconz and phoscon on the raspberry pi, how do i do that? Which commandline to use in terminal?

` sudo reboot`

Agressive method, but working ^^, you have headless setup ? Can just restart the deconz service, but I never tried this command.

`sudo systemctl restart deconz`

edwin-2023, 2023-05-03:
Making connection via VNC i'm unable to create a file in /usr/share/deCONZ/devices/ via the file explorer. Must be a rights issue. No rightclick sudo option or something like that. Any suggestion?

edwin-2023, 2023-05-03:
got the file created.
Made a copy of button_maps.json but unable to edit (sudo nano button_maps.json) the original file. My guess is that some (deconz?) locks the file. tried  sudo systemctl stop deconz but no luck
Service -status-all does not show a service that i recognize as being deconz.

So Q. How to edit the button_maps.json file?

BabaIsYou, 2023-05-03:
From inside terminal you can type `sudo pcmanfm`, will  launch file explorer with privs.
AFAIK `button_maps.json` is not locked by DeConz, then, after having made a copy, you can modify it directly.
`sudo systemctl status deconz `or `sudo systemctl status deconz-gui` will give you information about deconz anyway.

edwin-2023, 2023-05-03:
Allright. Created the file, edited the button file. Still no luck.
Device does show in phoscon
[image]

But in deconz on/off and level are still greyed out
[image]

BabaIsYou, 2023-05-03:
> But in deconz on/off and level are still greyed out

And they will stay because that's the way that the device is exposing these clusters as server clusters.

edwin-2023, 2023-05-03:
> Be advised it wont show anything in Phoscon. You need to check what the Websocket provides (in phoscon : under Help > Api Information > events). There you should see events when clicking buttons.

> 
> Phoscon needs to add support on its own later.

This is what a have now when pressing button;
[image]
Does this mean that for deconz everything is working like its supposed to be?

How do we get the support going in Phoscon?

edwin-2023, 2023-05-03:
> > But in deconz on/off and level are still greyed out
> 
> And they will stay because that's the way that the device is exposing these clusters as client clusters.

Just looked at another device with dimming and switch capabilities and noticed the same. So all looks good.

Smanar, 2023-05-03:
> For the button map can try first to add the device in LDS part in the file button_maps.json
> ```
>
>         "ZBT-DIMController-D0800Map": {
>             "vendor": "LDS",
>             "doc": [...]

edwin-2023, 2023-05-03:
i did edited the button_maps.json file (adding ZBT-DIMSwitch-D0000)

edwin-2023, 2023-05-03:
this is center switch

16:07:53:611 Poll APS request to 0x385CFBFFFECBB96F cluster: 0x0006 dropped, values are fresh enough
16:07:54:238 [INFO] - No button map for: ZBT-DIMSwitch-D0000, broadcast to: 0x400A, endpoint: 0x01, cluster: ONOFF (0x0006), command: ON (0x01), payload: None, zclSeq: 84
16:07:54:240 Websocket 192.168.140.13:64617 send message: {"attr":{"id":"5","lastannounced":null,"lastseen":"2023-05-03T15:07Z","manufacturername":"LDS","modelid":"ZBT-DIMSwitch-D0000","name":"YNOA ZBT-DIMSwitch-D0000 5","swversion":"2.22","type":"ZHASwitch","uniqueid":"08:6b:d7:ff:fe:fc:a3:66-01-1000"},"e":"changed","id":"5","r":"sensors","t":"event","uniqueid":"08:6b:d7:ff:fe:fc:a3:66-01-1000"} (ret = 2125066264)
16:07:54:242 Websocket 192.168.140.236:50382 send message: {"attr":{"id":"5","lastannounced":null,"lastseen":"2023-05-03T15:07Z","manufacturername":"LDS","modelid":"ZBT-DIMSwitch-D0000","name":"YNOA ZBT-DIMSwitch-D0000 5","swversion":"2.22","type":"ZHASwitch","uniqueid":"08:6b:d7:ff:fe:fc:a3:66-01-1000"},"e":"changed","id":"5","r":"sensors","t":"event","uniqueid":"08:6b:d7:ff:fe:fc:a3:66-01-1000"} (ret = 2125066264)
16:07:54:558 poll node 00:15:8d:00:07:5b:47:f5-01

edwin-2023, 2023-05-03:
need to go, continue  tomorrow. Sorry,.....

edwin-2023, 2023-05-04:
i did a full reboot of the pi.
checked the button file, seems ok:
[image]

edwin-2023, 2023-05-04:
Alright, i eddited the button file to add the "
[image]
Now i see this in API info when i press the button
[image]
and in debug
[image]

edwin-2023, 2023-05-04:
So, this is how it should be?
Is there a way to get the device supported in Phoscon?

Mimiix, 2023-05-04:
In your screen, i see twice 1002, not sure if that is correct?

On Homeseer and HA you need to use the Events codes to map to buttons( i am sure for HA, not suire for Homeseer. For HA: https://www.home-assistant.io/integrations/deconz/#remote-control-devices)

For Phoscon, you need to request support on the forums.

Mimiix, 2023-05-04:
forums.phoscon.de :)

Smanar, 2023-05-04:
If you miss somes event, just give me the debug output message and the action/button used.

Just need to convert
`16:07:54:238 [INFO] - No button handler for for: ZBT-DIMSwitch-D0000, broadcast to: 0x400A, endpoint: 0x01, cluster: ONOFF (0x0006), command: ON (0x01), payload: None, zclSeq: 84`
in
`
 [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],`

edwin-2023, 2023-05-06:
> If you miss somes event, just give me the debug output message and the action/button used.
>
> Just need to convert `16:07:54:238 [INFO] - No button handler for for: ZBT-DIMSwitch-D0000, broadcast to: [...]

Sorry, missed the message.

Not sure what you mean. Which debug info do you need and where do I need to make a change?

Smanar, 2023-05-06:
Do you miss somes events ?

edwin-2023, 2023-05-06:
Not home at the moment. I think I missed long press but will check and come back

edwin-2023, 2023-05-28:
Just got home and will pickup this later next week.

edwin-2023, 2023-05-28:
Just tried, it get as status;
- on/off short
- dim up holding
- dim up released long
- dim down holding
- dim down released
So it would say, device is working!
What would be the next step?

Smanar, 2023-05-28:
Making a PR for the change will be included officially.
I can do it if you have problem, but you can try first, you can't break something, so you can retry if you fail ^^.

edwin-2023, 2023-05-28:
Hi. If you guide me how to improve will give it a try 😁

Smanar, 2023-05-28:
I think it will be easier to find a tutorial in your language.
But to make it simple, you need to create a fork (this is automatic is you try to edit the button map file on the official branch).
Then you need to upload a new DDF
And make the PR.
After that your fork will be linked to the PR on this girhub, so you can update your fork, and the change will be automatic

You can do all just with a web browser.

edwin-2023, 2023-05-30:
Alright, I'm afraid this is totally Spanish for me 🤣
Would appreciate it if you could do this.

Smanar, 2023-05-30:
PR done https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7027
If you can confirm all is ok, and don't need more code edition, Will remove the draft mode (sorry this issue is old, I don't remeber all change)

edwin-2023, 2023-05-30:
@Smanar. All is ok. Device is operating like it should. Together with Homeseer and the JowiHue plugin I'm able use the button to control event's. Thnx for support and opening PR

Smanar, 2023-05-31:
Ok thx, PR is ready, now need to wait.

edwin-2023, 2023-05-31:
Great!

edwin-2023, 2023-06-22:
Any news on this?

Smanar, 2023-06-22:
The PR is merged.
It's not woking ATM ?

edwin-2023, 2023-06-22:
It's absolutely working!

Smanar, 2023-06-23:
Lol, so which one news do you want  ?

edwin-2023, 2023-06-23:
Just afraid that when this thread got closed the PR part would stop.

Mimiix, 2023-06-23:
The pr is done. @Smanar please link the issue to the pr next time so it auto closes :)

closed 2023-06-23 by Mimiix

Smanar, 2023-06-24:
Ha yep, sorry my bad.