dresden-elektronik/deconz-rest-plugin #6940: Teekar Zigbee light switch

pimperator, 2023-05-01:
## Device

- Product name: The device name as shown on the product or package.
- Manufacturer: unknown
- Model identifier: unknown
- Device type : 
  - Switch

https://www.amazon.de/-/en/gp/product/B095LXWZS5/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

## Screenshots

[image]
[image]
[image]

pimperator, 2023-05-01:
additionally I am not sure wether and how identical these devices are with mine:
https://zigbee.blakadder.com/Tuya_DS-112L-3.html
https://zigbee.blakadder.com/Moes_ZS-EU1-PNW.html

at least the exterior appearance seems to be the same

Mimiix, 2023-05-01:
Can you please read the template properly and provide all screenshots and details as requested ?

pimperator, 2023-05-01:
@Mimiix I've updated the data. I guess you miss manufacturer and model but no matter how often I push the 'read' button in 'Cluster info' the fields are not being provided.
Sadly only after messing around for a longer time I found those information:
[image]

I will adjust these above

BabaIsYou, 2023-05-01:
Surprisingly this device was already recognized in 2021 #4896 ... and no more today ?

pimperator, 2023-05-02:
@BabaIsYou it works! 
Indeed it seems like the same device is handled in #4896
At first when I created the ticket (note: this is my first one) I sadly was unaware where to gather the information and the manufacturer and model was shown as unknown. Also there are a bunch of other switches with different model id so I was not sure.

Additionally even when #4896 covered this device and I can see the device id by searching in two different [tickets](https://github.com/dresden-elektronik/deconz-rest-plugin/search?q=TS0011) the manufacturer is missing in the whole repo (see [here](https://github.com/dresden-elektronik/deconz-rest-plugin/search?q=_TYZB01_qeqvmvti))

PR #6943 (2023-05-21): DDF - Add Teekar Zigbee light switch support
Fix #6940 6940 and close #4896
commit 7907fe65 (2023-05-21): DDF Tuya / Teekar Zigbee light switch (_TYZB01_qeqvmvti) (#6943)

Fix #6940 6940 and close #4896
closed 2023-05-21 by manup