dresden-elektronik/deconz-rest-plugin #6941: Device name

xvolte, 2023-05-02:
## Device

- Product name: Tuya Smart Zigbee Siren Alarm Waterproof Outdoor With Solar
- Manufacturer: TUYA
- Model identifier: As per deCONZ GUI Basic cluster.
- Device type : Please remove all unrelated device types. 
  - Siren
- guys already made a converter here : https://github.com/Koenkk/zigbee2mqtt/discussions/17325#discussioncomment-5772907

## Screenshots
[image]
[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2023-05-02:
Can you please read the template properly and provide all screenshots and details as requested ?

xvolte, 2023-05-02:
i think this is the Best Link (and quite new : From yesterday) to add support : https://github.com/Koenkk/zigbee2mqtt/discussions/17325#discussioncomment-5772907

Mimiix, 2023-05-02:
Still missing Groups screenshot.

Mimiix, 2023-05-02:
All good :)

BabaIsYou, 2023-05-02:
First time I try this kind of device  ... may be all is not working at first try 

```
 {
	"schema": "devcap1.schema.json",
	"manufacturername": "_TZE200_nlrfgpny",
	"modelid": "TS0601",
	"product": "Tuya Smart Zigbee Siren Alarm",
	"sleeper": false,
	"status": "Silver",
	"subdevices": [{
			"type": "$TYPE_WARNING_DEVICE",
			"restapi": "/lights",
			"uuid": [
				"$address.ext",
				"0x01"
			],
			"items": [{
					"name": "attr/id"
				},
				{
					"name": "attr/lastannounced"
				},
				{
					"name": "attr/lastseen"
				},
				{
					"name": "attr/manufacturername"
				},
				{
					"name": "attr/modelid"
				},
				{
					"name": "attr/name"
				},
				{
					"name": "attr/swversion",
					"parse": {
						"fn": "zcl",
						"ep": 1,
						"cl": "0x0000",
						"at": "0x0001",
						"script": "tuya_swversion.js"
					},
					"read": {
						"fn": "zcl",
						"ep": 1,
						"cl": "0x0000",
						"at": "0x0001"
					}
				},
				{
					"name": "attr/type"
				},
				{
					"name": "attr/uniqueid"
				},
				{
					"name": "config/alert",
					"description": "Alarm mode, 0 : sound, 1: light, 2 : Sound and light. Default 0",
					"values": [
						["0", "Sound"],
						["1", "Light"],
						["2", "Sound & light"]
					],
					"read": {
						"fn": "none"
					},
					"write": {
						"dpid": 102,
						"dt": "0x2b",
						"eval": "Mode = ['0','1','2']; Attr.val = Math.max(0,Mode.indexOf(Item.val));",
						"fn": "tuya"
					},
					"parse": {
						"dpid": 102,
						"eval": "Item.val = String(Attr.val);",
						"fn": "tuya"
					},
					"default": "0"
				},
				{
					"name": "config/melody",
					"description": "Alarm melody, from 0 to 2. Default 1",
					"range": [0, 2],
					"read": {
						"fn": "none"
					},
					"write": {
						"dpid": 21,
						"dt": "0x2b",
						"eval": "Item.val;",
						"fn": "tuya"
					},
					"parse": {
						"dpid": 21,
						"eval": "Item.val = Attr.val;",
						"fn": "tuya"
					},
					"default": 1
				},
				{
					"name": "config/duration",
					"description": "Alarm duration. Default 10",
					"read": {
						"fn": "none"
					},
					"write": {
						"dpid": 5,
						"dt": "0x2b",
						"eval": "Item.val;",
						"fn": "tuya"
					},
					"parse": {
						"dpid": 5,
						"eval": "Item.val = Attr.val;",
						"fn": "tuya"
					},
					"default": 10
				},
				{
					"name": "state/battery",
					"description": "Charge state 0: Not Charging, 1: Charging",
					"read": {
						"fn": "none"
					},
					"parse": {
						"dpid": 6,
						"eval": "Item.val = Attr.val;",
						"fn": "tuya"
					}
				},
				{
					"name": "state/reachable"
				}
			]
		},
		{
			"type": "$TYPE_ALARM_SENSOR",
			"restapi": "/sensors",
			"uuid": [
				"$address.ext",
				"0x01",
				"0x0500"
			],
			"fingerprint": {
				"profile": "0x0104",
				"device": "0x0403",
				"endpoint": "0x01",
				"in": [
					"0x0000",
					"0x0001",
					"0x0004",
					"0xef00"
				]
			},
			"items": [{
					"name": "attr/id"
				},
				{
					"name": "attr/lastannounced"
				},
				{
					"name": "attr/lastseen"
				},
				{
					"name": "attr/manufacturername"
				},
				{
					"name": "attr/modelid"
				},
				{
					"name": "attr/name"
				},
				{
					"name": "attr/swversion",
					"parse": {
						"fn": "zcl",
						"ep": 1,
						"cl": "0x0000",
						"at": "0x0001",
						"script": "tuya_swversion.js"
					},
					"read": {
						"fn": "zcl",
						"ep": 1,
						"cl": "0x0000",
						"at": "0x0001"
					}
				},
				{
					"name": "attr/type"
				},
				{
					"name": "attr/uniqueid"
				},
				{
					"name": "config/enrolled",
					"public": false
				},
				{
					"name": "config/tuya_unlock"
				},
				{
					"name": "config/on",
					"description": "Alarm switch",
					"range": [0, 1],
					"read": {
						"fn": "none"
					},
					"write": {
						"dpid": 13,
						"dt": "0x10",
						"eval": "Item.val == 1 ? 1 : 0;",
						"fn": "tuya"
					},
					"parse": {
						"dpid": 13,
						"eval": "Item.val = Attr.val;",
						"fn": "tuya"
					},
					"default": 0
				},
				{
					"name": "config/pending"
				},
				{
					"name": "config/reachable"
				},
				{
					"name": "state/tampered",
					"read": {
						"fn": "none"
					},
					"parse": {
						"dpid": 20,
						"eval": "Item.val = Attr.val;",
						"fn": "tuya"
					}
				},
				{
					"name": "state/alarm",
					"read": {
						"fn": "none"
					},
					"parse": {
						"dpid": 101,
						"eval": "Item.val = Attr.val;",
						"fn": "tuya"
					}
				},
				{
					"name": "config/battery",
					"refresh.interval": 3600,
					"read": {
						"fn": "none"
					},
					"parse": {
						"dpid": 15,
						"eval": "Item.val = Attr.val / 2;",
						"fn": "tuya"
					}
				},
				{
					"name": "state/lastupdated"
				},
				{
					"name": "state/deviceruntime",
					"description": "Alarm time",
					"read": {
						"fn": "none"
					},
					"parse": {
						"dpid": 7,
						"eval": "Item.val = Attr.val;",
						"fn": "tuya"
					}
				},
				{
					"name": "state/lowbattery",
					"read": {
						"fn": "none"
					},
					"parse": {
						"dpid": 110,
						"eval": "Item.val = Attr.val;",
						"fn": "tuya"
					}
				}
			]
		}
	],
	"bindings": [{
		"bind": "unicast",
		"src.ep": 1,
		"cl": "0xEF00"
	}]
}
```

xvolte, 2023-05-02:
nice:) thank you!
how do i test it ?

i got a conbee 2 attached to a rpi 4 running Home Assistant :D 

Thanks in advance!

BabaIsYou, 2023-05-02:
May be this coud help : https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6806#issuecomment-1476606167

xvolte, 2023-05-02:
Nice thank you.
I was able to add the DDF, however i can't see the device in Home Assistant. i'm now rebooting HA, in case something is done at startup, but else, i don't know how to test it !

xvolte, 2023-05-02:
So the only thing i changed was to make the melody a public item.

it now looks like this : 
[image]

[image]

[image]

[image]

but still got nothing in Home Assistant : No device / no entity

Mimiix, 2023-05-02:
Because the DDF is silver and that probably isn't enabled..

BabaIsYou, 2023-05-02:
> Because the DDF is silver and that probably isn't enabled..

Weird because the DDF has been loaded anyway, but I miss this precision, DDF level had to be Silver (https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-change-DDF-modes-and-enable-or-disable-DDF-levels)

xvolte, 2023-05-02:
progress:
changing it to gold and doing a hot reload and read node, changed things : 

[image]

[image]

but from there, i'm stuck:

[image]

looks like it is a wrong device type. maybe we should consider this as a switch, so that i can on/off the alarm ?

BabaIsYou, 2023-05-02:
REST-API only knows lights or sensors, no switch device type. This DDF should have add two new devices, one light that support most of the configuration parts, and one sensor that support most of the reported events from the device.

xvolte, 2023-05-02:
so i reverted to silver and enabled support for silver.
can't find sensors or light ...

BabaIsYou, 2023-05-02:
> so i reverted to silver and enabled support for silver. can't find sensors or light ...

If it has been loaded once changed into Gold it should be loaded when changing level to Silver and revert the DDF to Silver. Have you deleted the device and added it again ?

xvolte, 2023-05-02:
if i delete the device, i need to pair it again, and since i need to put the custom DDF file on the device ... it will be configured after joining ... or am i missing something ?

BabaIsYou, 2023-05-02:
If the DDF is stored in the right folder it will be loaded automatically when DeConz will detect the manufactueername and modelid match those in a DDF. If you have to load it by hands it's because the DDF is not in the right place.

xvolte, 2023-05-02:
the DDF has been stored here : 
/data/.local/share/dresden-elektronik/deCONZ/devices/
does the name of the js file matters ?

BabaIsYou, 2023-05-02:
> does the name of the js file matters ?

Not really, it just have to have extension` .json`, like t`uyasiren.json`

xvolte, 2023-05-02:
a reboot of Home Assistant did the trick:

[image]

still waiting for the switch to come online

on the sensor side : 
battery says 0% but it should be 100% charged 
[image]

BabaIsYou, 2023-05-02:
Hmmm may be DDF need some adjustements or waitings ... . Tuya usually use battery update after max 4 hours as a kind of heartbeat. If that dosn't work it could be that calculation is wrong.

xvolte, 2023-05-02:
device kept saying offline, i removed it, i'm going to pair it again after the reboot

xvolte, 2023-05-02:
> Hmmm may be DDF need some adjustements or waitings ... . Tuya usually use battery update after max 4 hours as a kind of heartbeat. If that dosn't work it could be that calculation is wrong.

i think the guy on the other website mentionned a "In charge" or not, so 0 or 1 sensor, when it is plugged to a usb cable.
anyway, we will see in couple of minute after Re pairing

BabaIsYou, 2023-05-02:
I masked the "in charge state" behind the `state/battery` of the light device but I may be light doesn't expose this on REST-API

xvolte, 2023-05-02:
so i did it again, it does create a device but i can't change the switches
[image]

BabaIsYou, 2023-05-02:
I'm a bit surprised that you just have one more device, should have two. Could you got aceess to the REST-API directly rather than using HA ? https://dresden-elektronik.github.io/deconz-rest-doc/

xvolte, 2023-05-02:
maybe it is also because I never configured the device ? like the alarm sound and so on? maybe I should first set those values ?

BabaIsYou, 2023-05-02:
> maybe it is also because I never configured the device ? like the alarm sound and so on? maybe I should first set those values ?

Don't think so. You should have two devices, configured or not. I you don't have two then may be this is a DDF issue ... What a request like `http;//<your deconz ip address>:<and port number>/api/<your apikey>/lights` gives ? same with `/sensors` at the end in place of `/lights`

xvolte, 2023-05-02:
Sensor:

[image]

switch:
[image]

light:
[image]

xvolte, 2023-05-02:
i noticed this in your script : 
[image]

do i have this script somewhere ?

BabaIsYou, 2023-05-02:
> do i have this script somewhere ?

Seesm you don't have it where you put the DDF file but it doesn't matter for now. It just updates swversion item and we can see that later ... Don't worry about that. ;-)

BabaIsYou, 2023-05-02:
Then you have to devices (light as "TuyaSiren" and sensor as "Alarm 47").
Seems that light device doesn't expose all items, as I was afraid of. We have to find other way to gain access to them.
BTW for the moment we can focus on sensor items to see if they are updated ...

xvolte, 2023-05-03:
ok, what do you want me to do ? so far, i can't see anything updating (or it is already "up to date")

BabaIsYou, 2023-05-03:
Well, i changed of approach after looking at what REST-API exposes : 
You will have to delte device one more time, changing DDF with this one and pairing device again please.
I based the DPID from the links you referenced, hope there were functionnal ;-) But to be honest I'm still confused between dpip 102 (alarm mode) and dpid 1 (alarm state)

```
 {
 	"schema": "devcap1.schema.json",
 	"manufacturername": "_TZE200_nlrfgpny",
 	"modelid": "TS0601",
 	"product": "Tuya Smart Zigbee Siren Alarm",
 	"sleeper": false,
 	"status": "Silver",
 	"subdevices": [{
 			"type": "$TYPE_WARNING_DEVICE",
 			"restapi": "/lights",
 			"uuid": [
 				"$address.ext",
 				"0x01"
 			],
 			"items": [{
 					"name": "attr/id"
 				},
 				{
 					"name": "attr/lastannounced"
 				},
 				{
 					"name": "attr/lastseen"
 				},
 				{
 					"name": "attr/manufacturername"
 				},
 				{
 					"name": "attr/modelid"
 				},
 				{
 					"name": "attr/name"
 				},
 				{
 					"name": "attr/swversion",
 					"parse": {
 						"fn": "zcl",
 						"ep": 1,
 						"cl": "0x0000",
 						"at": "0x0001",
 						"script": "tuya_swversion.js"
 					},
 					"read": {
 						"fn": "zcl",
 						"ep": 1,
 						"cl": "0x0000",
 						"at": "0x0001"
 					}
 				},
 				{
 					"name": "attr/type"
 				},
 				{
 					"name": "attr/uniqueid"
 				},
 				{
 					"name": "config/tuya_unlock"
 				},
 				{
 					"name": "state/alert",
 					"default": "none"
 				},
 				{
 					"name": "config/on",
 					"description": "Alarm switch On or Off",
 					"range": [0, 1],
[... 307 lines cut ...]
 				{
 					"name": "state/lowbattery",
 					"read": {
 						"fn": "none"
 					},
 					"parse": {
 						"dpid": 110,
 						"eval": "Item.val = Attr.val;",
 						"fn": "tuya"
 					}
 				},
 				{
 					"name": "state/lastupdated"
 				}
 			]
 		}
 	],
 	"bindings": [{
 		"bind": "unicast",
 		"src.ep": 1,
 		"cl": "0xEF00"
 	}]
 }

```

BabaIsYou, 2023-05-03:
It should have 1 light and 2 sensors now.
Will try something else, for the light part, can you change

```
 				{
 					"name": "config/on",
 					"description": "Alarm switch On or Off",
 					"range": [0, 1],
 					"read": {
 						"fn": "none"
 					},
 					"write": {
 						"dpid": 13,
 						"dt": "0x10",
 						"eval": "Item.val == 1 ? 1 : 0;",
 						"fn": "tuya"
 					},
 					"parse": {
 						"dpid": 13,
 						"eval": "Item.val = Attr.val;",
 						"fn": "tuya"
 					},
 					"default": 0
 				},

```
by
```

 				{
 					"name": "state/on",
 					"description": "Alarm switch On or Off"
 					"read": {
 						"fn": "none"
 					},
 					"write": {
 						"dpid": 1,
 						"dt": "0x10",
 						"eval": "Item.val == 1 ? 1 : 0;",
 						"fn": "tuya"
 					},
 					"parse": {
 						"dpid": 1,
 						"eval": "Item.val = Attr.val;",
 						"fn": "tuya"
 					},
 					"default": 0
 				},
```

xvolte, 2023-05-03:
new file, new bug : 
[image]
i didn't change the filename, only the content .... but he won't read it anymore ?

Mimiix, 2023-05-03:
Error in the content.

xvolte, 2023-05-03:
it was missing a "," in the modification after "Alarm switch On or Off"**,**

BabaIsYou, 2023-05-03:
You're right, my mistake. Sorry. Then are you able to activate the device ?

xvolte, 2023-05-03:
nope

xvolte, 2023-05-03:
does this help ? https://github.com/Koenkk/zigbee2mqtt/discussions/17325#discussioncomment-5772907

Mimiix, 2023-05-03:
Are you using the rest api to activate or home assistant?

I strongly advice to use the rest api

xvolte, 2023-05-03:
home assistant :D 
i'll try

Mimiix, 2023-05-03:
> home assistant :D i'll try but it is not that easy as the deconz is running within a container, not accessible from my LAN, only home assistant has access to it, so i have trouble sending command to the api within the container

The addon has the option to forward the ports.

xvolte, 2023-05-03:
So what do you want me to try ?

i did that : 
PUT http://192.168.210.205:40850/api/xxxxxx/lights/15/state

with this json:
{
   "alert": "lselect", 
   "on": true
}

results:
[{"success":{"/lights/15/state/on":true}},{"success":{"/lights/15/state/alert":"lselect"}}]

but no sound

BabaIsYou, 2023-05-03:
> does this help ? [Koenkk/zigbee2mqtt#17325 (comment)](https://github.com/Koenkk/zigbee2mqtt/discussions/17325#discussioncomment-5772907)

That's exactly the main link I used to transform this in DDF. Thx ;-)

May be it doesn't work if the alarm switch (dpid 13) is off. This is the "on" item in the alarm sensor

xvolte, 2023-05-03:
are you talking about the "on" here ? (i changed it with the rest api)

[image]

xvolte, 2023-05-03:
and the sensors: 
[image]

BabaIsYou, 2023-05-03:
Exactly, but a lot of items are missing (tampered, melody, volume, ...). Are you sure about the DDF used ?

xvolte, 2023-05-03:
i can try removing it again and re pair it again

xvolte, 2023-05-03:
might be because the "public item" is not switched on ?

BabaIsYou, 2023-05-03:
> might be because the "public item" is not switched on ?

Yes it could be ! ;-)

xvolte, 2023-05-03:
[image]

BabaIsYou, 2023-05-03:
It's certainly same thing about volume. These items have missing default generic item json files ... Will correct it later.

BabaIsYou, 2023-05-03:
> i noticed sometimes in the ddf editor, it is listed "zcl" and sometimes "tuya", is this normal ?

Yes, it depends if this is "standard" zcl Zigbee attributes or Tuya specific item. As you can see this device as quite nothing in standard Zigbee cluster/Item ... About all is hidden in Tuya Cluster

xvolte, 2023-05-03:
would it help if you have other TS0601 DDF devices i saw there are a couple of them in the DDF folder?

BabaIsYou, 2023-05-03:
Yes but none like yours : Siren on battery with so many configuration parameters ;-(

xvolte, 2023-05-03:
> Yes but none like yours : Siren on battery with so many configuration parameters ;-(

if you want the files i can extract them. i don't know what to say apart A very big thanks with all your help :D

BabaIsYou, 2023-05-03:
They are all here : https://github.com/dresden-elektronik/deconz-rest-plugin/tree/master/devices/tuya
The next step is to catch DeConz log to see what happens from and to the device. But my main concern is that I don't undestand the way to activate the siren with so many parameters.

BabaIsYou, 2023-05-03:
To action the siren, if I believe https://github.com/zigbeefordomoticz/z4d-certified-devices/blob/4766f25fe014e220949d7beb1e139e6d4d7ccdae/z4d_certified_devices/Certified/Tuya/TS0601-Solar-Siren.json it's dp13, that correspond to 
```
 				{
 					"name": "config/on",
 					"description": "Alarm switch On or Off",
 					"range": [0, 1],
 					"read": {
 						"fn": "none"
 					},
 					"write": {
 						"dpid": 13,
 						"dt": "0x10",
 						"eval": "Item.val == 1 ? 1 : 0;",
 						"fn": "tuya"
 					},
 					"parse": {
 						"dpid": 13,
 						"eval": "Item.val = Attr.val;",
 						"fn": "tuya"
 					},
 					"default": 0
 				},
```
in the alarm sensor. It was also the `config/on` in light that  asked you to change previously ...?!

[35 more replies]

closed 2023-06-03 by github-actions[bot]