dresden-elektronik/deconz-rest-plugin #6944: GiEXperience SMART Water Valve

BEskandari, 2023-05-02:
## Device

- Product name: GiEXperience SMART Water Valve
- Manufacturer: _TZE200_a7sghmms
- Model identifier: TS0601
- Device type : 
  - Other: Water Valve
- Link to product
[https://www.aliexpress.com/item/1005002767720399.html?spm=a2g0o.productlist.main.23.2a167054om2ca6&algo_pvid=6cac93dd-63d8-4b5e-9e19-2381e9e3a8fa&algo_exp_id=6cac93dd-63d8-4b5e-9e19-2381e9e3a8fa-11&pdp_npi=3%40dis%21CHF%2165.38%2144.46%21%21%21%21%21%40211bf3f116830512797731945d080f%2112000022089158096%21sea%21CH%210&curPageLogUid=S5NqvJVz8CTR](url)

## Screenshots

### Node Info
[image]
[image]
[image]
### Basic
[image]
[image]
[image]
### Groups
[image]
[image]
[image]
### Scenes
[image]
[image]
[image]
[image]
[image]
[image]
### Other: Tuya specific
[image]
### Other: OTAU
[image]
[image]
[image]
### Other: Time
[image]

BabaIsYou, 2023-05-03:
Could try this DDF ? You,Ll have two devices, one On/off "light" and one Consumption "sensor"
```
{
	"schema": "devcap1.schema.json",
	"manufacturername": ["_TZE200_a7sghmms", "_TZE200_sh1btabb"],
	"modelid": ["TS0601", "TS0601"],
	"vendor": "GIEXperience",
	"product": "Smart water valve",
	"sleeper": false,
	"status": "Gold",
	"subdevices": [{
			"type": "$TYPE_ON_OFF_OUTPUT",
			"restapi": "/lights",
			"uuid": [
				"$address.ext",
				"0x01"
			],
			"items": [{
					"name": "attr/id"
				},
				{
					"name": "attr/lastannounced"
				},
				{
					"name": "attr/lastseen"
				},
				{
					"name": "attr/manufacturername"
				},
				{
					"name": "attr/modelid"
				},
				{
					"name": "attr/name"
				},
				{
					"name": "attr/swversion"
				},
				{
					"name": "attr/type"
				},
				{
					"name": "attr/uniqueid"
				},
				{
					"name": "state/on",
					"refresh.interval": 300,
					"read": {
						"fn": "tuya"
					},
					"write": {
						"dpid": 2,
						"dt": "0x10",
						"eval": "Item.val == 1 ? 1 : 0;",
						"fn": "tuya"
					},
					"parse": {
						"dpid": 2,
						"eval": "Item.val = Attr.val;",
						"fn": "tuya"
					},
					"default": 0
				},
				{
					"name": "state/reachable"
				}
			]
		},
		{
			"type": "$TYPE_CONSUMPTION_SENSOR",
			"restapi": "/sensors",
			"uuid": [
				"$address.ext",
				"0x01",
				"0x0702"
			],
			"items": [{
					"name": "attr/id"
				},
				{
					"name": "attr/lastannounced"
				},
				{
					"name": "attr/lastseen"
				},
				{
					"name": "attr/manufacturername"
				},
				{
					"name": "attr/modelid"
				},
				{
					"name": "attr/name"
				},
				{
					"name": "attr/swversion"
				},
				{
					"name": "attr/type"
				},
				{
					"name": "attr/uniqueid"
				},
				{
					"name": "config/battery",
					"parse": {
						"dpid": 108,
						"eval": "Item.val = Attr.val;",
						"fn": "tuya"
					},
					"default": 0
				},
				{
					"name": "config/duration",
					"description": "Last irrigation duration",
					"parse": {
						"dpid": 114,
						"eval": "Item.val = Attr.val;",
						"fn": "tuya"
					}
				},
				{
					"name": "state/consumption",
					"description": "Rcurrent_summ_delivered",
					"parse": {
						"dpid": 111,
						"eval": "Item.val == Attr.val;",
						"fn": "tuya"
					},
					"default": 0
				},
				{
					"name": "config/reachable"
				},
				{
					"name": "state/lastupdated"
				},
				{
					"name": "config/delay",
					"description": "Cycle irrigation interval in seconds min=0 max=3600",
					"range": [0, 3600],
					"write": {
						"dpid": 104,
						"dt": "0x2b",
						"eval": "Attr.val = Item.val;",
						"fn": "tuya"
					},
					"parse": {
						"dpid": 104,
						"eval": "Item.val = Attr.val ;",
						"fn": "tuya"
					}
				},
				{
					"name": "state/seconds_remaining",
					"description": "Irrigation target, duration in seconds or capacity in litres (depending on mode) min=0 max=3600",
					"range": [0, 3600],
					"write": {
						"dpid": 104,
						"dt": "0x2b",
						"eval": "Attr.val = Item.val;",
						"fn": "tuya"
					},
					"parse": {
						"dpid": 104,
						"eval": "Item.val = Attr.val ;",
						"fn": "tuya"
					}
				}, {
					"name": "config/mode",
					"description": "Device mode: [0] duration [1] capacity",
					"write": {
						"dpid": 1,
						"dt": "0x10",
						"eval": "Item.val == '1' ? 1 : 0;",
						"fn": "tuya"
					},
					"parse": {
						"dpid": 1,
						"eval": "Item.val = String(Attr.val) ;",
						"fn": "tuya"
					}
				}
			]
		}
	]
}
```

BEskandari, 2023-05-03:
I just added to the DDF those lines twice:
```
"name": "attr/swversion",
"parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
"read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
```
I can turn on/off the valve and have the battery level.

I have some doubt on the **ZHAConsumption** type sensor, because through Homebridge Deconz it gives me as below without any control on the duration. See below:
[image]

BabaIsYou, 2023-05-03:
The Consumption sensor could eventually be replaced by 
```
      "type": "$TYPE_OPEN_CLOSE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
```

instead of
```

		"type": "$TYPE_CONSUMPTION_SENSOR",
			"restapi": "/sensors",
			"uuid": [
				"$address.ext",
				"0x01",
				"0x0702"
			],
```

BEskandari, 2023-05-03:
REST-API for /lights:
```
{
    "config": {
        "groups": []
    },
    "etag": "1d9980c55c5ca365b9597d2bf16ea338",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2023-05-03T19:03Z",
    "manufacturername": "_TZE200_a7sghmms",
    "modelid": "TS0601",
    "name": "Water Valve",
    "state": {
        "on": false,
        "reachable": true
    },
    "swversion": "1.0.8",
    "type": "On/Off output",
    "uniqueid": "a4:c1:38:17:9e:c3:8c:06-01"
}
```
REST-API for /sensors:
```
{
    "config": {
        "battery": 100,
        "delay": 0,
        "duration": null,
        "mode": "false",
        "on": true,
        "reachable": true
    },
    "etag": "c05d2ebb4c348c3abd2e8eb3fe497423",
    "lastannounced": null,
    "lastseen": "2023-05-03T19:07Z",
    "manufacturername": "_TZE200_a7sghmms",
    "modelid": "TS0601",
    "name": "Consumption 22",
    "state": {
        "consumption": 0,
        "lastupdated": "2023-05-03T16:50:15.724",
        "seconds_remaining": 0
    },
    "swversion": "1.0.8",
    "type": "ZHAConsumption",
    "uniqueid": "a4:c1:38:17:9e:c3:8c:06-01-0702"
}
```
Let me check **$TYPE_OPEN_CLOSE_SENSOR**

BEskandari, 2023-05-04:
By using $TYPE_OPEN_CLOSE_SENSOR
REST-API for /lights:
```
{
    "config": {
        "groups": []
    },
    "etag": "204faf3a975f99b35e2c0b6879bf0797",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2023-05-04T17:41Z",
    "manufacturername": "_TZE200_a7sghmms",
    "modelid": "TS0601",
    "name": "Water Valve",
    "state": {
        "on": false,
        "reachable": true
    },
    "swversion": "1.0.8",
    "type": "On/Off output",
    "uniqueid": "a4:c1:38:17:9e:c3:8c:06-01"
}
```
REST-API for /sensors:
```
{
    "config": {
        "battery": 100,
        "delay": 0,
        "duration": null,
        "mode": null,
        "on": true,
        "reachable": true
    },
    "etag": "56246383b299b40b671f3fe11dc60747",
    "lastannounced": null,
    "lastseen": "2023-05-04T17:43Z",
    "manufacturername": "_TZE200_a7sghmms",
    "modelid": "TS0601",
    "name": "OpenClose 23",
    "state": {
        "consumption": 0,
        "lastupdated": "none",
        "seconds_remaining": 0
    },
    "swversion": "1.0.8",
    "type": "ZHAOpenClose",
    "uniqueid": "a4:c1:38:17:9e:c3:8c:06-01-0500"
}
```
but in this case **I get error in Homebridge Deconz** `Water Valve: error: TypeError: Cannot read properties of undefined`

I would like to quote @ebaauw 

> Not sure I appreciate the full features of the device, but this looks like a dirty hack, abusing existing API attributes to mean something completely different. No API client is going to understand that, without whitelisting the device and treating it special. I won’t be doing that in Homebridge deCONZ. In fact I just introduced the `capabilities` and `config` you `/lights` to get rid of whitelisting lights.
> 
> Looks like you’d need a new device type and new attributes to support the device properly. That cannot be done in a DDF alone, you’d need changes to the C++ code. I’m happy to support a new device type in Homebridge deCONZ if it’s well-defined.

BabaIsYou, 2023-05-04:
I know that some devices are an in-between mode that's perhaps not clearly identified and that some measure are only considered under one angle (electricity) in REST-API. But from my point of view is too limited regarding to Zigbee specification. The formatting of the measure (attribute 0x0300 of cluster 0x0702) is here to precise unit of measure (from kWh, m3, BTU, liters, etc ...). Then REST-API is too limitied (miising an item to transfer unit) , and so are Third party app expectations (just KW are attendee).

Then I agree with Erik. And he is in better position than me to have C++ modification approuved.

BEskandari, 2023-05-04:
I rather prefer to use $TYPE_CONSUMPTION_SENSOR as the change doesn't bring any additional feature. 
Do we need just to add the swversion or can we do more?

@BabaIsYou if you give me the name of the existing DDF I can try to make a PR for your additions.

BabaIsYou, 2023-05-04:
I see no more addition at this step than swversion. I suggest also to change` "status": "Gold"`, for `"status": "Silver",` and to put the DDF in https://github.com/dresden-elektronik/deconz-rest-plugin/tree/f837452b4eec614746e3ba56f214b1096342a797/devices/tuya with name `_TZE200_TS0601_water_valve.json`

BabaIsYou, 2023-05-04:
If you add something like `close #6944` in the comment of the PR, github will automatically close this issues once PR merge with a next version.

BEskandari, 2023-05-06:
@BabaIsYou, could you please let me know if I made the PR #6947 correctly or not? it seems stuck on a validation process.

BabaIsYou, 2023-05-06:
Yes seems good. Waiting for approval.

miloit, 2023-09-12:
Hello is this integrated???