dresden-elektronik/deconz-rest-plugin #6948: Device support - Leviton Dimmer Switch DG6HD-1BW (Switch Support)

mdolnik, 2023-05-05:
## Device

[image]

- Product name: `Zigbee Dimmer Switch`
- Manufacturer: `Leviton`
- Model identifier: `DG6HD-1BW`
- Device type :
  - Light
  - Switch 

## Issues:

**_Note: Right now this Dimmer switch _does_ work as a light with most functionality intact._**

**But it does have two issues:**
1. **When you physically turn the switch on/off it can take a long time to register in Phoscon / Home Assistant:**
    - All other physical mains switches I have (different makes/models) will register almost instantly (~ 1 second) but this switch can take anywhere from 5 - 20 seconds to update its state.
    - I assume there are no events being pushed from the device when physically changed and the updates from from polling the status.
2. **I would like for this to register as a switch and not just a light**
    - In other words, whenever buttons are physically pressed, it should send out events of the button presses.
    - One example would be to send an On/Up button press event even though the light is already on.

## Screenshots (01 Home Automation Dimmable Light)

### Node Info
[image]

### Basic
[image]

### Identify
[image]

### Groups
[image]

### Scenes
[image]

### On/Off
[image]
[image]

### Level Control
[image]

### Ballast Configuration
[image]

### Diagnostics
[image]

## Screenshots (02 c4c8 Dimmable Light)

### Unknown
[image]

## Screenshots (03 Home Automation Dimmer Switch)

### Basic
[image]

### Identify
[image]

## Screenshots (F2 Green Power GP Combo Basic)

### Green Power
[image]

mdolnik, 2023-05-05:
Similar closed issue: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2483

mdolnik, 2023-05-05:
I am willing / able to make code changes to try out on my end.

Smanar, 2023-05-10:
Hello I think it's possible to do only with DDF
Can you share the DDF auto generated for this device ?

> I assume there are no events being pushed from the device when physically changed and the updates from from polling the status.

Yes, I think you are right, but it's old working mode, even philip use reporting now on recent stuff.
Can reduce the polling using the DDF.

For the second part, few device support that, but it seem you device is able to do that, using the endpoint 0x03, from that I m seing the 0x01 is for the "light" and the 0x03 for the "switch".
This is possible too using the DDF, need to add a ZHASwitch sensor ann probably edit the buttonmap file.

mdolnik, 2023-05-11:
Here is the auto-generated DDF:
```json
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Leviton",
  "modelid": "DG6HD",
  "product": "DG6HD",
  "sleeper": false,
  "status": "Draft",
  "subdevices": [
    {
      "type": "$TYPE_DIMMABLE_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/bri",
          "description": "The current brightness.",
          "refresh.interval": 5
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```

mdolnik, 2023-05-11:
> Can reduce the polling using the DDF

After I made the original post I was deep-diving to try figure this out all myself, changing the poll interval down to 1 second had no effect, it would still take well over 5 seconds to update.

But I _did_ have success in dragging the `0006 On/Off` and `0007 Level Control` `Dimmable Light` clusters to the Bindings tab and now manually changing the light is reflected within one second in Home Assistant, I call that a win.
[image]
```json
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Leviton",
  "modelid": "DG6HD",
  "product": "DG6HD",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/dg6hd.json",
  "subdevices": [
    {
      "type": "$TYPE_DIMMABLE_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/bri",
          "description": "The current brightness.",
          "refresh.interval": 5
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0008"
    }
  ]
}
```
Note: I changed status to `Gold` because I don't know what I'm doing and had issues getting deconz to update my DDF changes.

mdolnik, 2023-05-11:
> For the second part, few device support that, but it seem you device is able to do that, using the endpoint 0x03, from that I m seing the 0x01 is for the "light" and the 0x03 for the "switch".
This is possible too using the DDF, need to add a ZHASwitch sensor ann probably edit the buttonmap file.

This is where I may need your help... In my deep dive, I attempted to manually add the `"type": "$TYPE_SWITCH"` section which I have seen in other DDF files and, while adding that section _does_ add a new entry to Phoscon's switches section... 
[image]
...it still does not appear in Home Assistant's event listener for `deconz_event`.

Also dragging in the same clusters for the `03 Dimmer Switch` endpoint, does not help.

```json
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Leviton",
  "modelid": "DG6HD",
  "product": "DG6HD",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/dg6hd.json",
  "subdevices": [
    {
      "type": "$TYPE_DIMMABLE_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/bri",
          "description": "The current brightness.",
          "refresh.interval": 5
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
[... 144 lines cut ...]
18:35:19:364 ZCL attribute report 0x0022A300003D4D64 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000
18:35:19:365 	payload: 00001001
18:35:19:372 ZDP skip fetch, node 0x0022A300003D4D64 has unconfirmed requests [1]
18:35:19:938 ZCL attribute report 0x0022A300003D4D64 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000
18:35:19:938 	payload: 00002029
18:35:20:478 ZCL attribute report 0x0022A300003D4D64 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000
18:35:20:478 	payload: 00001000
18:35:20:966 ZCL attribute report 0x0022A300003D4D64 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000
18:35:20:967 	payload: 00002000
18:35:21:480 ZCL attribute report 0x0022A300003D4D64 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000
18:35:21:480 	payload: 00001001
18:35:21:941 ZCL attribute report 0x0022A300003D4D64 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000
18:35:21:942 	payload: 00002014
18:35:22:985 ZCL attribute report 0x0022A300003D4D64 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000
18:35:22:985 	payload: 00002014
18:35:23:040     * neighbor: 0x0022A300003D4D64 (0xC13B), LQI: 56, relation: 0x02, depth: 15, rxOnWHenIdle: 1
18:35:23:040 ZDP add 0x0022A300003D4D64, nwk: 0xC13B to discover queue
18:35:24:170 Node 0x0022A300003D4D64 is known by 7 neighbors, last seen 1002 s
18:35:24:349 ZCL attribute report 0x0022A300003D4D64 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000
18:35:24:350 	payload: 00002014
```

I assume that payload _may_ include info on the button that got pressed, but there are more unique payload values (14) than there are buttons to press (4).

Any advice on what I should try next would be greatly appreciated.

Smanar, 2023-05-11:
Fot the first part, you have set the bind, but not the reporting, you need to have something like
```

    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 5,
          "max": 300
        }
      ]
    }
```

But take care here the value used for the poll NEED to be superior at the value used for reporting ("refresh.interval", the poll is the last chance action).

>So I think the next step is the button map?

Exactly
And yes, there is something missing, there is no event from the device on your logs, can use just logs with "info" and "info_l2" (so --dbg-info=2 with command line) to see them, and the usefull log not contain the mac adress, it's something like 

`19:42:08:507 [INFO] - No button map for: Remote switch Wake up / Sleep, broadcast to: 0xFFF5, endpoint: 0x01, cluster: SCENES (0x0005), command: RECALL_SCENE (0x05), payload: F5FF010000`

mdolnik, 2023-05-13:
> Fot the first part, you have set the bind, but not the reporting, you need to have something like

> But take care here the value used for the poll NEED to be superior at the value used for reporting ("refresh.interval", the poll is the last chance action).

I don't quite understand what value poll is supposed to be... I kept `refresh.interval` at its default of 5 but I added your suggested bind config and it seems to work, but I didn't notice any difference.

> can use just logs with "info" and "info_l2" (so --dbg-info=2 with command line) to see them

👍

> The uid for the sensor is incomplete

I have updated this to your suggestion, but I'm curious, I assume `0x03` + `0x0006` refers to the ON/OFF cluster of the Dimmer Switch endpoint? Does this mean that the switch would only listen to the physical on/off buttons and not the dimmer buttons?
Would adding another `$TYPE_SWITCH` entry for the `0x0006` Level Control add listeners for the dimmer buttons?

I have made your suggestions and the following is my current version of the DDF file, but I am not seeing any `No button map` logs for the Leviton switch when I press any of the buttons...

```json
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Leviton",
  "modelid": "DG6HD",
  "product": "DG6HD",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/dg6hd.json",
  "subdevices": [
    {
      "type": "$TYPE_DIMMABLE_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/bri",
          "description": "The current brightness.",
          "refresh.interval": 5
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x03",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/group",
          "default": "auto"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent",
          "awake": true
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 5,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0008",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 5,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 3,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 3,
      "cl": "0x0008"
    }
  ]
}
```

Do you have any other suggestions? If need-be, I am able to put extra print statements within the cpp files and compile in order to get a better sense as why I may not be getting button map logs.

Smanar, 2023-05-14:
>I have updated this to your suggestion, but I'm curious, I assume 0x03 + 0x0006 refers to the ON/OFF cluster of the Dimmer Switch endpoint? Does this mean that the switch would only listen to the physical on/off buttons and not the dimmer buttons?

No it's more "cosmetic", can use all, need to choose one, and we generaly choose the on/off.

>I don't quite understand what value poll is supposed to be... I kept refresh.interval at its default of 5 but I added your suggested bind config and it seems to work, but I didn't notice any difference

Better to set a bigger value for "refresh.interval", but on logs you can see (with 'info" and "info_l2" if the device use report or poll, not the same message)

>I have made your suggestions and the following is my current version of the DDF file, but I am not seeing any No button map logs for the Leviton switch when I press any of the buttons...

Perhaps the device finally don't support it, it's not used on z2m too https://www.zigbee2mqtt.io/devices/DG6HD-1BW.html

An error message can be visible on logs ? A not working bind.

BabaIsYou, 2023-05-19:
My 2 cents : I hoped that some "magic" was hidden in specific cluster 0x8007 (as suspected in #2483), but I just received informations for this cluster and that will be no help for getting button.

BTW I proposed to update general.xml with gathered informations in PR #6987 ;-)

Smanar, 2023-05-19:
Ha right, perhaps a setting "decoupling" like for the Xiaomi.

closed 2023-06-18 by github-actions[bot]