dresden-elektronik/deconz-rest-plugin #6950: IKEA Tradfri Open/Close Remote E1766 - Stopped Working

jotzet79, 2023-05-05:
## Describe the bug

- Pressing any button on the remote has no effect 
- Previously ("some" months ago) this was working correctly without further ado...

## Steps to reproduce the behavior

- I have 2 IKEA Fyjrtur Shades connected to deCONZ and the IKEA Tradfri Open/Close Remote. 
- In the deconz Legacy UI Tool I created the group which includes the 2 shades 
- Pressing any button on the remote has no effect 
- If I control the shades using Actions mapped to the Remote within Homebridge / HomeKit it works, meaning the Events seem to be triggered
- Deleting the remote from within deConz and Re-connecting it has also no effect

## Expected behavior

- All devices assigned to that group are controlled correctly

## Screenshots

## Environment

 - Host system: Raspberry Pi 
 - Running method: Raspbian 
 - Firmware version: 0x26690700
 - deCONZ version: v2.21.2
 - Device: RaspBee II
 - Do you use an USB extension cable: n/a
 - Is there any other USB or serial devices connected to the host system? nope

## deCONZ Logs

## Additional context

[image]

Mimiix, 2023-05-05:
Hi,

I have this one at home. Will test this afternoon.

Mimiix, 2023-05-05:
Hi,

I have tested on 2.21.1 and 2.22.0 ( can't test on 2.21.2 as i am on beta). On both occasions, it works as expected when i add it to the group:
[image]

So i am not able to reproduce this.

ebaauw, 2023-05-05:
The remote no longer sends group commands with its latest firmware.  You should still be able to control the blinds directly, by creating unicast bindings from the remote to each blind.

Note that IKEA probably changed this, as it was stretching the Zigbee spec, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3881.

jotzet79, 2023-05-05:
@ebaauw Wow, Thank you - I will definitely look into this! 

@Mimiix, Can you check what is your firmware version, please? Just to rule out different things here...

Thanks,
Joachim

Mimiix, 2023-05-05:
> @ebaauw Wow, Thank you - I will definitely look into this!
> 
> @Mimiix, Can you check what is your firmware version, please? Just to rule out different things here...
> 
> Thanks, Joachim

> @ebaauw Wow, Thank you - I will definitely look into this!
> 
> @Mimiix, Can you check what is your firmware version, please? Just to rule out different things here...
> 
> Thanks, Joachim

I never updated it since I got it. Date code is 20190311 , application version is 34.

Anyway, as i can't reproduce and Ebaauw giving a possible explanation, i am closing this issue.

We can still discuss a bit.

closed 2023-05-05 by Mimiix

olemr, 2023-05-07:
Is this what I'm seing here?
https://forum.phoscon.de/t/tradfri-open-close-switch-with-fw-2-3-079-fails-to-pair-with-blind/3518

ebaauw, 2023-05-07:
Yes, looks like.  If memory serves, the "association" in the old web app consists of adding the group bound to the controller (i.c. the open/close remote) to the actuator (i.c. the blind).

olemr, 2023-05-07:
IC. 
It is possible to do the association from somewhere else, and if so, how? (Have just used the Old Web App so far)

ebaauw, 2023-05-07:
The "association" is still there, but the open/close remote no longer sends group commands.  As I mentioned above, you could add unicast bindings from the remote to each blind individually, from the _Binding_ panel in the GUI.  See the user manual for details.