dresden-elektronik/deconz-rest-plugin #6958: 2 gang Switch module (_TZ3000_zmy4lslw - TS0002)

ThiemeNL, 2023-05-07:
## Device
Product name: 2CH Zigbee Switch Module (Tuya Smart ZigBee 3.0 Switch Module)
[AliExpress](https://aliexpress.com/item/1005003417422886.html)
Manufacturer: _TZ3000_bvrlqyj7
Model identifier: TS0002
Device type : 2 gang/ch Switch

The device is connected reports the states properly but switching on/off is always applied to both channels.

Currently I have loaded this file: https://gist.githubusercontent.com/ThiemeNL/ce8f0db07da2c4d78b99ad7ab576ccfe/raw/990585a12a6ed5fe6519a639db18df2656287a3f/TZ3000_bvrlqyj7.json
I change the vendor to something custom, so I could see that my changes where loaded.

[image]

Related to #6446

Mimiix, 2023-05-07:
Please add all requested screenshots.

Smanar, 2023-05-08:
ATM the actual used DDf by the user is 

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_bvrlqyj7",
  "vendor": "MYNAME",
  "modelid": "TS0002",
  "product": "Tuya 2 output module",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",                   
          "refresh.interval": 86400,
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on"
        },
        {
          "name": "state/reachable"
        },
        {
          "name": "config/interfacemode",
          "refresh.interval": 86400,
          "read": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": [
              "0x0004",
              "0x0000",
              "0x0001",
              "0x0005",
              "0x0007",
              "0xfffe"
            ]
          },
          "parse": {
            "fn": "zcl",
            "ep": 1,
            "cl": "0x0000",
            "at": "0xfffe",
            "eval": "Item.val = Attr.val"
          },
          "public": false,
          "default": 0
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x02"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",                   
          "refresh.interval": 86400,
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on"
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```

There is the tuya unlock sequence, but it seem this one is not working.

BabaIsYou, 2023-05-08:
A bit strange because in other projects they also use the Tuya magic sequence to unlock this device among others TS0002 : https://github.com/Koenkk/zigbee-herdsman-converters/issues/5040#issuecomment-1329927044

ThiemeNL, 2023-05-08:
Hi thank you all for the enthusiastic response. I'll provide you with the appropriate information, before I do. I also want to make sure I'm running on the latest firmware and app version.

ThiemeNL, 2023-05-08:
Hi guys, updating deconz and rebooting the system fixed the issue. I'm really really sorry.
Thanks for the support.

[image]
To
[image]

closed 2023-05-08 by ThiemeNL