dresden-elektronik/deconz-rest-plugin #6961: Tuya  IH012-RT01 Présence sensor

EricJeedom, 2023-05-09:
## Device

- Product name: Présence sensor
- Manufacturer: Tuya
- Model identifier: IH012-RT01 / ZN291438 (2 P/N in the box)
- Device type : 
  - Sensor
   

## Screenshots

[image]

https://fr.aliexpress.com/item/1005003371270522.html?spm=a2g0o.productlist.0.0.53764d625njsuE&algo_pvid=67148651-00e9-4ce2-8ef1-a42bd6d6d031&aem_p4p_detail=20220913045740335693757769800000488720&algo_exp_id=67148651-00e9-4ce2-8ef1-a42bd6d6d031-3&pdp_ext_f=%7B%22sku_id%22%3A%2212000025470295348%22%7D&pdp_npi=2%40dis%21EUR%2151.93%2131.16%21%21%21%21%21%400b0a187916630702602865316e918d%2112000025470295348%21sea&curPageLogUid=GNg3cCKIattE&ad_pvid=20220913045740335693757769800000488720_4

{
    "ieee": "a4:c1:38:37:3f:8d:2f:11",
    "nwk": 28007,
    "status": 2,
    "lqi": "116",
    "rssi": "-71",
    "last_seen": "1663068647.562169",
    "node_descriptor": "02:40:80:41:11:42:42:00:00:2a:42:00:00",
    "endpoints": [
        {
            "id": 1,
            "status": 1,
            "device_type": 1026,
            "profile_id": 260,
            "manufacturer": "_TZ3000_mcxw5ehu",
            "model": "TS0202",
            "output_clusters": [
                {
                    "id": 4096,
                    "name": "LightLink",
                    "attributes": []
                },
                {
                    "id": 6,
                    "name": "On\/Off",
                    "attributes": []
                },
                {
                    "id": 25,
                    "name": "Ota",
                    "attributes": []
                },
                {
                    "id": 10,
                    "name": "Time",
                    "attributes": []
                }
            ],
            "input_clusters": [
                {
                    "id": 1,
                    "name": "Power Configuration",
                    "attributes": []
                },
                {
                    "id": 1280,
                    "name": "IAS Zone",
                    "attributes": [
                        {
                            "id": 16,
                            "name": "cie_addr",
                            "value": [
                                137,
                                107,
                                150,
                                254,
                                255,
                                249,
                                227,
                                180
                            ]
                        },
                        {
                            "id": 19,
                            "name": "current_zone_sensitivity_level",
                            "value": 1
                        }
                    ]
                },
                {
                    "id": 3,
                    "name": "Identify",
                    "attributes": []
                },
                {
                    "id": 0,
                    "name": "Basic",
                    "attributes": [
                        {
                            "id": 0,
                            "name": "zcl_version",
                            "value": 3
                        },
                        {
                            "id": 1,
                            "name": "app_version",
                            "value": 70
                        },
                        {
                            "id": 2,
                            "name": "stack_version",
                            "value": 0
                        },
                        {
                            "id": 3,
                            "name": "hw_version",
                            "value": 1
                        },
                        {
                            "id": 4,
                            "name": "manufacturer",
                            "value": "_TZ3000_mcxw5ehu"
                        },
                        {
                            "id": 5,
                            "name": "model",
                            "value": "TS0202"
                        },
                        {
                            "id": 6,
                            "name": "date_code",
                            "value": ""
                        },
                        {
                            "id": 7,
                            "name": "power_source",
                            "value": 3
                        }
                    ]
                }
            ]
        }
    ],
    "signature": {
        "manufacturer": "_TZ3000_mcxw5ehu",
        "model": "TS0202",
        "node_desc": {
            "logical_type": 2,
            "complex_descriptor_available": 0,
            "user_descriptor_available": 0,
            "reserved": 0,
            "aps_flags": 0,
            "frequency_band": 8,
            "mac_capability_flags": 128,
            "manufacturer_code": 4417,
            "maximum_buffer_size": 66,
            "maximum_incoming_transfer_size": 66,
            "server_mask": 10752,
            "maximum_outgoing_transfer_size": 66,
            "descriptor_capability_field": 0
        },
        "endpoints": {
            "1": {
                "profile_id": 260,
                "device_type": 1026,
                "input_clusters": [
                    1,
                    1280,
                    3,
                    0
                ],
                "output_clusters": [
                    4096,
                    6,
                    25,
                    10
                ]
            }
        }
    },
    "class": "zigpy.device"
}

Mimiix, 2023-05-09:
Please add all screenshots.

BabaIsYou, 2023-05-10:
> Please add all screenshots.

Will be essential because if this device is really ```
 "manufacturer": "_TZ3000_mcxw5ehu", "model": "TS0202"``` , it is already supported : https://github.com/dresden-elektronik/deconz-rest-plugin/blob/f837452b4eec614746e3ba56f214b1096342a797/devices/tuya/ts0202_presence_sensor.json#L3

Mimiix, 2023-05-31:
Was already supported :)

closed 2023-05-31 by Mimiix