dresden-elektronik/deconz-rest-plugin #6972: New device request - TS0601 / TZE200_locansqn

Gorole, 2023-05-14:
Hi,
i buy this thermomether on Aliexpress :-)
[](https://www.aliexpress.com/item/1005005097691633.html?spm=a2g0o.order_list.order_list_main.22.21ef1802YgTjQo
)
but deconz is not supported :-(

- Product name: Tuya TH01Z-WHITE/BLACK
- Manufacturer: _TZE200_locansqn
- Model identifier: TS0601
- Device type : Please remove all unrelated device types. 
  - Thermostat
[image]
[image]
[image]
[image]
[image]

Mimiix, 2023-05-14:
Hi, can you please add all screenshots ?

Gorole, 2023-05-15:
Done

Mimiix, 2023-05-15:
Great :)

BabaIsYou, 2023-05-15:
Can you try this DDF ?
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZE200_locansqn","_TZE200_lve3dvpy","_TZE200_c7emyjom"],
  "modelid": ["TS0601","TS0601","TS0601"],
  "product": "Tuya temp and humidity sensor with clock",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",                   
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = 10 * Attr.val;" },
          "read": {"fn": "tuya"},
          "default": 0
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",                   
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = 10 * Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        }
      ]
    }
  ]
}
```

Could get/configure other informations using other dpip but will see furthermore if we could use them in REST-APi

 TempUnitConvert: 9,
 MaxTemp: 10,
 MinTemp: 11,
 MaxHumi: 12,
  MinHumi: 13,
 TempAlarm: 14,
  HumiAlarm: 15,
 TempSensitivity: 19,
  ReportInterval: 17

Gorole, 2023-05-17:
Iam running on HAOS.
I dont know how i edit DDF... :-(

BabaIsYou, 2023-05-17:
May be you could find some information here : https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6666#issuecomment-1387646724

Gorole, 2023-05-23:
It's much better.
The device already has the correct name, but unfortunately it does not send values.
[image]
[image]

Gorole, 2023-06-18:
> Do you see something special in log ?

Whitch log do you mean?
[image]

mkress, 2023-06-28:
I have also bought this device - looks like it is related to
[_TZE200_locansqn](https://dresden-elektronik.github.io/deconz-rest-doc/devices/tuya/nous_hum_temp_e6/)

see https://github.com/zigpy/zha-device-handlers/issues/1286 and https://smarthomescene.com/reviews/zigbee-temperature-humidity-sensor-with-a-screen-sz-t04-review/

mkress, 2023-06-28:
adding the manufacturername _TZE200_locansqn to the ddf file /usr/share/deCONZ/devices/tuya/_TZE200_TS0601_humidity_temp.json helped

temperature and humidity are now there - only battery is missing/0

` "manufacturername": ["_TZE200_locansqn","_TZE200_bjawzodf"`...

api/... /sensors now returns:
```
"41": {
"config": {
"battery": 0,
"offset": 0,
"on": true,
"reachable": true
},
"etag": "734bcdef1d539696aff6dc08e3fd90f9",
"lastannounced": "2023-06-28T19:15:00Z",
"lastseen": "2023-06-28T19:15Z",
"manufacturername": "_TZE200_locansqn",
"modelid": "TS0601",
"name": "Humidity 41",
"state": {
"humidity": 4500,
"lastupdated": "2023-06-28T19:15:40.296"
},
"swversion": "1.0.8",
"type": "ZHAHumidity",
"uniqueid": "a4:c1:38:09:a1:c1:63:cd-01-0405"
},
"42": {
"config": {
"battery": 0,
"offset": 0,
"on": true,
"reachable": true
},
"etag": "4ecd62a4f980326caf25889d126d2ffe",
"lastannounced": "2023-06-28T19:15:00Z",
"lastseen": "2023-06-28T19:15Z",
"manufacturername": "_TZE200_locansqn",
"modelid": "TS0601",
"name": "Temperature 42",
"state": {
"lastupdated": "2023-06-28T19:15:40.176",
"temperature": 2440
},
"swversion": "1.0.8",
"type": "ZHATemperature",
"uniqueid": "a4:c1:38:09:a1:c1:63:cd-01-0402"
},
```

closed 2023-07-28 by github-actions[bot]

azlux, 2023-08-12:
On my side, I've got the same device, but the humidity was reported as 5.5% instead of 55%.

So I've modify the json with :
```json
        {
          "name": "state/humidity",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = 100 * Attr.val;" },
          "read": {"fn": "tuya"},
          "default": 0
        },

```

Not sure if it's the good way (first time I modify deconz config).
Thank for the hardware and tools btw, it's amazing.
Azlux

azlux, 2023-08-12:
About the `TempSensitivity` configurable option, do you have a wiki/tuto to edit them manually, since it's not in the API yet ?
I didn't found the option with the VNC Deconz App.

BabaIsYou, 2023-08-12:
Where did you get this 5.5% ? Not thru Rest-Api surely !

BabaIsYou, 2023-08-12:
﻿

About the TempSensitivity configurable option, do you have a wiki/tuto to edit them manually, since it's not in the API yet ?
I didn't found the option with the VNC Deconz App.

Using GUI and request command from Tuya Cluster or could be added into DDF using attribute config/sensitivity into temperature sensor subdevice. A bit like it as been done for https://github.com/dresden-elektronik/deconz-rest-plugin/blob/0063906e24b99ddb952b9078329423ae868472d5/devices/tuya/ZY-M100_human_breathing_presence.json#L200C20-L200C20 but with dpid 19 Message ID: ***@***.***>

azlux, 2023-10-31:
My current file is : 
```json
{
  "schema": "devcap1.schema.json",
  "manufacturername": [
    "_TZE200_locansqn",
    "_TZE200_lve3dvpy",
    "_TZE200_c7emyjom"
  ],
  "modelid": [
    "TS0601",
    "TS0601",
    "TS0601"
  ],
  "vendor": "Tuya",
  "product": "Tuya temp and humidity sensor with clock",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 4,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "read": {
            "fn": "tuya"
          },
          "parse": {
            "dpid": 2,
            "eval": "Item.val = 100 * Attr.val;",
            "fn": "tuya"
          },
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 4,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/sensitivity",
          "description": "Report temperature sensibility (from 1 to 10), in *10 value (6 for 0.6 precision). Default 6",
          "range": [1, 10],
          "parse": {
            "dpid": 19,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "read": {
            "fn": "none"
          },
          "write": {
            "dpid": 19,
            "eval": "Item.val / 10;",
            "fn": "tuya"
          },
          "default": 0
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 1,
            "eval": "Item.val = 10 * Attr.val;",
            "fn": "tuya"
          },
          "default": 0
        }
      ]
    }
  ]
}
```
I didn't find a good doc about the read/write config, not sure if I've read it well.

To config my sensitivity, I use the API:
```bash
curl -X GET -H "Content-Type: application/json" "http://127.0.0.1:70/api/<API-KEY>/sensors/" | jq
```
-> search for the ID of your sensor
```bash
curl -X PUT -H "Content-Type: application/json" "http://127.0.0.1:70/api/<API-KEY>/sensors/<SENSOR-ID>/config" --data '{"sensitivity": 1}'
```
-> confirmation with : 
```bash
curl -X GET -H "Content-Type: application/json" "http://127.0.0.1:70/api/<API-KEY>/sensors/<SENSOR-ID>/" | jq
```

That is the max I can do, I don't know how to get this value spawn on home-assistant (for my case)

> Where did you get this 5.5% ? Not thru Rest-Api surely !

@BabaIsYou I divide by /100 on my home-assistant. Deconz was giving me 500 instead of 5000

Gorole, 2024-02-09:
Thank you for replies :-)
MY DDF for _TZE200_locansqn is:
`{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZE200_locansqn","_TZE200_lve3dvpy","_TZE200_c7emyjom"],
  "modelid": ["TS0601","TS0601","TS0601"],
  "product": "Tuya temp and humidity sensor with clock",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",                   
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = 100 * Attr.val;" },
          "read": {"fn": "tuya"},
          "default": 0
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",                   
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = 10 * Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        }
      ]
    }
  ]
}`

Gorole, 2024-02-09:
`{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZE200_locansqn","_TZE200_lve3dvpy","_TZE200_c7emyjom"],
  "modelid": ["TS0601","TS0601","TS0601"],
  "product": "Tuya temp and humidity sensor with clock",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",                   
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = 100 * Attr.val;" },
          "read": {"fn": "tuya"},
          "default": 0
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",                   
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = 10 * Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        }
      ]
    }
  ]
}`