dresden-elektronik/deconz-rest-plugin #6976: plug _TZ3000_gvn91tmx

lepeno82, 2023-05-15:
## Device

- Product name: aubess plug
- Manufacturer: _TZ3000_gvn91tmx
- Model identifier: TS011F
- Device type : plug 
 

## Screenshots

### Basic
[image]

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off
[image]

### Simple Metering
[image]

### Diagnostics

### Other clusters that are not mentioned above
[image]

BabaIsYou, 2023-05-15:
Looks like duplicate of #6423

Mimiix, 2023-05-16:
Yep, Agree.

Closing.

closed 2023-05-16 by Mimiix